Xilinx/Vitis-AI #1240: Can not build Petalinux 2022.1 with VitisAI 2.5 - Fetch fails, mirrors not available

fschramka, 2023-05-17:
Added VitisAI 2.5 as user package like described in the manual. 

While building VitisAI this output is shown: 

```
WARNING: unilog-2.5-r0 do_fetch: Failed to fetch URL git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.5, attempting MIRRORS if available
ERROR: unilog-2.5-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; (...)
ERROR: unilog-2.5-r0 do_fetch: Fetcher failure for URL: 'git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.5'. Unable to fetch URL from any source. 
WARNING: xir-2.5-r0 do_fetch: Failed to fetch URL git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.5, attempting MIRRORS if available
ERROR: xir-2.5-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; (...)
WARNING: target-factory-2.5-r0 do_fetch: Failed to fetch URL git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.5, attempting MIRRORS if available
ERROR: target-factory-2.5-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; (...)
ERROR: target-factory-2.5-r0 do_fetch: Fetcher failure for URL: 'git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.5'. Unable to fetch URL from any source.
WARNING: vitis-ai-library-2.5-r0 do_fetch: Failed to fetch URL git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.5, attempting MIRRORS if available
ERROR: vitis-ai-library-2.5-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; (...)
WARNING: vart-2.5-r0 do_fetch: Failed to fetch URL git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.5, attempting MIRRORS if available
ERROR: vart-2.5-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; (...)
ERROR: vart-2.5-r0 do_fetch: Fetcher failure for URL: 'git://gitenterprise.xilinx.com/Vitis/vitis-ai-staging;protocol=https;branch=2.5'. Unable to fetch URL from any source. 
```

I removed the paths within the error message to make it more "readable" 

Rly have no idea what to do here...

AlbertaBeef, 2023-05-17:
There are two releases : one is a branch called "2.5", and the other is a tag called "v2.5".
You are using the last one, but should be using the first one.

Make sure you are using this one (which has a valid (public) github.com repository URL):
https://github.com/Xilinx/Vitis-AI/blob/2.5/setup/petalinux/recipes-vitis-ai/vitis-ai-library/vitisai.inc

Do not use this one (which has the invalid (private) gitenterprise.xilinx.com URL):
https://github.com/Xilinx/Vitis-AI/blob/v2.5/setup/petalinux/recipes-vitis-ai/vitis-ai-library/vitisai.inc

Cheers !
Mario.

fschramka, 2023-05-24:
Well this was it! Thanks a lot @AlbertaBeef !

I took the https://github.com/Xilinx/Vitis-AI/releases/tag/v2.5 Release - This should be changed in there as well. Releases are here to be used :D