Xilinx/Vitis-AI #1245: VAI3.0 generates multiple subgraphs if activation = HardSwish

three-cups, 2023-05-24:
Hi

I'm trying to deploy YOLOX-Nano/Tiny into DPU.
https://github.com/Megvii-BaseDetection/YOLOX

Vitis-AI Version : 3.0
DPU IP Version : DPUCZDX8G 4.1
WorkFlow : pytorch

DPU supports HardSwish.
I had customized network scripts to use activation `nn.Hardswish()`.
Quantize calib/test and compile process goes with no error, but generated xmodel contains 11 subgraphs. (bad)
I had modified activation to `nn.ReLU()` and `nn.LeakyReLU()`, generated xmodels contains only 1 subgraph. (good)

I checked generated xmodel, which contains 11 subgraphs.
Some activation layers are assigned to DPU, but some are not (assigned to CPU).
Is there any way to generate xmodel contains single subgraph?

[yolox_nano.zip](https://github.com/Xilinx/Vitis-AI/files/11551426/yolox_nano.zip)

This problem is similer to #593, #1214.
I want to try optional argument `separate_conv_act=False`, but this argument is only in TensorFlow version.
PyTorch version not accepts this argument. (by config.json)

quentonh, 2023-08-09:
@jiaz-xlnx Is it reasonable to assume that for the DPUCZ that Hardswish is not supported as an activation type following eltwise, simply because the Eltwise engine is a separate block?  This is to say that though the DPU supports Hardswish for operators such as CONV2D, it will not support this activation following Eltwise?

JasonXijieJia, 2023-08-10:
hi @quentonh , @three-cups 
  Yes, currently DPUCZ/DPUCV don't support hardswish as an activation type following eltwise. Please try to insert fix between eltwise and hardswish to compute them in sequence on dpu.

three-cups, 2023-08-10:
hi @quentonh @JasonXijieJia 

Where are the eltwise layers that affect hardswish placement?

I'm checking for `yolox_nano.xmodel' I'd uploaded.
This `hard-sigmoid` layer is placed on CPU.
`YOLOX__YOLOX_YOLOPAFPN_backbone__CSPDarknet_backbone__Sequential_dark2__CSPLayer_1__BaseConv_conv2__Hardswish_act__22793_i0`

but previous layer type is `conv2d-fix`, not `eltwise`.
`YOLOX__YOLOX_YOLOPAFPN_backbone__CSPDarknet_backbone__Sequential_dark2__CSPLayer_1__BaseConv_conv2__Conv2d_conv__input_25`

According to the documentation, both conv2d and depthwise-conv2d support HardSwish. (DPUCZDX8G).
https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Currently-Supported-Operators

JasonXijieJia, 2023-08-11:
Hi @quentonh ,
  Yes, on the DPUCZ for the old device, we can't support h-swish as the activation of CONV like what we have supported on DPUCV and DPUCV2 powered by AIE and AIE-ML. 
  Thus on the DPUCZ we instead support the Conv + h-swish by mapping h-swish as a Non-Conv OP as [here](https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_runtime/target_factory/targets/DPUCZDX8G_ISA1_B3136.prototxt#L123) which has to require model to insert a fix between Conv and h-swish.

> Sorry @three-cups. A colleague and I were discussing a very closely related issue and I incorrectly assumed that your post was from the same team.
> 
> @JasonXijieJia is it possible that we have this incorrectly documented? In UG1414 we do state support for h-swish for the DPUCZ, and yet, when I look [here](https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_runtime/target_factory/targets/DPUCZDX8G_ISA1_B3136.prototxt?rgh-link-date=2023-08-10T18%3A43%3A34Z#L83) and compare it to what I see [here](https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_runtime/target_factory/targets/DPUCV2DX8G_ISA1_C20B1.prototxt?rgh-link-date=2023-08-10T18%3A43%3A34Z#L55) it would seem that DPU non-linear support for the CONV engine appears to differ.

quentonh, 2023-08-17:
@JasonXijieJia are you suggesting that we replace hwish with hsigmoid and insert a FIXNEURON between the CONV and activation?  Can you provide any pointers as to how we insert a FIXNEURON?  I cannot find a reference for this.

JasonXijieJia, 2023-09-12:
Hi @quentonh  @three-cups 
  The issue should have been resolved in vai3.5. Please re-try with vai3.5 and feel free to feedback any new issues.
  Thank you

closed 2023-09-14 by three-cups