Xilinx/Vitis-AI #1249: ViT-Demo

hocken-li, 2023-05-29:
I find the ViT-Demo in the vitis-ai-2.5, examples/transformer.
I find there is only the .xclbin file that complied by VCK5000? I want to develop it in U250, Could you share the files that how to build the .xclbin？？？

SuZhan0322, 2023-06-13:
Hi @hocken-li , The VIT demo released in VAI2.5 only can be supported by VCK5000.  There is no corresponding xclbin for U250. Thanks!

hocken-li, 2023-07-28:
> Hi @hocken-li , The VIT demo released in VAI2.5 only can be supported by VCK5000. There is no corresponding xclbin for U250. Thanks!

Hi, @SuZhan0322 
May I ask you a question?
I have buy the VCK5000 and try the demo, but it didn't success
The dmesg is
```
Run BERT_Base:
 
[XCLBIN    ] ./testcase/bert_base.xclbin
[MODEL     ] ./testcase/bert_base.bm
[INPUT     ] ./testcase/squad1.1/dev-v1.1.json
[THREADNUM ] 3
Begin parsing dateset...
Finish parsing dateset
======= Total Number of Examples: 10570
======= Total Number of features: 21363
XRT build version: 2.14.384
Build hash: 090bb050d570d2b668477c3bd0f979dc3a34b9db
Build date: 2022-12-09 00:55:08
Git branch: 2022.2
PID: 8292
UID: 1001
[Thu Jul 27 11:09:14 2023 GMT]
HOST: host-scut-hg-fpga-1
EXE: /data/Vitis-AI/lhq_workspace/VCK5000_BERT_Demo_2.5.0/build/bin/bert_accuracy
[XRT] ERROR: See dmesg log for details. err = -22
terminate called after throwing an instance of 'xrt_core::system_error'
  what():  failed to load xclbin: Invalid argument
./test_accuracy.sh: line 24:  8292 Aborted                 (core dumped) ./build/bin/bert_accuracy -b ./testcase/bert_base.xclbin -m ./testcase/bert_base.bm -i ./testcase/squad1.1/dev-v1.1.json -v ./testcase/squad1.1/vocab_en.txt -t 3
calculating the accuracy...
Traceback (most recent call last):
  File "/data/Vitis-AI/lhq_workspace/VCK5000_BERT_Demo_2.5.0/scripts/evaluate-v1.1.py", line 104, in <module>
    with open(args.prediction_file) as prediction_file:
FileNotFoundError: [Errno 2] No such file or directory: 'demo_pred.json'
```
```
[ 1844.159927] cgroup: cgroup: disabling cgroup2 socket matching due to net_prio or net_cls activation
[ 1898.756014] xocl 0000:56:00.1:  ffff9ca8ddb170b0 _xocl_drvinst_open: OPEN 1
[ 1898.756145] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_create_client: created KDS client for pid(5280), ret: 0
[ 1898.756163] xocl 0000:56:00.1:  ffff9ca8ddb170b0 _xocl_drvinst_open: OPEN 2
[ 1898.756279] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_create_client: created KDS client for pid(5280), ret: 0
[ 1898.756296] xocl 0000:56:00.1:  ffff9ca8ddb170b0 _xocl_drvinst_open: OPEN 3
[ 1898.756881] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_create_client: created KDS client for pid(5280), ret: 0
[ 1898.769300] xocl 0000:56:00.1: xmc.u.18874370 ffff9ca876ce2010 xmc_read_from_peer: reading from peer
[ 1898.769320] xocl 0000:56:00.1: mailbox.u.9437186 ffff9ca876ce1410 _mailbox_request: sending request: 10 via HW
[ 1898.785982] xclmgmt 0000:56:00.0: mailbox.m.9437186 ffff9ca8d7669010 process_request: received request from peer: 10, passed on
[ 1898.785993] xclmgmt 0000:56:00.0: xclmgmt_read_subdev_req: req kind 0
[ 1898.786081] xclmgmt 0000:56:00.0: mailbox.m.9437186 ffff9ca8d7669010 mailbox_post_response: posting response for: 10 via HW
[ 1898.873139] xocl 0000:56:00.1: icap.u.22020098 ffff9ca8d7686410 icap_read_from_peer: reading from peer
[ 1898.873147] xocl 0000:56:00.1: mailbox.u.9437186 ffff9ca876ce1410 _mailbox_request: sending request: 10 via HW
[ 1898.873518] xclmgmt 0000:56:00.0: mailbox.m.9437186 ffff9ca8d7669010 process_request: received request from peer: 10, passed on
[ 1898.873528] xclmgmt 0000:56:00.0: xclmgmt_read_subdev_req: req kind 1
[ 1898.873557] xclmgmt 0000:56:00.0: mailbox.m.9437186 ffff9ca8d7669010 mailbox_post_response: posting response for: 10 via HW
[ 1898.874450] xocl 0000:56:00.1: icap.u.22020098 ffff9ca8d7686410 icap_cached_ocl_frequency: no cached data for 3
[ 1898.874494] xocl 0000:56:00.1: icap.u.22020098 ffff9ca8d7686410 icap_cached_ocl_frequency: no cached data for 3
[ 1898.874580] xocl 0000:56:00.1: icap.u.22020098 ffff9ca8d7686410 icap_cached_ocl_frequency: no cached data for 3
[ 1898.875468] xocl 0000:56:00.1:  ffff9ca8ddb170b0 _xocl_drvinst_open: OPEN 4
[ 1898.875568] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_create_client: created KDS client for pid(5280), ret: 0
[ 1898.875959] xocl 0000:56:00.1:  ffff9ca8ddb170b0 _xocl_drvinst_open: OPEN 5
[ 1898.876050] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_create_client: created KDS client for pid(5280), ret: 0
[ 1898.876131] xocl 0000:56:00.1:  ffff9ca8ddb170b0 _xocl_drvinst_open: OPEN 6
[ 1898.876519] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_create_client: created KDS client for pid(5280), ret: 0
[ 1898.926384] xocl 0000:56:00.1: icap.u.22020098 ffff9ca8d7686410 icap_cached_ocl_frequency: no cached data for 3
[ 1898.926820] xocl 0000:56:00.1: icap.u.22020098 ffff9ca8d7686410 icap_cached_ocl_frequency: no cached data for 3
[ 1898.927198] xocl 0000:56:00.1: icap.u.22020098 ffff9ca8d7686410 icap_cached_ocl_frequency: no cached data for 3
[ 1898.985189] [drm] Finding MEM_TOPOLOGY section header
[ 1898.985191] [drm] Section MEM_TOPOLOGY details:
[ 1898.985193] [drm]   offset = 0x29654f0
[ 1898.985193] [drm]   size = 0x80
[ 1898.985222] xocl 0000:56:00.1: icap.u.22020098 ffff9ca8d7686410 icap_download_bitstream_axlf: check interface uuid
[ 1898.985235] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_fdt_check_uuids: Can not find uuid e221a0ff8695d5eb8725fc65147f90c3
[ 1898.985238] xocl 0000:56:00.1: icap.u.22020098 ffff9ca8d7686410 icap_download_bitstream_axlf: interface uuids do not match
[ 1898.985241] xocl 0000:56:00.1: icap.u.22020098 ffff9ca8d7686410 icap_download_bitstream_axlf: err: -22
[ 1898.985246] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_read_axlf_helper: Failed to download xclbin, err: -22
[ 1899.196907] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_destroy_client: client exits pid(5336)
[... cut ...]
[ 1899.196950] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_destroy_client: client exits pid(5337)
[ 1899.196951] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_drvinst_close: CLOSE 2
[ 1899.196954] xocl 0000:56:00.1:  ffff9ca8ddb170b0 xocl_drvinst_close: NOTIFY 0000000028f6eff0
```

It seems is the uuid not match? How to solve it?
What is the Platform of the xclbin file? Is gen4×8_xdma_base_2 or gen4x8_qdma_base_2 ???
My VCK5000 Platform is already xilinx_vck5000_gen4x8_qdma_base_2. Is that the problem of the error?

SuZhan0322, 2023-07-30:
Hi @hocken-li , Thanks for your feedback! Please install the "xilinx_vck5000_gen3x16_xdma_base_1" platform which released in VAI2.5.  And run the "/opt/xilinx/xrt/bin/xbutil reset -d"  command before you run the VIT/BERT_base demo. Thanks!

hocken-li, 2023-07-31:
Hi, @SuZhan0322 
Thanks for your reply. There is a bad new that the latest VCK5000 platform "xilinx_vck5000_gen4x8_qdma_base_2" implements some features that make it incompatible with previous platforms (especially SC version 4.4.35).  So it is not success for me to return back to the platform "xilinx_vck5000_gen3x16_xdma_base_1".
Will you guys build ViT/BERT xclbin files on this platform? Why is it not available in VAI3.0?
Or are there any Tutorials to build it? 
Thanks!

SuZhan0322, 2023-07-31:
Hi @hocken-li , the VIT/BERT_base's xclbin need to match the platform version.  And the ViT/BERT's xclbin will not update on VCK5000. For transformer-related updates, please pay attention to the Ryzen AI release. Thanks!