Xilinx/Vitis-AI #1250: Using vai_q_tensorflow2 API quantize .h5 file error: list index out of range.

ChuDavid0624, 2023-05-29:
We are working on a FPGA project.Our target is to put a Yolov4 .weight file on KV260.
We had already convert our .weight(yolov4) file into .h5 file but now there is a problem.
When we use the vai_q_tensorflow2 API quantize our .h5 file. 
There is always an error:list index out of range.

[image]

[image]

We want to know how to solve this problem.
We are using vitis-AI 2.5.

minecraftdixit, 2023-07-03:
have you got the solution of this problem?

zhenzhen-AMD, 2023-07-04:
hi @ChuDavid0624 ,
Could you please provide the complete code, '2022_no_mish.h5' model file, and '2022_dataset' for us to better assist you?

zhenzhen-AMD, 2023-08-23:
Hi @ChuDavid0624,

Thank you for bringing this to our attention. However, due to the lack of detailed information in the initial report and no response to our follow-up questions, it's challenging for us to pinpoint and address the issue you're experiencing.

In order to maintain an organized issue tracker, we will be closing this issue for now. If you can provide more details or answer the previous questions, please feel free to reopen the issue or create a new one with the additional information.

Thank you for understanding.

closed 2023-08-23 by renjiangyong