Xilinx/Vitis-AI #1251: Matrix Multiplies on DPU

Rellek72, 2023-05-30:
I'm implementing a transformer-based architecture using Vitis AI and am having great difficulty using the simple two-input matrix multiply.

In Tensorflow, Dense layers work fine, however tf.linalg.matmul is unable to quantize since TensorFlow wraps it as a lambda function ([my existing github issue](https://github.com/Xilinx/Vitis-AI/issues/1217) on this).

Using PyTorch, both Linear layers and torch.matmul quantize without error, but when attempting to use VART the torch.matmul instances only ever compile to the CPU, regardless of the size of the matrices. Example code that demonstrates this is below:

```    
CLASSES = 10
     
class ExampleModel(Module):
     
    def __init__(self, dim1, dim2, device='cpu'):
        super(ExampleModel, self).__init__()
        self.lin_1 = nn.Linear(dim2, 1)
        self.lin_2 = nn.Linear(dim1, CLASSES)
     
    def forward(self, x):
        # For inputs of size [batch, dim1, dim2]
        y = x.permute(0, 2, 1)
        x = torch.matmul(x, y)
        x = self.lin_1(x)
        x = x.reshape(x.shape[0], x.shape[1])
        x = self.lin_2(x)
        return x
```

I've tested this both on versions 2.5 and 3.0 of Vitis AI and had the same issues on each. Any assistance or guidance would be greatly appreciated.

leoflcnatgm, 2023-07-20:
try:
from pytorch_nndct.nn.modules import functional
x = functional.matmul()(x,y)

Rellek72, 2023-07-20:
I have tried this, as well as all other forms of matmul in Pytorch, and none of them successfully quantize. I also replaced the transpose with a linear layer, and it still does not quantize. I have an updated test bench attached.

[matmul_example.zip](https://github.com/Xilinx/Vitis-AI/files/12114344/matmul_example.zip)

neuhsm, 2023-07-24:
i meet the same problem using torch.bmm, and get : [VAIQ_WARN]: The quantizer recognize new op `bmm` as a float operator by default.
how can i use matrix multiply in dpu?

quentonh, 2023-07-24:
@neuhsm May I suggest this thread?

https://support.xilinx.com/s/question/0D74U000007t4IsSAI/matmuls-wont-compile-for-ultrascale?language=en_US&fromEmail=1&s1oid=00D2E000000nHq7&s1nid=0DB2E000000XdtH&s1uid=0052E00000MpQYo&s1ext=0&emkind=chatterBestAnswerNotification&emtm=1690154656642

neuhsm, 2023-07-24:
> @neuhsm May I suggest this thread? 
> 
> https://support.xilinx.com/s/question/0D74U000007t4IsSAI/matmuls-wont-compile-for-ultrascale?language=en_US&fromEmail=1&s1oid=00D2E000000nHq7&s1nid=0DB2E000000XdtH&s1uid=0052E00000MpQYo&s1ext=0&emkind=chatterBestAnswerNotification&emtm=1690154656642

sorry，i am not able to operate it, and i'm trying to do something else to avoid the matrix multiply.

neuhsm, 2023-07-24:
> Perhaps this link is better: https://support.xilinx.com/s/question/0D54U0000786ERFSA2/matmuls-wont-compile-for-ultrascale?language=en_US

i'm sorry,it's time for sleep now, can I bother you again tomorrow?

Rellek72, 2023-09-13:
Per this issue, two-input matmuls are not supported (only one-input matmul with static weights). This is very understandable given the architecture of the DPU, however I would _highly_ suggest that the documentation be updated to make that explicitly clear. The current language of the user guide clearly announces support for matrix multiplies with no notice of this limitation, and in my previous conversations with Xilinx support staff it would seem many are under the same impression I was (see [here](https://support.xilinx.com/s/question/0D54U00006usdRHSAY/matrix-multiply-on-dpu?language=en_US)). Had I understood this the first time I read the docs, it would have saved me months of work.

Regardless, I cannot thank you enough for the support and workarounds with this issue, especially you @quentonh. a Senior Manager taking time out of his day to give paragraph responses on the [forums](https://support.xilinx.com/s/question/0D54U0000786ERFSA2/matmuls-wont-compile-for-ultrascale?language=en_US)? Seriously, what a hero. It's thanks to you I finished my Master's thesis on pace.

closed 2023-09-13 by Rellek72