Xilinx/Vitis-AI #1254: Having issues running a compiled model using vitis ai 2.0, smart-cam

DarkVader135, 2023-06-02:
Hi so the issue i am having is when i run the modified smart-cam i followed these two guides [Xilinx_Documentation](https://xilinx.github.io/kria-apps-docs/kv260/2022.1/build/html/docs/smartcamera/docs/customize_ai_models.html) & [related_github_workshop](https://github.com/Xilinx/Xilinx_Kria_KV260_Workshop/blob/main/Part%204%3A%20Optional%20Demo.md) to help me change the ai model from the simple face detect to a yolov3 model after compiling to xmodel format using vitis_AI 2.0 tensorflow conda-env for the reasons that it supports the DPUCZDX8G_ISA0_B3136_MAX_BG2 which was the accelerator used in the smart cam application i used this command `vai_c_tensorflow -m quantized.h5 -a arch.json -o ./ -n quantized --options "{'mode':'normal','save_kernel':'', 'input_shape':'1,416,416,3'}"` i copied the xmodel and md5sum.txt and prototxt to the kria via scp and ran this command after loading the smart cam application `sudo smartcam --file=video.h264 -W 1920 -H 1080 -r 30 --target=rtsp --audio --screenfps` on the same directory as the video.h264 which has been tested with the default facedetect model, after starting the smart-cam app i get this error 
[image]
i am not sure to be honest what i did wrong or what i missed, if anyone knows anything please reply

in this zip i included aiinference.json prototxt xmodel and the md5sum with the meta.json and arch.json for reference, the xmodel was generated using an already quantized model from vitis ai model zoo and compiled for b3136 dpu which the smart-cam uses [Download link](https://workupload.com/file/qr3NCcvbBzM) file is 60mb which is past 25mb github limit.

hetianz, 2023-06-13:
Hi [DarkVader135](https://github.com/DarkVader135), thanks for your feedback.
To verify the xmodel file is usable, you can run a vitis ai test program to test it. It's included in the vitit-ai-library
(usually in /usr/share/vitis_ai_library/test) and the input could be a simple jpeg file. 
About the potential software issue, because it's not the newest version of image and application, the local environment shall be set up in the laboratory using the same configuration as yours to investigate this bug.

hetianz, 2023-06-14:
The sample/test source code could be found [here](https://github.com/Xilinx/Vitis-AI/tree/2.0/demo/Vitis-AI-Library/samples). After compilation, you could use the test_jpeg_yolov3 in yolov3 folder to test the xmodel file.