Xilinx/Vitis-AI #1258: SSH into ZCU102

zijian98, 2023-06-16:
Host platform: Ubuntu 20.04
Target board: ZCU102
Using the latest SD card system image files (PetaLinux 2022.2)

Hi, does anyone knows how to SSH into the board? I have tried many times
```
ssh root@192.168.33.30
```
and entered the password: root
However, I keep getting root@192.168.33.30: Permission denied (publickey,password).

Pinging this address was successful
[image]

Any suggested solutions?

yangyecs, 2023-06-25:
Please refer to this page https://docs.xilinx.com/r/2022.2-English/ug1144-petalinux-tools-reference-guide/Login-Changes

zijian98, 2023-06-26:
> Please refer to this page https://docs.xilinx.com/r/2022.2-English/ug1144-petalinux-tools-reference-guide/Login-Changes

I am getting petalinux-config command not found. 

The OS image was downloaded from https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Flashing-the-OS-Image-to-the-SD-Card and and I can access the USB-UART port directly using minicom but not through SSH connection.

closed 2023-06-30 by zijian98