Xilinx/Vitis-AI #1259: OOM Cuda error when using  "include_fast_ft" in tf2 model quantization

JaviMota, 2023-06-16:
Hi

Im quantizing a very basic model in tf2 and the output model has very low accuracy (32%) in comparison with the float one (85%). According to the documentation, with tf2 we can use the flag "include_fast_ft" in combination with "fast_ft_epochs" to try to improve the performance in those cases. Im executing your "dogs_vs_cats" tutorial with some minor changes in the customcnn (less layers). Without the use of these flags Im able to quantize the model with a batch size of 32, but if I allow them I get the following error: 

OOM when allocating tensor with shape[2,240,240,32] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc [[{{node Identity_1-0-0-TransposeNCHWToNHWC-LayoutOptimizer}}]]

As you can see even with a batch size of 2 Im getting this out of memory error, using a RTX 2080 TI.

Do you know how can I solve this issue?

Thank you

gengxinwu, 2023-06-20:
Hi @JaviMota,
Depending on the GPU device, you can try to reduce the size of the calib_dataset. For example, if you used 1000 samples as the calibration dataset, please try changing it to 100, or even less.

zhenzhen-AMD, 2023-09-20:
Closing this for now. Please re-open if there are any other concerns. Thanks.

closed 2023-09-20 by zhenzhen-AMD