Xilinx/Vitis-AI #1260: Please add a (cross-) compile guide for custom Ultrascale+ MPSoC with

RapidEdgeAI, 2023-06-21:
Hi,

We have a custom Ultrascale+ MPSoC device with a Debian derived OS. Unfortunately, the VAI 3.0 User Guide and older versions of it only contains a build guide for Petalinux ([link](https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Setting-Up-the-Custom-Board)). We have built XRT (Xilinx Runtime), ZOCL-Driver, VART (Vitis AI Runtime) and the Vitis-AI-Library on the device itself. But finding out the dependencies and how to do it was a little bit challenging.
Is somewhere maybe a custom build guide, which includes the installation of the necessary dependencies, compilation of XRT and ZOCL and compilation of the VART und Vitis-AI-Library? Or could you create a custom build guide? This would be extremely helpful.
AFAIK there is an Ubuntu apt-package, developed by Canonical, which installs Vitis-AI ([link](https://packages.ubuntu.com/lunar/vitis-ai-library)). So, there might be already somewhere a build guide.

Thank you very much for your help!

Mickeyyyang, 2023-06-22:
Very good question, looking forward to an official reply from Xilinx...

yangyecs, 2023-06-27:
Unfortunately, Xilinx currently maintains the Petalinux tool exclusively. Regarding the VitisAI framework, it does not provide support for any other systems. However, Xilinx has a project called PYNQ, which is based on the Ubuntu system and can support the specified version of VitisAI and DPU. Below are some links to the project that you can refer to:
https://github.com/Xilinx/DPU-PYNQ
https://github.com/Xilinx/Kria-PYNQ/blob/main/README.md

RapidEdgeAI, 2023-06-27:
Ok, what a pity. But thank you very much for your response. My request has been clarified.

quentonh, 2023-07-20:
@RapidEdgeAI have you also been discussing this with our ML Specialist in Europe, Sophie?  I just want to understand if this is the same request that I am aware of.

RapidEdgeAI, 2023-07-20:
Hi @quentonh. Yes, but not directly. In February we asked the Xilinx support if there are prebuilt Debian packages for Vitis-AI because Xilinx offers the certified Ubuntu for the Kria boards. But the answer was no because the Ubuntu image that includes those packages is distributed by Canonical. Because of this it would be super helpful if a build guide, including dependencies, cross compilation, etc. for custom Linux systems would be included in a future Vitis AI release.
Petalinux works very well for us with our prototype. But unfortunately, we cannot use Petalinux in the product because our whole product ecosystem is based on our own Linux derivate due to maintainability reasons.

RapidEdgeAI, 2023-07-20:
I'm sorry. I accidentally skipped "Sophie". No, we didn't talk to Sophie. We talked to another Xilinx employee via the Xilinx support page.

tyler274, 2023-08-23:
This is also an issue for our usecase. Having detailed guides for bringing up the runtime environment is crucial.