Xilinx/Vitis-AI #1264: Error start docker on KV260

oriolorra, 2023-06-28:
Hello,

I have a KV260, and I want to try Vitis-AI.

When I try to run docker image like this: 

´´´
sudo ./docker_run.sh xilinx/vitis-ai-cpu
´´´

And I am getting this error:
[image]

Any suggestion/solution?

Thanks

qianglin-xlnx, 2023-07-04:
Hi @oriolorra 
The vitis-ai-cpu docker we released can only be run on X86 host not on the edge target.

aidanrhind, 2023-07-06:
Hi @oriolorra,
Please make sure you are including the framework you wish to use in the version of the docker image you are pulling.  
Ex:
1. docker pull xilinx/vitis-ai-pytorch-cpu:latest
2. ./docker_run.sh xilinx/vitis-ai-pytorch-cpu:latest
(Replace pytorch with tf or tf2 if you are using tensorflow)

Additionally, make sure that your user is in the docker group so you can run docker commands without sudo. 

https://docs.docker.com/engine/install/linux-postinstall/

oriolorra, 2023-07-07:
HI @qianglin-xlnx @aidanrhind,

Thanks, I was able to run the docker. I misunderstood where to run the docker images. I was doing it in the KV260 (Target), and I must run it into a x86 PC (Host)

closed 2023-07-30 by qianglin-xlnx