Xilinx/Vitis-AI #1274: YOLOv5 prototxt file

zijian98, 2023-06-30:
Hello, I have trained a yolov5n model with my own dataset and successfully compiled for ZCU102 with vitis-ai-pytorch. However, I noticed that it needs a prototxt file for my model. May I know how to generate this file? It returned segmentation fault when running with ./test_video_yolov5

Vitis-Ai 3.0

Thank You.

lishixlnx, 2023-07-04:
yolov5 model has its own prototxt file. 
You should copy from one of the yolov5 models (such as /usr/share/vitis_ai_library/models/yolov5m_pt/yolov5m_pt.prototxt in your board)
and make modification due to your own parameter used.
for example, you should change below values as show in my comments below.

`  kernel {
     mean: 0.0   # change your own mean value here.
     mean: 0.0
     mean: 0.0
     scale: 0.00392156   # change your own scale value here.
     scale: 0.00392156
     scale: 0.00392156
  }
  model_type : YOLOv3
  yolo_v3_param {
    num_classes: 80   # change your own classes number 
    anchorCnt: 3     
    layer_name: "271_fix"   # your model's output layer name , that must be different than this one.
    layer_name: "270_fix"
    layer_name: "269_fix"
    conf_threshold: 0.5
    nms_threshold: 0.65
    biases: 10
    biases: 13
    biases: 16
    biases: 30
    biases: 33
    biases: 23
    biases: 30
    biases: 61
    biases: 62
    biases: 45
    biases: 59
    biases: 119
    biases: 116
    biases: 90
    biases: 156
    biases: 198
    biases: 373
    biases: 326
    test_mAP: true
    type: YOLOV5
  }
  is_tf: true
`

zijian98, 2023-07-04:
Thank You. The model was able to deploy successfully.

closed 2023-07-04 by zijian98

nhanhust, 2023-07-06:
> Hello, I have trained a yolov5n model with my own dataset and successfully compiled for ZCU102 with vitis-ai-pytorch. However, I noticed that it needs a prototxt file for my model. May I know how to [...]

Hi zijian98. 

Can you guide me how to quantize and compile the above yolov5 model.  Can you send the files needed to perform the above steps?
I am a student and am doing an assignment.

Thank you.

zijian98, 2023-07-07:
> > Hello, I have trained a yolov5n model with my own dataset and successfully compiled for ZCU102 with vitis-ai-pytorch. However, I noticed that it needs a prototxt file for my model. May I know how to generate this file? It returned segmentation fault when running with ./test_video_yolov5
> > Vitis-Ai 3.0
> > Thank You.
> 
> Hi @zijian98 .
> 
> Can you guide me how to quantize and compile the above yolov5 model. Can you send the files needed to perform the above steps (you can send the file via gmail: [sung13335@gmail.com](mailto:sung13335@gmail.com)) ? I am a student and am doing an assignment.
> 
> Thank you.

Hello, you may refer to this link regarding the quantization and compilation of YOLOv5n model: https://github.com/Xilinx/Vitis-AI/issues/1252 

You need to first modify the unsupported activation functions used by YOLOv5 and retrain your model. This can be done by replacing all SiLU function in models/common.py and models/experimental.py.

After that, launch the pytorch gpu docker and change the forward function in models/yolo.py. this will allow you to quantize and compile the model successfully and the xmodel will be generated.

Regarding the prototxt file, you may copy and paste the prototxt file configuration above and modify the classes needed for your own model. Run the command ```xir subgraph yolov5n.xmodel | grep DPU``` and you will be able to see the layer_name and its value as shown below:

```
subgraph_DetectMultiBackend__DetectMultiBackend_DetectionModel_model__C3_model__C3_13__Conv_cv1__Conv2d_conv__input_133 [device=DPU,fingerprint=0x603000b56011861,DPU=DPUCVDX8G_ISA3_C32B6,I=[xir_tensor{DetectMultiBackend__input_0_fix:(1,640,640,3), fixpos=4 # of elements= 1228800}],O=
[xir_tensor{DetectMultiBackend__DetectMultiBackend_DetectionModel_model__Detect_model__Detect_24__Conv2d_m__ModuleList_0__9285_fix:(1,80,80,21), fixpos=3 # of elements= 134400},
xir_tensor{DetectMultiBackend__DetectMultiBackend_DetectionModel_model__Detect_model__Detect_24__Conv2d_m__ModuleList_1__9304_fix:(1,40,40,21), fixpos=3 # of elements= 33600},
xir_tensor{DetectMultiBackend__DetectMultiBackend_DetectionModel_model__Detect_model__Detect_24__Conv2d_m__ModuleList_2__9323_fix:(1,20,20,21), fixpos=2 # of elements= 8400}]]
```
where 9285_fix, 9304_fix, and 9323_fix are your layer_name.

With these, you should be able to deploy successfully.

nhanhust, 2023-07-07:
> > > Hello, I have trained a yolov5n model with my own dataset and successfully compiled for ZCU102 with vitis-ai-pytorch. However, I noticed that it needs a prototxt file for my model. May I know how to generate this file? It returned segmentation fault when running with ./test_video_yolov5
> > > Vitis-Ai 3.0
> > > Thank You.
> > 
> > 
> > Hi @zijian98 .
> > Can you guide me how to quantize and compile the above yolov5 model. Can you send the files needed to perform the above steps (you can send the file via gmail: [sung13335@gmail.com](mailto:sung13335@gmail.com)) ? I am a student and am doing an assignment.
> > Thank you.
> 
> Hello @nhanhust , you may refer to this link regarding the quantization and compilation of YOLOv5n model: #1252
> 
> You need to first modify the unsupported activation functions used by YOLOv5 and retrain your model. This can be done by replacing all SiLU function in models/common.py and models/experimental.py.
> 
> After that, launch the pytorch gpu docker and change the forward function in models/yolo.py. this will allow you to quantize and compile the model successfully and the xmodel will be generated.
> 
> Regarding the prototxt file, you may copy and paste the prototxt file configuration above and modify the classes needed for your own model. Run the command `xir subgraph yolov5n.xmodel | grep DPU` and you will be able to see the layer_name and its value as shown below:
> 
> ```
> subgraph_DetectMultiBackend__DetectMultiBackend_DetectionModel_model__C3_model__C3_13__Conv_cv1__Conv2d_conv__input_133 [device=DPU,fingerprint=0x603000b56011861,DPU=DPUCVDX8G_ISA3_C32B6,I=[xir_tensor{DetectMultiBackend__input_0_fix:(1,640,640,3), fixpos=4 # of elements= 1228800}],O=
> [xir_tensor{DetectMultiBackend__DetectMultiBackend_DetectionModel_model__Detect_model__Detect_24__Conv2d_m__ModuleList_0__9285_fix:(1,80,80,21), fixpos=3 # of elements= 134400},
> xir_tensor{DetectMultiBackend__DetectMultiBackend_DetectionModel_model__Detect_model__Detect_24__Conv2d_m__ModuleList_1__9304_fix:(1,40,40,21), fixpos=3 # of elements= 33600},
> xir_tensor{DetectMultiBackend__DetectMultiBackend_DetectionModel_model__Detect_model__Detect_24__Conv2d_m__ModuleList_2__9323_fix:(1,20,20,21), fixpos=2 # of elements= 8400}]]
> ```
> 
> where 9285_fix, 9304_fix, and 9323_fix are your layer_name.
> 
> With these, you should be able to deploy successfully.

Thanks for answer @zijian98!

I followed the steps at: https://github.com/Xilinx/Vitis-AI/issues/1252#issuecomment-1608431434 and I trained with gg colab and got the **best.pt** file (in runs folder) and downloaded it.
But I don't know how to quantize?
I saw this quantize.py file (https://github.com/Xilinx/Vitis-AI/issues/1252#issuecomment-1612238795). I also edited the forward(self, x) function (as per https://github.com/Xilinx/Vitis-AI/issues/1252#issuecomment-1618812057) and put it in the yolov5 folder.
In the end I have a yolov5 folder in vitis-ai folder as follows:

/home/nhan/Vitis-AI-3.0/yolov5/yolov5
├── benchmarks.py
├── CITATION.cff
├── classify
│   ├── predict.py
│   ├── train.py
│   ├── tutorial.ipynb
│   └── val.py
├── CONTRIBUTING.md
├── data
│   ├── Argoverse.yaml
│   ├── coco128-seg.yaml
│   ├── coco128.yaml
│   ├── coco.yaml
│   ├── GlobalWheat2020.yaml
│   ├── hyps
│   ├── ImageNet.yaml
│   ├── images
│   ├── Objects365.yaml
│   ├── scripts
│   ├── SKU-110K.yaml
│   ├── VisDrone.yaml
│   ├── VOC.yaml
│   ├── xView.yaml
**_│   └── YOLODataset_**
│       ├── images
│       │   ├── test : .png, .....
│       │   ├── train
│       │   └── val
│       └── labels
│           ├── test: .txt, ...
│           ├── train
│           └── val
├── detect.py
├── export.py
[... 87 lines cut ...]
│   │   ├── __init__.py
│   │   ├── loss.py
│   │   ├── metrics.py
│   │   └── plots.py
│   ├── torch_utils.py
│   └── triton.py
└── val.py

Then I run:
./docker_run.sh xilinx/vitis-ai-cpu:latest
(i don't have gpu on pc)

pip install -r requirements.txt
[image]

conda activate vitis-ai-pytorch 

Then I run
python quantize.py --build_dir runs/train/exp --quant_mode calib --weights runs/train/exp/best.pt --dataset data/YOLODataset
I don't know what to do when there is an error here:
[image]

I don't know if my steps are correct or not? Or do I make a mistake? 

Looking forward to your help. Thank you!

IkrameBeggar, 2023-09-21:
> > > > Hello, I have trained a yolov5n model with my own dataset and successfully compiled for ZCU102 with vitis-ai-pytorch. However, I noticed that it needs a prototxt file for my model. May I know how to generate this file? It returned segmentation fault when running with ./test_video_yolov5
> > > > Vitis-Ai 3.0
> > > > Thank You.
> > > 
> > > 
> > > Hi @zijian98 .
> > > Can you guide me how to quantize and compile the above yolov5 model. Can you send the files needed to perform the above steps (you can send the file via gmail: [sung13335@gmail.com](mailto:sung13335@gmail.com)) ? I am a student and am doing an assignment.
> > > Thank you.
> > 
> > 
> > Hello @nhanhust , you may refer to this link regarding the quantization and compilation of YOLOv5n model: #1252
> > You need to first modify the unsupported activation functions used by YOLOv5 and retrain your model. This can be done by replacing all SiLU function in models/common.py and models/experimental.py.
> > After that, launch the pytorch gpu docker and change the forward function in models/yolo.py. this will allow you to quantize and compile the model successfully and the xmodel will be generated.
> > Regarding the prototxt file, you may copy and paste the prototxt file configuration above and modify the classes needed for your own model. Run the command `xir subgraph yolov5n.xmodel | grep DPU` and you will be able to see the layer_name and its value as shown below:
> > ```
> > subgraph_DetectMultiBackend__DetectMultiBackend_DetectionModel_model__C3_model__C3_13__Conv_cv1__Conv2d_conv__input_133 [device=DPU,fingerprint=0x603000b56011861,DPU=DPUCVDX8G_ISA3_C32B6,I=[xir_tensor{DetectMultiBackend__input_0_fix:(1,640,640,3), fixpos=4 # of elements= 1228800}],O=
> > [xir_tensor{DetectMultiBackend__DetectMultiBackend_DetectionModel_model__Detect_model__Detect_24__Conv2d_m__ModuleList_0__9285_fix:(1,80,80,21), fixpos=3 # of elements= 134400},
> > xir_tensor{DetectMultiBackend__DetectMultiBackend_DetectionModel_model__Detect_model__Detect_24__Conv2d_m__ModuleList_1__9304_fix:(1,40,40,21), fixpos=3 # of elements= 33600},
> > xir_tensor{DetectMultiBackend__DetectMultiBackend_DetectionModel_model__Detect_model__Detect_24__Conv2d_m__ModuleList_2__9323_fix:(1,20,20,21), fixpos=2 # of elements= 8400}]]
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > where 9285_fix, 9304_fix, and 9323_fix are your layer_name.
> > With these, you should be able to deploy successfully.
> 
> Thanks for answer @zijian98!
> 
> I followed the steps at: [#1252 (comment)](https://github.com/Xilinx/Vitis-AI/issues/1252#issuecomment-1608431434) and I trained with gg colab and got the **best.pt** file (in runs folder) and downloaded it. But I don't know how to quantize? I saw this quantize.py file ([#1252 (comment)](https://github.com/Xilinx/Vitis-AI/issues/1252#issuecomment-1612238795)). I also edited the forward(self, x) function (as per [#1252 (comment)](https://github.com/Xilinx/Vitis-AI/issues/1252#issuecomment-1618812057)) and put it in the yolov5 folder. In the end I have a yolov5 folder in vitis-ai folder as follows:
> 
> /home/nhan/Vitis-AI-3.0/yolov5/yolov5 ├── CONTRIBU **_│   └── YOLODataset_** │   ├── images │   │   ├── test : .png, ..... │   │   ├── train │   │   └── val │   └── labels │   ├── test: .txt, ... │   ├── train │   └── val ├── detect.py ├── export.py ├── hubconf.py ├── LICENSE ├── models **_│   ├── common.py_** **_│   ├── experimental.py_** **_├── quantize.py_** ├── README.md ├── README.zh-CN.md **_├── requirements.txt_** ├── runs │   └── train │   └── exp **_│   └── best.pt_**
> 
> Then I run: ./docker_run.sh xilinx/vitis-ai-cpu:latest (i don't have gpu on pc)
> 
> pip install -r requirements.txt [image]
> 
> conda activate vitis-ai-pytorch
> 
> Then I run python quantize.py --build_dir runs/train/exp --quant_mode calib --weights runs/train/exp/best.pt --dataset data/YOLODataset I don't know what to do when there is an error here: [image]
> 
> I don't know if my steps are correct or not? Or do I make a mistake? I use Xilinx ZCU104 Evaluation Board.
> 
> Looking forward to your help. Thank you!
> 
> Update: I have quantized the yolov5 . model [image]
> 
> Every time quantization I need to run: pip install -r requirements.txt in docker cpu I take very long time to wait for execution. Is there a faster way? And can yolov6 do the same as the tutorial?

Hello, I have a question, did you notice a drop on the performance of you yolov5 model after quantization and testing it on the board?

IkrameBeggar, 2023-11-09:
> yolov5 model has its own prototxt file. You should copy from one of the yolov5 models (such as /usr/share/vitis_ai_library/models/yolov5m_pt/yolov5m_pt.prototxt in your board) and make modification [...]

Hello, can you please tell me how to modify scale and mean values? Based on what?

quentonh, 2023-11-09:
Here is an excerpt that I wrote some time back.  I did start working on integrating this into the Github.IO documentation and will finish this in a future docs release.

~~~~~~~~~~~~~~~~~~~~~~~~~~

Input normalization serves to “center” the input data such that the data associated with each input sample has a similar range. In the context of data science there are potentially many techniques that can be used to “standardize the statistical properties of the input”, but the most common method used in pixel-based processing for CNNs is mean subtraction followed by division by the standard deviation.  What does this mean more precisely?  What it means is that we calculate the per-color channel mean and standard deviation across the entire training dataset.  For example, for ImageNet, these values, calculated across the entire dataset are, where the ordering of the elements in each vector is [R,G,B]:

        mean=[123.675, 116.28, 103.53] 
        std=[58.395, 57.12, 57.375]
        
So, for each image that is fed to the network for both training and inference we subtract the mean value and divide by the standard deviation per channel:

  image=(image-mean)/std

Pre-processing of the pixels at inference time must precisely emulate the preprocessing technique that was used for training.  When using the Vitis AI Library for inference, the pre-processing implementation is communicated via a .prototxt file that lists the mean and scale (where scale=1/std) per channel.  ImageNet example: 

model {
   name : "resnet18_pt"
   kernel {
         name: "resnet18_pt_0"
         mean: 103.53
         mean: 116.28
         mean: 123.675
         scale: 0.017429
         scale: 0.017507
         scale: 0.01712475
   }
   model_type : CLASSIFICATION
   classification_param {
          top_k : 5
          test_accuracy : false
          preprocess_type : VGG_PREPROCESS
   }
}

quentonh, 2023-11-09:
Here is a better version....
[image]

IkrameBeggar, 2023-11-12:
> Here is a better version.... [image]

Thank you very much

TurcanVeli, 2025-03-07:
> Thanks for answer [@zijian98](https://github.com/zijian98)!
>
> I followed the steps at: [#1252 (comment)](https://github.com/Xilinx/Vitis-AI/issues/1252#issuecomment-1608431434) and I trained with gg [...]

How did you edit foward and use it when applying quantize?

218w1d7706, 2025-05-22:
@zijian98 how we can evaluate the quantizeed  model. if exported the quantized model to both onnx and xmodel and stuck and how we can evaluate the quantized model. any suggestions are much appreciated.

zijian98, 2025-05-23:
> [@zijian98](https://github.com/zijian98) how we can evaluate the quantizeed model. if exported the quantized model to both onnx and xmodel and stuck and how we can evaluate the quantized model. any suggestions are much appreciated.

Hi, its been a while since I last used Vitis-AI, however, you may refer to this repository for their integration of vitis-ai and yolov5. 
https://github.com/Xilinx/Vitis-AI-Copyleft-Model-Zoo/tree/6882ff7cadfad7c98b6c2ed2ab17a9fb4462426e/yolov5_nano

218w1d7706, 2025-05-24:
@zijian98 thank you. i will follow the reference you provided

quang29122k3, 2025-06-17:
I have trained YOLOv8n with a dataset for UAV detection. I then quantized the model to INT8, specifically converting both the weights and intermediate activations from float32 to int8.
After static quantization, my model is in ONNX format and has a size of approximately 3MB.
I have evaluated the quantized model on a sample image, and the confidence scores are approximately the same as those from the original model.
However, I would like to ask: if I intend to deploy this model on an edge device such as the PYNQ-Z2, is my current quantized model sufficient?
Initially, I did not modify the activation functions. If this is a problem, should the correct workflow be to first replace SiLU with ReLU (e.g., ReLU6) before performing static quantization?
Thank you very much. I look forward to your guidance.

218w1d7706, 2025-06-18:
i dont know much but it will tell you what i know. so 
_should the correct workflow be to first replace SiLU with ReLU_
yes. it i what i did when for yolov5 as silu is not supported on DPU.
_I intend to deploy this model on an edge device such as the PYNQ-Z2_
never has the oppurtunity to work on pynq but as per my understanding if you want to deploy on FPGA the quantizatin has to be performed using vitis ai quantization then you have to same the model in .xmodel format then  use the vitis ai compiler to compile the model. then you can deploy it.

i wanted to tell you something also, when i was working with yolov5 there are some changes in code [refer

TurcanVeli, 2025-06-18:
for all model prototxt is same? for example yolov6 anchor free model so in head anchor is 1 and there is no constant anchors. So how to add bias in prototxt? How to find them?

smtalds-knt, 2025-06-18:
@TurcanVeli  Hi,

Vitis AI 3.5 officially supports only the [YOLOv6-m model](https://docs.amd.com/r/en-US/ug1354-xilinx-ai-sdk/YOLOv6-Detection).

During my research, I found some helpful files that you can use. They include example `prototxt` configurations:
🔗 https://github.com/Xilinx/Vitis-AI/blob/3.0/model_zoo/model-list/pt_yolov6m_coco_640_640_82.4G_3.0/model.yaml

best regards,

TurcanVeli, 2025-06-18:
Yeah you were right. Thanks it is more much cleaner now.