Xilinx/Vitis-AI #1275: Non accessible Vitis-AI-Copyleft-Model-Zoo repository

iraadit, 2023-06-30:
In the Vitis-AI 3.5 doc, at https://xilinx.github.io/Vitis-AI/3.5/html/docs/workflow-model-zoo.html, it is written:

```
Vitis AI Copyleft Model Zoo[¶](https://xilinx.github.io/Vitis-AI/3.5/html/docs/workflow-model-zoo.html#vitis-ai-copyleft-model-zoo)

Many open-source models are released under reciprocal license terms which are not compatible with Apache 2.0. In order to faciliate the support of such models, and clearly distinguish the source license for each, we have created a separate Model Zoo repository. Users will find the training code for these models (for example, YOLOv7) in the [Vitis AI Copyleft Model Zoo](https://github.com/Xilinx/Vitis-AI-Copyleft-Model-Zoo). All other models are found in the primary Vitis AI repository.
```
However, [Vitis-AI-Copyleft-Model-Zoo](https://github.com/Xilinx/Vitis-AI-Copyleft-Model-Zoo) is not accessible. Maybe you forgot to make it public?

Than you

quentonh, 2023-06-30:
@iraadit Thanks for mentioning this!  Yes, it has been a bit delayed.  We just received final legal approval in the past 24 hours.  You can expect it to be online very shortly.

iraadit, 2023-07-03:
Thank you for your prompt answer!

I'll be looking forward to the public release of this repo.

Closing the issue.

quentonh, 2023-07-10:
@iraadit I can confirm that the repo is now public.  We will be adding some additional guides to this repo for some AGPL models in the near future, but for now this is what we have released:  https://github.com/Xilinx/Vitis-AI-Copyleft-Model-Zoo

iraadit, 2023-07-11:
Thank you for the code and the follow-up !