Xilinx/Vitis-AI #1277: cannot import name 'decent_q' Vitis-AI 3.0 tf1

JaviMota, 2023-07-04:
Hi

I want to evaluate the quantized model using the script of the tf_ssdmobilenetv1 (taken from the example in vitis ai library) but Im getting this error:

**cannot import name 'decent_q'**

Is the decent_q library necessary to evaluate a quantized model? In such case, how can I get it?

Im using the docker image xilinx/vitis-ai-tensorflow-gpu from Vitis-AI 3.0

Thank you.

gengxinwu, 2023-07-04:
From Vitis-AI 3.0, tf1 has changed from a contributor to a Tensorflow plugin. Please try changing "from tensorflow.contrib import decent_q" to "import vai_q_tensorflow as decent_q".

zhenzhen-AMD, 2023-08-23:
Closing this for now. Please re-open if there are any other concerns. Thanks.

closed 2023-08-23 by renjiangyong

amonni5, 2024-03-21:
simil issue

 import tensorflow.contrib.decent_q
ModuleNotFoundError: No module named 'tensorflow.contrib.decent_q'

in evaluating step of 02-mnist-tutorial

vitis-ai 3.5 gpu