Xilinx/Vitis-AI #1278: VART compilation error

Rellek72, 2023-07-04:
Following the [Quick Start for Embedded](https://github.com/Xilinx/Vitis-AI/blob/v3.5/src/vai_runtime/quick_start_for_embedded.md) guide, I error out at this step:

```
cd <vitis-ai-install-path>/src/vai_runtime/vart
./cmake.sh --clean
```
With the following error:
```
Cross-platform building...
Found target sysroot /home/user/petalinux_sdk_2023.1/sysroots/cortexa72-cortexa53-xilinx-linux
cd /home/user/build/build.linux.2023.1.aarch64.Debug/vart
cmake -DBUILD_TEST=ON -DENABLE_CPU_RUNNER=ON -DENABLE_SIM_RUNNER=ON -DBUILD_SHARED_LIBS=ON -DCMAKE_BUILD_TYPE=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DBUILD_PYTHON=OFF -DINSTALL_HOME=OFF -DCMAKE_TOOLCHAIN_FILE=/home/user/petalinux_sdk_2023.1/sysroots/x86_64-petalinux-linux/usr/share/cmake/OEToolchainConfig.cmake -DCMAKE_PREFIX_PATH=/install/Debug -DCMAKE_INSTALL_PREFIX=/home/user/petalinux_sdk_2023.1/sysroots/cortexa72-cortexa53-xilinx-linux/install/Debug /home/user/dev/Vitis-AI/src/vai_runtime/vart
-- building for edge ON
-- Found importable target unilog::unilog: /home/user/petalinux_sdk_2023.1/sysroots/cortexa72-cortexa53-xilinx-linux/install/Debug/lib/libunilog.so.3.5.0
-- Find glog in public mode
-- Found Boost: /home/user/petalinux_sdk_2023.1/sysroots/cortexa72-cortexa53-xilinx-linux/usr/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: system filesystem 
-- Found importable target xir::xir: /home/user/petalinux_sdk_2023.1/sysroots/cortexa72-cortexa53-xilinx-linux/install/Debug/lib/libxir.so.3.5.0
-- Found importable target target-factory::target-factory: /home/user/petalinux_sdk_2023.1/sysroots/cortexa72-cortexa53-xilinx-linux/install/Debug/lib/libtarget-factory.so.3.5.0
-- Could NOT find XRT (missing: XRT_DIR)
-- XRT FOUND: XRT_LIBRARYIES= XRT_INCLUDE_DIRS=/home/user/petalinux_sdk_2023.1/sysroots/cortexa72-cortexa53-xilinx-linux/usr/include/xrt
-- Found Boost: /home/user/petalinux_sdk_2023.1/sysroots/cortexa72-cortexa53-xilinx-linux/usr/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: filesystem 
+ cat /home/user/dev/Vitis-AI/src/vai_runtime/vart/dpu-runner/test/word_list.txt
+ sed 's/\(.*\)/"\1",/g'
CMake Error at CMakeLists.txt:107 (add_subdirectory):
  add_subdirectory given source "sim-runner" which is not an existing
  directory.

CMake Error at CMakeLists.txt:107 (add_subdirectory):
  add_subdirectory given source "cpu-runner" which is not an existing
  directory.

-- Configuring incomplete, errors occurred!
```

I've been following the guide to the letter in terms of install locations and such, any advice would be greatly appreciated.

qianglin-xlnx, 2023-07-05:
@Rellek72  Thank you for your feedback.
The bug is fixed now. Please git pull the latest code and try it again.
You can also refer to : https://github.com/Xilinx/Vitis-AI/commit/904abc2f724d1483964714e68b8e7db851fdb19f

Rellek72, 2023-07-05:
This resolved my issue, thank you for the quick update!

closed 2023-07-05 by Rellek72