Xilinx/Vitis-AI #1279: Profiler (vaitrace) - ERROR:root:This xmodel does not support fine grained profiling

flyinged, 2023-07-05:
I am trying the profiler on the resnet50 example running on the VCK190 board.

The board is loaded with the DPU reference design (https://github.com/Xilinx/Vitis-AI/tree/3.0/dpu). Vitis-AI runtime 3.0 is installed on the target.

The profiler works normally when I run `vaitrace ./resnet50 /usr/share/vitis_ai_library/models/resnet50/resnet50.xmodel`

but if I run the command with the `--fine_grained` option, then I get the following error:

`ERROR:root:This xmodel does not support fine grained profiling`

How does one enable the fine-grained profiling on an xmodel?

li900309, 2023-07-21:
hi @flyinged ，
Unfortunately, this feature has been removed in 3.5 due to xcompiler optimizations, and the results produced by the release version of the xcompiler will no longer contain fine-grained network information and therefore cannot be profiling

flyinged, 2023-07-21:
hi @li900309 

Thanks for your answer. Is there an alternative to that? How is one supposed to debug a design, in case a network has a performance much lower than expected? Coarse profiling just does not provide enough information for a proper analysis.

kkzqp, 2023-10-04:
Did you solved it?

flyinged, 2023-10-04:
No. Our client has decided that they don't want to use Vitis-AI anymore for
their project. Thus I've stopped with the attempts too.

shaibal13, 2023-12-25:
Can anyone solve this issue?

TiantianZhang, 2024-03-09:
Is anyone solved this problem?

The following reference is not work on KV260.

https://github.com/Xilinx/Vitis-AI/tree/v3.5/examples/vai_profiler
https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Vitis-AI-Profiler?tocId=Luv7e02rhpEpdYpzLa0cpg