Xilinx/Vitis-AI #1280: Linking fails on base VCK190 DPUCVDX8G reference design

flyinged, 2023-07-06:
I am trying to generate an image for the VCK190 DPUCVDX8G reference design by following the instructions in https://github.com/Xilinx/Vitis-AI/blob/3.0/dpu/ref_design_docs/README_DPUCVDX8G.md

No changes were made to the project.

The "link" step fails with the following message:

```
[17:52:27] Starting bitstream generation..
[18:07:52] Run vpl: Step impl: Failed
[18:07:54] Run vpl: FINISHED. Run Status: impl ERROR
ERROR: [VPL 60-773] In '/home/amalatesta/Vitis-AI/DPUCVDX8G_VAI_v3.0/vitis_prj/hw/binary_container_1/link/vivado/vpl/vivado.log', caught Tcl error:  problem implementing dynamic region, impl_1: write_device_image ERROR, please look at the run log file '/home/amalatesta/Vitis-AI/DPUCVDX8G_VAI_v3.0/vitis_prj/hw/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, impl_1: write_device_image ERROR, please look at the run log file '/home/amalatesta/Vitis-AI/DPUCVDX8G_VAI_v3.0/vitis_prj/hw/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [18:07:55] Run run_link: Step vpl: Failed
Time (s): cpu = 00:02:21 ; elapsed = 04:58:56 . Memory (MB): peak = 458.805 ; gain = 0.000 ; free physical = 20467 ; free virtual = 20876
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [Makefile:199: hw/binary_container_1.xsa] Error 1
```

The end of the implementation log (`Vitis-AI/DPUCVDX8G_VAI_v3.0/vitis_prj/hw/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log`) reads:

```
[17:52:27] Starting bitstream generation..
[18:07:52] Run vpl: Step impl: Failed
[18:07:54] Run vpl: FINISHED. Run Status: impl ERROR
ERROR: [VPL 60-773] In '/home/amalatesta/Vitis-AI/DPUCVDX8G_VAI_v3.0/vitis_prj/hw/binary_container_1/link/vivado/vpl/vivado.log', caught Tcl error:  problem implementing dynamic region, impl_1: write_device_image ERROR, please look at the run log file '/home/amalatesta/Vitis-AI/DPUCVDX8G_VAI_v3.0/vitis_prj/hw/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
WARNING: [VPL 60-732] Link warning: No monitor points found for BD automation.
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, impl_1: write_device_image ERROR, please look at the run log file '/home/amalatesta/Vitis-AI/DPUCVDX8G_VAI_v3.0/vitis_prj/hw/binary_container_1/link/vivado/vpl/prj/prj.runs/impl_1/runme.log' for more information
ERROR: [VPL 60-1328] Vpl run 'vpl' failed
ERROR: [VPL 60-806] Failed to finish platform linker
INFO: [v++ 60-1442] [18:07:55] Run run_link: Step vpl: Failed
Time (s): cpu = 00:02:21 ; elapsed = 04:58:56 . Memory (MB): peak = 458.805 ; gain = 0.000 ; free physical = 20467 ; free virtual = 20876
ERROR: [v++ 60-661] v++ link run 'run_link' failed
ERROR: [v++ 60-626] Kernel link failed to complete
ERROR: [v++ 60-703] Failed to finish linking
INFO: [v++ 60-1653] Closing dispatch client.
make: *** [Makefile:199: hw/binary_container_1.xsa] Error 1
```

The full log is attached: [runme.log](https://github.com/Xilinx/Vitis-AI/files/11964845/runme.log)
The compilation run on Ubuntu 22_04 under VirtualBox (24 GB of RAM available).

Any suggestions on how to solve this?

yangyecs, 2023-07-21:
It seems that it's a free memory issue. If you don’t create swap memory, 24GB memory might be not proper to build a large FPGA design. Vivado needs free pages to manage some file structures. Can you make sure there is a swap partition available on your machine and then see how it goes.
 
The following are some useful links to help you allocate properly system memory and create a swap file.
 
https://www.xilinx.com/products/design-tools/vivado/vivado-ml.html#memory
https://linuxize.com/post/create-a-linux-swap-file/

flyinged, 2023-07-21:
Thanks for the info. Increasing the amount of RAM actually solved the problem. Many thanks.

closed 2023-07-21 by flyinged