Xilinx/Vitis-AI #1281: Quantize Yolov5 for instance segmentation

oriolorra, 2023-07-07:
Hi everyone,

I am trying to quantize a yolov5 for instance segmentation ([link](https://github.com/ultralytics/yolov5/releases/download/v7.0/yolov5l-seg.pt)).

But, now, I get an error during the fast finetuning:
```
[VAIQ_WARN][QUANTIZER_TORCH_TENSOR_TYPE_NOT_QUANTIZABLE]: The tensor type of DetectMultiBackend::DetectMultiBackend/SegmentationModel[model]/Segment[model]/Segment[24]/ret.483 is torch.int32. Only support float32/double/float16 quantization.
100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 149/149 [37:23<00:00, 15.06s/it]

[VAIQ_NOTE]: =>Fast finetuning module parameters for better quantization accuracy...
  0%|                                                                                                                                                       | 0/129 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "quantizer_test.py", line 119, in <module>
    quantizer.fast_finetune(evaluate, (quant_model, dataset))
  File "/opt/vitis_ai/conda/envs/vitis-ai-wego-torch/lib/python3.8/site-packages/pytorch_nndct/apis.py", line 123, in fast_finetune
    self.processor.finetune(run_fn, run_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-wego-torch/lib/python3.8/site-packages/pytorch_nndct/qproc/base.py", line 293, in finetune
    adaquant.finetune_v2(run_fn, run_args)
  File "/opt/vitis_ai/conda/envs/vitis-ai-wego-torch/lib/python3.8/site-packages/pytorch_nndct/qproc/adaquant.py", line 803, in finetune_v2
    net_loss = self.optimize_layer_v2(qnode, fmod, layer_act_pair, net_inputs, net_outputs, net_loss, device, need_cache)
  File "/opt/vitis_ai/conda/envs/vitis-ai-wego-torch/lib/python3.8/site-packages/pytorch_nndct/qproc/adaquant.py", line 947, in optimize_layer_v2
    loss.backward()
  File "/opt/vitis_ai/conda/envs/vitis-ai-wego-torch/lib/python3.8/site-packages/torch/_tensor.py", line 488, in backward
    torch.autograd.backward(
  File "/opt/vitis_ai/conda/envs/vitis-ai-wego-torch/lib/python3.8/site-packages/torch/autograd/__init__.py", line 197, in backward
    Variable._execution_engine.run_backward(  # Calls into the C++ engine to run the backward pass
RuntimeError: could not create a primitive descriptor iterator
```

Anyone has a clue, why am I getting this error? Can be quantize the yolov5l-seg?

Thank you very much,

LouisLepez23, 2023-08-08:
Hi, could you resolve the problem ?

oriolorra, 2023-10-09:
No, seems that VitisAI could not optimize/transform this kind of network