Xilinx/Vitis-AI #1283: No module named 'tf_nndct'

Rellek72, 2023-07-09:
Attempting to use the iterative pruning runner (such as in the [mnist_convnet example](https://github.com/Xilinx/Vitis-AI/blob/bf7cca32b0f45144531a77a9a2a6fa8e4bcb6c4b/examples/vai_optimizer/tensorflow2/mnist_convnet/mnist_convnet.py#L21C11-L21C11)) in the vitis-ai-tensorflow2-cpu docker container returns the error:

```
  File "/workspace/testbench.py", line 162, in prune
    from tf_nndct.optimization import IterativePruningRunner
ModuleNotFoundError: No module named 'tf_nndct'
```

quentonh, 2023-07-18:
@Rellek72 Please change this to:

_**from tf_nndct.pruning import IterativePruningRunner**_

You will then likely see errors related to:

**_INTERNAL: libdevice not found at ./libdevice.10.bc_**

I am posting a temporary workaround / patch here that you can test:  https://support.xilinx.com/s/question/0D54U0000795OLvSAM/temporary-patch-to-resolve-internal-libdevice-not-found-at-libdevice10bc?language=en_US

Please update us as to whether or not this helps you to solve this.

Rellek72, 2023-07-18:
@quentonh Thank you so much, this not only resolved this issue but also allowed me to use the GPU accelerated TF2 container (which wasn't working before, but I assumed it was a problem with my machine).

closed 2023-07-18 by Rellek72