Xilinx/Vitis-AI #1286: Is AI compiler part open source?

Canvas-Atomer, 2023-07-11:
I cannot find the source code of AI compiler in src folder. Is it not open source?

quentonh, 2023-07-18:
@Canvas-Atomer No, it is not currently open source.  We are considering opening it up in the future to enable users to build their own custom operators using the DPU ISA, but not final decision has been made, and we may or may not do this.  Tell us more about your requirement so that we can incorporate your feedback into this decision making process.

akuzminski, 2023-07-19:
This is also something that my team would want. Currently we must use vitis containers to do quantization and export of xmodels. This is sometimes tedious because we constantly install our python packages which are not present in container. If the compiler would be open source, we could just install it locally or construct our custom docker image with all the necessary tools for training, pruning, quantization and compilation.

quentonh, 2023-07-19:
@akuzminski @Canvas-Atomer setting aside the source-code discussion for a moment, for both of your use cases would installation via shell scripts or pip install packages be helpful, and if so, do you have a preference for one or the other?

akuzminski, 2023-07-20:
@quentonh if I had to choose, I'd say pip install but either is fine really