Xilinx/Vitis-AI #1287: Cannot modify conda environment in VAI 3.5 vitis-ai-pytorch-cpu container

danielstumpp, 2023-07-11:
I'm attempting to upgrade my workflow to VAI3.5 to leverage the new features, particularly the optimizer. I need to install python packages not included in the default conda environment, specifically `mlflow`. However, the environment requires root permissions to modify. I am running the container via singularity on an HPC system and do not have root privileges. This is the error I receive: 

```
Docker Image Version: ubuntu2004-3.5.0.306   (CPU) 
Vitis AI Git Hash: 6a9757a 
Build Date: 2023-06-26
WorkFlow: pytorch

[1]+  Exit 1                  sudo chown -R vitis-ai-user /opt/vitis_ai/conda
Apptainer> conda activate vitis-ai-pytorch
(vitis-ai-pytorch) Apptainer> conda install -c conda-forge mlflow
Collecting package metadata (current_repodata.json): failed

NotWritableError: The current user does not have write permissions to a required path.
  path: /opt/vitis_ai/conda/pkgs/urls.txt
  uid: 100230
  gid: 5000

If you feel that permissions on this path are set incorrectly, you can manually
change them by executing

  $ sudo chown 100230:5000 /opt/vitis_ai/conda/pkgs/urls.txt

In general, it's not advisable to use 'sudo conda'.
```

The VAI3.0 container does not have this problem. I was able to successfully build and run with singularity and modify the conda environments as needed. Is this intentional? Is there a workaround?

janifer112x, 2023-07-21:
Hi @danielstumpp , 
there is an error message when you launch the docker.

[1]+  Exit 1                  sudo chown -R vitis-ai-user /opt/vitis_ai/conda

I think that's the reason, could you please launch a new container and try again, I've tried from my side. it works.
or if you want to keep the current running container, you can use '   sudo chown -R vitis-ai-user /opt/vitis_ai/conda' command to change  the permission again

danielstumpp, 2023-07-26:
@janifer112x, yes that seems to be the root cause. I was able to work around it in this case by installing additional packages using the `--user` flag with `pip install`. 

The cause is because I'm running the container via singularity (I can't run docker on the HPC environment I work in). It may be beneficial to support cases like this in some way in the future.

closed 2023-07-26 by danielstumpp