Xilinx/Vitis-AI #1290: KV260 and Vitis-AI 3.5 compatibility

iraadit, 2023-07-14:
Is it possible to run Vitis-AI 3.5 (and so, yolov7 and yolov8) on the KV260 target?

In [board_setup folder](https://github.com/Xilinx/Vitis-AI/tree/master/board_setup), there is only v70 and vek280.

Will KV260 be made compatible later on?
Or can I for now compile it by myself to be able to use Vitis-AI 3.5 on KV260? If so, how?

Or is KV260 considered obsolete and will not be maintained anymore?

Thank you

iraadit, 2023-07-14:
I now saw [on this documentation page](https://docs.xilinx.com/r/en-US/ug1354-xilinx-ai-sdk/Downloading-the-Vitis-AI-Library):

> Vitis AI 3.5 supports the VEK280 Evaluation Board and the V70 Development Card for AI Inference.
> 
> For the following evaluation boards, refer to Vitis AI 3.0:
> 
>     AMD Zynq™ UltraScale+™ MPSoC ZCU102 Evaluation Board
>     AMD Zynq™ UltraScale+™ MPSoC ZCU104 Evaluation Board
>     AMD Versal™ VCK190 Evaluation Board
>     AMD Kria™ KV260 Vision AI Starter Kit
>     AMD Versal™ VCK5000 Development Card

Does it mean I should also only use code (and models) of 3.0 on KV260?
Or just that I have to install following the instructions of 3.0, but can run 3.5 models on it?

ahnhy1324, 2023-07-17:
I think you can check this document for that question.

https://xilinx.github.io/Vitis-AI/3.5/html/docs/workflow.html

they said this :

**Vitis™ AI 3.5 supports Zynq™ Ultrascale+™ and Versal™ AI Core architectures, however the IP for these devices is now considered mature and will not be updated with each release. Users should understand that we will continue to support these targets into the future and Vitis AI will update the pre-built board images and reference designs for these architectures with each major release (ie, 4.0).

For minor releases (x.5), users wishing to leverage these architectures have two choices:

Leverage Vitis™ AI 3.0 for initial evaluation and development.**

Build a custom board Petalinux image for their target leveraging the Vitis AI 3.5 runtime and libraries.

I use vitis 3.5 for my ultrascale+ board and tools are working like quantizer or some make make models for dpu flow.

but you'd better use 3.0 branch if you want to use pre-built board setup files.

quentonh, 2023-07-18:
Yes, we fully anticipate that the Vitis AI 3.5 release is compatible with ZU+ targets, however, with the 3.5 release we did not do extensive testing, benchmarking nor do we provide pre-built images for those targets.  We will update ZU+ targets with the 4.0 release.  Keep in mind that there were no updates to the DPU IP with 3.5, so it should be very feasible to update the Petalinux image with the latest VART and VAIL versions.

iraadit, 2023-07-28:
Thank you for your answers.
I'll try to update the Petalinux image.

quentonh, 2023-07-28:
@iraadit Please note that the packagegroup-petalinux-vitisai has not yet been released.  However, the package group is simply a matter of convenience, and the availability of that package does not stop users from building their own image.  If you look closely at that package group, you will quickly see that the package group is simply invoking BitBake to build the various components of Vitis AI.  In fact, the Vitis AI repository provides the required BitBake recipes for the same components (albeit discretely) that are included in the package group.  For instance, here is the BitBake recipe for the vitis-ai-library:

[Vitis-AI/src/vai_petalinux_recipes/recipes-vitis-ai/vitis-ai-library/vitis-ai-library_3.5.bb at master · Xilinx/Vitis-AI (github.com)](https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_petalinux_recipes/recipes-vitis-ai/vitis-ai-library/vitis-ai-library_3.5.bb)

The instructions to integrate these packages are quite simple and are provided in the Github.IO documentation:
[DPU IP Details and System Integration — Vitis™ AI 3.5 documentation (xilinx.github.io)](https://xilinx.github.io/Vitis-AI/3.5/html/docs/workflow-system-integration.html#rebuilding-the-linux-image-with-petalinux)

Note!  There is an error in the above documentation that we just noticed this week and I will work on fixing it next week.  Specifically, you can see that Step 5 tells the user to run petalinux-build, but the command that is listed is:

cp Vitis-AI/src/petalinux_recipes/recipes-vai-kernel <petalinux project>/project-spec/meta-user/

Somehow this documentation became muddled in the more recent release.

This older link contains some information that is not found in the latest documentation.  I have some work to do to fix the latest docs and fix this...
https://github.com/Xilinx/Vitis-AI/tree/2.5/setup/petalinux

You will need to use PetaLinux 2023.1 to build an image for Vitis AI 3.5 as there have been updates to packages such as OpenCV in the base PetaLinux image that will not be pulled in if you try to use an older version of PetaLinux with these recipes.

iraadit, 2023-09-08:
Hi @iraadit ,

Sorry for the delayed answer, I worked on other subject and had long holidays.

Did you have the opportunity to update the documentation?

I think I successfully built  PetaLinux 2023.1 for Vitis AI 3.5, here are my notes:

> ## Set Up PetaLinux Working Environment
> 
> See <https://docs.xilinx.com/r/2021.1-English/ug1144-petalinux-tools-reference-guide/Steps-to-Set-Up-PetaLinux-Working-Environment>
> See <https://xilinx.github.io/Vitis-AI/3.5/html/docs/workflow-system-integration.html#rebuilding-the-linux-image-with-petalinux>
> 
> ```
> $ source petalinux-v2023.1-05012318/settings.sh
> $ echo $PETALINUX
> ```
> 
> Download KV260 BSP
> 
> See <https://www.xilinx.com/support/download/index.html/content/xilinx/en/downloadNav/embedded-design-tools.html>
> 
> ```bash
> $ petalinux-create -t project -s xilinx-kv260-starterkit-v2023.1-05080224.bsp
> $ cp -r $PATH_TO_VITIS_AI/Vitis-AI/src/vai_petalinux_recipes/recipes-vitis-ai/ xilinx-kv260-starterkit-2023.1/project-spec/meta-user/
> $ rm -rf xilinx-kv260-starterkit-2023.1/project-spec/meta-user/recipes-vitis-ai/vart/vart_3.5_vivado.bb
> ```
> 
> Edit <petalinux project>/project-spec/meta-user/conf/user-rootfsconfig file, appending the following lines:
> 
> ```bash
> CONFIG_vitis-ai-library
> CONFIG_vitis-ai-library-dev
> CONFIG_vitis-ai-library-dbg
> ```
> 
> ```bash
> $ cd xilinx-kv260-starterkit-2023.1
> $ petalinux-config -c rootfs
> ```
> 
> Select the following option (while leaving package packagegroup-petalinux-kria selected?):
> ```
> Select user packages --->
> Select [*] vitis-ai-library
> ```
> 
> ```
> petalinux-build
> ```
> 
> The full compilation log build.log is stored in the build sub-directory of your PetaLinux project. The final image, <plnx-proj-root>/images/linux/image.ub, is a FIT image. The kernel image (including RootFS) is Image for Zynq® UltraScale+™ MPSoC and Versal™ platform. (from https://docs.xilinx.com/r/2021.1-English/ug1144-petalinux-tools-reference-guide/Default-Image)
> 
> Generate Boot Image
> 
> See <https://docs.xilinx.com/r/2021.1-English/ug1144-petalinux-tools-reference-guide/Generate-Boot-Image?tocId=3fqGlFncsr7XtKH0ay0cVA>
> ```
> petalinux-package --boot --u-boot --format BIN
> ```
> 
> See <https://xilinx.github.io/kria-apps-docs/kv260/2021.1/build/html/docs/build_petalinux.html>
> 
> ```
> petalinux-package --wic --bootfiles "ramdisk.cpio.gz.u-boot boot.scr Image system.dtb"
> ```
> 
> The generated image file will be located at images/linux/petalinux-sdimage.wic.
> 
> Tip: The generated wic file assumes a fixed partition size. The total file size is several GB but the actual usage on the ext4 partition is much lower. The file size can be significantly lower by compressing the file e.g. using gzip:
> 
[... 136 lines cut ...]
> ```
> 
> If the target board is connected to the Internet, you can also use the wget command to download the model directly to the board.
> Log in to the target board (using ssh or serial port) and install the model package.
> If the /usr/share/vitis_ai_library/model folder does not exist on the target, create it first.
> 
> ```
> # sudo mkdir -p /usr/share/vitis_ai_library/models
> ```
> 
> Install the model on the target board.
> 
> ```
> # tar -xzvf <model>.tar.gz -C /usr/share/vitis_ai_library/models
> ```
> 
> By default, the models are located in the /usr/share/vitis_ai_library/models directory on the target.

I think I have a zorking Petalinux with Vitis AI 3.5 on KV260. At least, it is booting. :D

But my problem now is that I cannot test it. Indeed, let's take the classification example: I need to add a model to use (for example, tf_inceptionv1_3.5) on the KV260 target. However, in the `model.yaml' file, I see available models for GPU, vek280 and v70;, but no KV260. Does it mean I have also to create my own compatible model?
Can I use models from 3.0, or can I convert the GPU models to KV260?
How should I proceed from here?

Thank you for your help

iraadit, 2023-09-13:
I cross-compiled some example code and tried to execute it. It failed with the following error message, and I then tried to dig deeper:

```
$ ./test_jpeg_yolov7 yolov7_pt sample_yolov7.jpg
WARNING: Logging before InitGoogleLogging() is written to STDERR
F20230913 05:12:28.760835  1116 dpu_controller.cpp:52] Check failed: !get_factory_methods().empty() 
Aborted

$ xbutil examine
System Configuration
  OS Name              : Linux
  Release              : 6.1.5-xilinx-v2023.1
  Version              : #1 SMP Fri Apr 21 07:47:58 UTC 2023
  Machine              : aarch64
  CPU Cores            : 4
  Memory               : 3931 MB
  Distribution         : PetaLinux 2023.1+release-S05010539 (langdale)
  GLIBC                : 2.36
  Model                : ZynqMP KV260 revB

XRT
  Version              : 2.15.0
  Branch               : 2023.1
  Hash                 : 64c933573e7e50a8aba939a74209590c2b739e8b
  Hash Date            : 2023-04-17 09:18:13
  ZOCL                 : 2.15.0, 64c933573e7e50a8aba939a74209590c2b739e8b

Devices present
  0 devices found
BDF  :  Shell  Platform UUID  Device ID  Device Ready*  
--------------------------------------------------------

* Devices that are not ready will have reduced functionality when using XRT tools

$ dmesg | grep dpu

$ xdputil query
WARNING: Logging before InitGoogleLogging() is written to STDERR
I20230913 05:39:56.710660  1171 xrt_bin_stream.cpp:49] Please check your /etc/vart.conf
Its format should be :
    firmware: xx
Example:
    firmware: /run/media/mmcblk0p1/dpu.xclbin
F20230913 05:39:56.710755  1171 xrt_bin_stream.cpp:53] [UNILOG][FATAL][VART_OPEN_DEVICE_FAIL][Cannot open device] open(/run/media/mmcblk0p1/dpu.xclbin) failed.
/usr/bin/xdputil: line 20:  1171 Aborted                 /usr/bin/python3 -m xdputil $*
xilinx-kv260-starterkit-20231:~/Vitis-AI/examples/vai_library/samples/yolov7$ cat /etc/vart.conf 
firmware: /run/media/mmcblk0p1/dpu.xclbin

$ ls /boot                
Image  boot.scr  ramdisk.cpio.gz.u-boot  system.dtb
```

It seems DPU is not installed / configured

First, it looks for it in /run/media/mmcblk0p1/dpu.xclbin.
However, my SD card is _mmcblk1p1_ (not _mmcblk0p1_), and has been mounted on _/boot_. (I also had to modify the 0 to 1 in fstab for the system to boot)
Second, there is no _dpu.xclbin_ file in _/boot_.

Where should I find, or how should I create this _dpu.xclbin_ file ?

Also, in https://github.com/Xilinx/Vitis-AI-Copyleft-Model-Zoo/tree/main/yolov7, pre-compiled models are available for VEK280 and V70, but not KV260. How should I produce/convert such model for KV260?

Thank you @quentonh

MRFiruzabadi, 2023-12-02:
Hi @iraadit ,

Have you been able to successfully run the YOLOv7 model on the KV260 target? If so, I would greatly appreciate any guidance or instructions you can provide.

Thank you!

iraadit, 2023-12-07:
Hi

No, I've never been able to successfully fully install Vitis-AI 3.5 on the KV260, and sadly had to resort to Vitis-AI 3.0 (and old yolo).

My conclusion from this: better support and more easy to run new models on Nvidia Jetson.