Xilinx/Vitis-AI #1292: vscode debug resnet18_quant.py Permission denied

xuyuanyuan1223, 2023-07-15:
I want to debug the quantization process, and I downloaded the official docker image. After entering, the debugging shows insufficient permissions. How can I solve it?
[image]

xuyuanyuan1223, 2023-07-15:
I try to start the docker container with sudo, but it says
[image]

danielstumpp, 2023-07-17:
This appears to be related to #1287, still unresolved

xuyuanyuan1223, 2023-07-25:
https://login.amd.com/app/amd_wwwxilinxcom_1/exk55966rlztShATk697/sso/saml?SAMLRequest=fZJfT8IwFMW%2FytJ31jG2GRpGMiVGEv8QmT74Qrr1Ag1dO3s7QT%2B9ZWjEB31rbu%2F9ndNzO0HeqJYVndvqR3jtAF1waJRG1l%2FkpLOaGY4SmeYNIHM1WxZ3tywOI9Za40xtFAkKRLBOGn1lNHYN2CXYN1nD0%2BNtTrbOtcgo3e%2F34UEqqQ9hbRp6FKDYUk9ZSwW0Negor5EEM%2B9Can7k%2FUwrs5E65I3oh3nbUn9eeeYJ6YurIYXDLk3HWWbVh1tui3KXjS8oounFSDCf5WQ1WsNQ8FFViVHKs6iOoUog8ZVUpELESTKsRFL5ZsQO5hod1y4ncRSPBtHFIE7LKGPRkKXjMIriFxJcG1tDH2BO1lwhkGDxFcyl1ELqzf8pVqcmZDdluRgsHpYlCZ7BYv9630Cmk6N71vuxZ9v5H8u%2FV0KmfyzA5%2BezFxN6hj9pteze8%2BazhVGyfg8Kpcz%2BygJ3kBNnOyB0epr6%2FXOmnw%3D%3D&RelayState=https%3A%2F%2Fwww%2Exilinx%2Ecom%2Fmember%2Fforms%2Fdownload%2Fdesign%2Dlicense%2Dxef%2Ehtml%3Ffilename%3Dxilinx%2Dkv260%2Ddpu%2Dv2022%2E1%2Dv2%2E5%2E0%2Eimg%2Egz

Alexiazzf, 2023-07-26:
Hi @danielstumpp , this issue is solved in https://github.com/Xilinx/Vitis-AI/issues/1287

quentonh, 2023-07-26:
@Alexiazzf #1287 was opened by @danielstumpp so it is not clear to be that it is resolved.  Perhaps @danielstumpp can confirm. 
 @janifer112x

Alexiazzf, 2023-07-26:
Hi @xuyuanyuan1223, can you open the link with permission denied in this docker images?

danielstumpp, 2023-07-26:
> @Alexiazzf #1287 was opened by @danielstumpp so it is not clear to be that it is resolved. Perhaps @danielstumpp can confirm. @janifer112x

Closed with a suitable workaround. Solution doesn't address this issue though

janifer112x, 2023-07-27:
@xuyuanyuan1223, the issue I saw from the screen  snapshot is the one /root/.vscode-server,   files under /root permission maybe very strict  so it cannot be accessed, how did you launch the docker ? if you use docker_run.sh, the user inside container should be your local uid. if you launch docker as root ,then it should  run as uid 0 inside the docker, you can double check with id -a to check the actual id