Xilinx/Vitis-AI #1295: engineer, how should the data for multitask be prepared, and what information is placed in each .txt file?

zcstart, 2023-07-20:
engineer, how should the data for multitask be prepared, and what information is placed in each .txt file?

for example kitti
 def _init(self):
        for line in open(osp.join(self.root, self.image_set + '.txt')):
            print('---- line ----')
            print(line)
            img_path, label_path, focal_length, baseline = line.split(' ')

line should contain contents?

wangxd-xlnx, 2023-07-21:
Hi @zcstart 

Could you describe which vitis-ai version of which model you are using？In vitis-ai 3.5, model zoo does not have this model.
Or you mean the examples of vitis-ai library?

zcstart, 2023-07-21:
i use vitis-ai 2.5 but  dataset i do not understand how to prepare data

zcstart, 2023-07-21:
01 cityexacpe get 16 class 。which 16 class?
02 depth dataset。 how to understand Depth estimation test images: kitti eigen split? 
03 lane dataset。1) data\multi_task_det5_seg16\lane\train\images    2)data\multi_task_det5_seg16\lane\train\seg
04 drivable dataset。data\multi_task_det5_seg16\drivable\train\images    2)data\multi_task_det5_seg16\drivable\train\seg

wangxd-xlnx, 2023-07-27:
Hi @zcstart 
1.The segmentation dataset is CityScapes, and the original data has a total of 19 categories.  
The specific 16 categories need further confirmation and will feedback later.
2.The depth dataset contains data meaning
img_path : Training RGB image path
label_path : label path
focal_length : focal length
baseline : the distance between the two cameras
3. lane dataset
data\multi_task_det5_seg16\lane\train\images: training data RGB images
data\multi_task_det5_seg16\lane\train\seg: The label for the RGB images of the training data
4. drivable dataset
     Contains classes: 'car', 'sign', 'road'
     data\multi_task_det5_seg16\drivable\train\images: training data RGB images
     data\multi_task_det5_seg16\drivable\train\seg: The label for the RGB images of the training data

zyh321, 2023-09-25:
> Hi @zcstart 1.The segmentation dataset is CityScapes, and the original data has a total of 19 categories. The specific 16 categories need further confirmation and will feedback later. 2.The depth [...]

Hi  [wangxd-xlnx](https://github.com/wangxd-xlnx)
About  the segmentation dataset from 19 categories to 16 categories, how does this happen? Is there any files to process the original CityScapes?