Xilinx/Vitis-AI #1297: What version of Vitis AI supports U200?

zyt1024, 2023-07-23:
Hello,What version of Vitis AI supports U200?Vitis AI3.5? or all?

quentonh, 2023-07-24:
@zyt1024 Please see this reference which documents that Alveo Uxxx card support was deprecated in Vitis AI 3.0.  You should leverage version 2.5.  https://xilinx.github.io/Vitis-AI/3.5/html/docs/workflow.html#supported-evaluation-targets

zyt1024, 2023-08-11:
> @zyt1024 Please see this reference which documents that Alveo Uxxx card support was deprecated in Vitis AI 3.0. You should leverage version 2.5. https://xilinx.github.io/Vitis-AI/3.5/html/docs/workflow.html#supported-evaluation-targets

Hello, this link is Vitis 3.5？， Does Vitis AI 3.5 supports U200?

zyt1024, 2023-08-11:
> @zyt1024 Please see this reference which documents that Alveo Uxxx card support was deprecated in Vitis AI 3.0. You should leverage version 2.5. https://xilinx.github.io/Vitis-AI/3.5/html/docs/workflow.html#supported-evaluation-targets

Hello, this link is Vitis 3.5? Does Vitis AI 3.5 supports U200?

quentonh, 2023-08-11:
@zyt1024 No, my purpose in sending the link was that if you read what is written there you will see "Vitis AI support for the Uxxx Alveo targets was discontinued in the 3.0 release. Please use Vitis AI 2.5 for those targets."

zyt1024, 2023-08-17:
> @zyt1024 No, my purpose in sending the link was that if you read what is written there you will see "Vitis AI support for the Uxxx Alveo targets was discontinued in the 3.0 release. Please use Vitis AI 2.5 for those targets."
Thank you. Unfortunately, my model contains Conv1D , see https://github.com/Xilinx/Vitis-AI/issues/981, the conv1D is not support in `Vitis AI 2.5`, my model contains the `Conv1D`, if I want to deploy my model ,I must be use the `Vitis AI 3.0 `? but I use the board is `U200`,and the `Vitis AI 3.0` is not support `U200`, how can I solve this issue?

zyt1024, 2023-08-17:
> @zyt1024 No, my purpose in sending the link was that if you read what is written there you will see "Vitis AI support for the Uxxx Alveo targets was discontinued in the 3.0 release. Please use Vitis AI 2.5 for those targets."

@quentonh 
Thank you. Unfortunately, my model contains Conv1D , see https://github.com/Xilinx/Vitis-AI/issues/981, the conv1D is not support in Vitis AI 2.5, my model contains the Conv1D, if I want to deploy my model ,I must be use the Vitis AI 3.0 ? but I use the board is U200,and the Vitis AI 3.0 is not support U200, how can I solve this issue?

quentonh, 2023-08-17:
@zyt1024 I know that at least one developer a few weeks back was attempting to leverage Alveo with a newer release but I do not know if this was successful.  You may be able to do this.  You may need to evaluate if you can use updated versions of XRM and XRT, compatible with Vitis AI 3.5, with the older xclbin that is provided in the historic Vitis AI release.  XRT and the Vitis AI Runtime are closely coupled and it will be important that the interactions between the two runtime code bases and the hardware are correct.  If you can make this work, you may not encounter many other issues.

For the most part, you should find that the hooks are there, and that all the source code that is needed exists in the newer repository, but it may or may not work "out of the box" as it has never been tested.  On a fundamental level we have not actually REMOVED Alveo Uxxx targets (for instance, you can see DPUCA [here](https://github.com/Xilinx/Vitis-AI/tree/master/src/vai_runtime/target_factory/targets)) but we have also not tested more recent versions of Vitis AI with the Alveo Uxxx targets.  

If I had to pursue this, I would study the install, setup and related scripts for Alveo installation.  There are minimal differences between the Alveo V70 [scripts ](https://xilinx.github.io/Vitis-AI/3.5/html/docs/quickstart/v70.html) in Vitis AI 3.5 and the historic Alveo Uxx scripts, so you could potentially update the Uxxx scripts to align with the Vitis AI 3.5 release.  From there you could test to determine if xbutil reports success with the install and then proceed to install the Docker and run a modified setup.sh that attempts to enable your U200 card in the newer Docker.

I am sure that there are things that I am not aware of that may block you, but again, all of the necessary Vitis AI components are provided as source code.  In theory, it may be possible, provided that you can get to the point where you can run inference on the card in the Docker.

Unfortunately, if you run into issues, there won't be much we can do to help.