Xilinx/Vitis-AI #1303: VAI profiler error

hocken-li, 2023-07-28:
I follow the setting of the README file. 
When I run
```
./resnet_v1_50_tf_graph_runner ../../resnet50_pruned_0_3_pt/resnet50_pruned_0_3_pt.xm
odel ./ILSVRC2012_val_00009346.JPEG
```
It has the error:
```
Analyzing symbol tables...
203 / 203
81 / 81
6 / 6
45 / 45
3 / 3
terminate called after throwing an instance of 'std::runtime_error'
  what():  XILINX_XRT not set
INFO:root:Generating VTF
ERROR:root:Xmodel info raw data empty
INFO:root:[vart_trace.csv]: Events number: 0
```

I already source /opt/xilinx/xrt/setup.sh
My vitis-ai version is 3.0. What can I do for it?

kkzqp, 2023-10-05:
@hocken-li Did you solve it?

Irelia1, 2024-01-13:
Did you solve it? i have the same question