Xilinx/Vitis-AI #1304: Possible regression in name key handling?

jeinstei, 2023-07-28:
#628 

I'm not sure if this is a regression, but I'm now running into this error again:

```python
Traceback (most recent call last):
  File "runme_tf2.py", line 39, in <module>
    quantized_model = quantizer.quantize_model(calib_dataset=ds)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 979, in quantize_model
    calib_steps)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 491, in _calibrate_without_loss
    target=self._target)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 1275, in create_quantize_model
    quantize_strategy.get_quantize_registry(), mode, target)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 182, in apply
    layer_metadata)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 55, in _apply_availables
    layer_metadata).recursive_transform()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 775, in recursive_transform
    self.layer_metadata).transform()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 711, in transform
    copy.deepcopy(match_layer_node))
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_quantize_transforms.py", line 816, in replacement
    act_layer_name = act_layer_node.layer['name']
KeyError: 'name'
```

I have no idea if anyone else has seen this, but it really feels like the same thing over again. I need to try the sed trick from #628 still.

zhenzhen-AMD, 2023-08-15:
Hi @jeinstei ，
   Could you please provide the code files runme_tf2.py for us to better assist you?

jeinstei, 2023-08-15:
`runme_tf2.py`

```python
#!/usr/bin/env python3

import os
os.environ['TF_CPP_MIN_LOG_LEVEL'] = '3'
os.environ['VAI_LOG_LEVEL']='-1'

import tensorflow as tf
import pandas as pd

tf.keras.backend.set_learning_phase(0)

# Load dataset
storer = pd.HDFStore("data.hdf5", mode="r")
x_train = storer.get("/input_train").to_numpy()
x_val = storer.get("/input_validate").to_numpy()
y_train = storer.get("/output_train").to_numpy()
y_val = storer.get("/output_validate").to_numpy()
storer.close()

# Create tensorflow dataset for ease of management
ds = tf.data.Dataset.from_tensor_slices((x_val, y_val))
ds = ds.batch(5, drop_remainder=True)

# Load model
mm = tf.keras.models.load_model("model_tuned/", compile=False)
mm.compile()
mm.summary()
model = mm

# Quantize model
from tensorflow_model_optimization.quantization.keras import vitis_quantize
quantizer = vitis_quantize.VitisQuantizer(model)
quantized_model = quantizer.quantize_model(calib_dataset=ds)

# Quantization-aware training
quantized_model.save('quantized_model.h5')
```

`runme.sh` -- note the commented out `sed` from fixing the previous bug referenced in the current issue

```bash
# Fix an issue with converting sigmoid functions
#sed -i "s/act_layer_node.layer\['name'\]/act_layer_node.layer\['config'\]\['name'\]/g" /opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/eight_bit/vitis_8bit_quantize_transforms.py

# Run model quantizer
python3 runme_tf2.py
#ipython -i -- runme_tf2.py

# Compile the model
./compile_for_zcu104.sh

# Generate graph
xir svg build/compiled_model_zcu104/radiasoft.xmodel build/compiled_model_zcu104/out.svg
```

`compile_for_zcu104.sh`

```bash
#!/bin/bash

vai_c_tensorflow2 \
        --model           quantized_model.h5 \
        --arch            /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json \
        --output_dir      build/compiled_model_zcu104 \
        --net_name        radiasoft
```

zhenzhen-AMD, 2023-08-15:
Hi @jeinstei ,
Could you please provide the float model files? These would be very helpful in resolving the issue.

jeinstei, 2023-08-15:
I'm getting those together right now -- annnnnd atttached.
[model.zip](https://github.com/Xilinx/Vitis-AI/files/12344980/model.
[data.zip](https://github.com/Xilinx/Vitis-AI/files/12344981/data.zip)
zip)

zhenzhen-AMD, 2023-08-15:
Hi @jeinstei ,
I am currently addressing this issue. Should there be any updates, I will inform you promptly.

zhenzhen-AMD, 2023-08-15:
Hi @jeinstei ,
I've managed to reproduce an error, but it's not consistent with the error you've provided in the issue. Do you have any idea what might be causing this discrepancy?

`=================================================================
 dense (Dense)               (None, 16)                608       
                                                                  
 dropout (Dropout)           (None, 16)                0         
                                                                  
 dense_1 (Dense)             (None, 16)                272       
                                                                  
 dropout_1 (Dropout)         (None, 16)                0         
                                                                  
 dense_2 (Dense)             (None, 5)                 85        
                                                                  
 dropout_2 (Dropout)         (None, 5)                 0         
                                                                  
 dense_3 (Dense)             (None, 1)                 6         
                                                                  
=================================================================
Total params: 971
Trainable params: 971
Non-trainable params: 0
_________________________________________________________________
[VAI INFO] Using func format quantizer
[VAI INFO] Quantizing without specific `target`.
[VAI DEBUG] Optimize Configurations:
[VAI DEBUG] - remove_dropout: True
[VAI DEBUG] - separate_conv_act: True
[VAI DEBUG] - fold_conv_bn: True
[VAI DEBUG] - convert_bn_to_dwconv: True
[VAI DEBUG] - convert_relu6_to_relu: False
[VAI DEBUG] - convert_tf_op_to_keras: True
[VAI DEBUG] - include_cle: True
[VAI DEBUG] - cle_to_relu6: False
[VAI DEBUG] - cle_steps: 10
[VAI DEBUG] - cle_balance_method: max
[VAI DEBUG] - cle_weight_threshold: 0.1
[VAI DEBUG] - train_with_bn: False
[VAI DEBUG] - separable_conv: True
[VAI DEBUG] Separate layer dense_1 activation: sigmoid
[VAI DEBUG] Separate layer dense_2 activation: sigmoid
[VAI DEBUG] Separate layer dense activation: sigmoid
Traceback (most recent call last):
  File "runme_tf2.py", line 39, in <module>
    quantized_model = quantizer.quantize_model(calib_dataset=ds)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 1168, in quantize_model
    self.optimize_model()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 770, in optimize_model
    self._create_optimized_model()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 454, in _create_optimized_model
    self._optimized_model, self._layer_metadata = self._quantizer.create_optimize_model(
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize_func.py", line 88, in create_optimize_model
    optimized_model, layer_metadata = optimize_pipeline.apply(
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 84, in apply
    transformed_model, layer_metadata = _apply_availables(
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 53, in _apply_availables
    transformed_model, layer_metadata = model_transformer.ModelTransformer(
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 833, in recursive_transform                                                                                                                                                          
    return ModelTransformer(transformed_model, self.transforms,                                                                                                                                                                                                                                                                                                                             
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.8/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 801, in transform                                                                                                                                                                    
    self._set_layer_weights(layer, weights_map)                                                                                                                                                                                                                                                                                                                                             
[... cut ...]
    variable.assign(value)                                                                                                                                                                                                                                                                                                                                                                  
ValueError: Cannot assign value to variable ' dense_1/kernel:0': Shape mismatch.The variable shape (37, 16), and the assigned value shape (16, 16) are incompatible.
`

jeinstei, 2023-08-15:
I honestly have no idea as I get past that step with no problems (see details below). Just to confirm, here is the output of some docker information as well:

Docker run command (shell script, truncated):

```bash
...
docker_run_params=$(cat <<-END
    -v /dev/shm:/dev/shm \
    -v /opt/xilinx/dsa:/opt/xilinx/dsa \
    -v /opt/xilinx/overlaybins:/opt/xilinx/overlaybins \
    -e USER=$user -e UID=$uid -e GID=$gid \
    -e VERSION=$VERSION \
    -v $DOCKER_RUN_DIR:/vitis_ai_home \
    -v $HERE:/workspace \
    -w /workspace \
    --rm \
    --network=host \
    ${DETACHED} \
    ${RUN_MODE} \
    $IMAGE_NAME \
    $DEFAULT_COMMAND
END
)

##############################
#--gpus all \

if [[ $IMAGE_NAME == *"gpu"* ]]; then
  docker run \
    $docker_devices \
    --gpus all \
    $docker_run_params
else
  docker run \
    $docker_devices \
    $docker_run_params
fi
```
Docker header

```
...
CUDA Version 11.3.1
...
Setting up jeinstei 's environment in the Docker container...
usermod: no changes
Running as vitis-ai-user with ID 0 and group 0 
...
Docker Image Version: latest   (GPU) 
Vitis AI Git Hash: 2057c588c 
Build Date: 2023-06-27
WorkFlow: tf2
```

<details>
<summary>Docker output when running script</summary>

```
/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/backend.py:452: UserWarning: `tf.keras.backend.set_learning_phase` is deprecated and will be removed after 2020-10-11. To update it, simply pass a True/False value to the `training` argument of the `__call__` method of your layer or model.
  "`tf.keras.backend.set_learning_phase` is deprecated and "
Model: "sequential"
_________________________________________________________________
 Layer (type)                Output Shape              Param #   
=================================================================
 dense (Dense)               (None, 16)                608       
                                                                 
 dropout (Dropout)           (None, 16)                0         
                                                                 
 dense_1 (Dense)             (None, 16)                272       
                                                                 
[... 79 lines cut ...]
    target=self._target)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/vitis_quantize.py", line 1275, in create_quantize_model
    quantize_strategy.get_quantize_registry(), mode, target)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 182, in apply
    layer_metadata)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_transforms_pipeline.py", line 55, in _apply_availables
    layer_metadata).recursive_transform()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 775, in recursive_transform
    self.layer_metadata).transform()
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/graph_transformations/model_transformer.py", line 711, in transform
    copy.deepcopy(match_layer_node))
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow_model_optimization/python/core/quantization/keras/vitis/quantize_strategy/pof2s/vitis_pof2s_quantize_transforms.py", line 816, in replacement
    act_layer_name = act_layer_node.layer['name']
KeyError: 'name'
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['quantized_model.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/radiasoft_DPUCZDX8G_ISA1_B4096_org.xmodel', proto=None)
[ERROR] Not found the file or directory: /workspace/quantized_model.h5
WARNING: Logging before InitGoogleLogging() is written to STDERR
F20230815 08:45:36.045058   200 serialize_v2.cpp:705] [UNILOG][FATAL][XIR_READ_PB_FAILURE][failed to read pb file] file = build/compiled_model_zcu104/radiasoft.xmodel
*** Check failure stack trace: ***
./runme.sh: line 23:   200 Aborted                 (core dumped) xir svg build/compiled_model_zcu104/radiasoft.xmodel build/compiled_model_zcu104/out.svg
```
</details>

zhenzhen-AMD, 2023-09-06:
Hi @jeinstei ,

Thank you for raising this issue. We've identified the problem and it will be fixed in the next version release. Meanwhile, as a workaround, you can avoid the error by using Keras's functional API to construct your model. Below is a reference example using the functional approach:

`
import tensorflow as tf                                                      
from tensorflow.keras.models import Model                                    
from tensorflow.keras.layers import Input, Dense, Dropout                                                                                                  

input_layer = Input(shape=(37))                                              
x = Dense(16, activation='sigmoid')(input_layer)                             
x = Dropout(0.5)(x)                                                          
x = Dense(16, activation='sigmoid')(x)                                       
x = Dropout(0.5)(x)                                                          
x = Dense(5, activation='sigmoid')(x)                                        
x = Dropout(0.5)(x)                                                                                                                                                                                                                                                                                                    
x = Dense(1)(x)                                                                                                                                                                                                                                                                                                        
model = Model(inputs=input_layer, outputs=x)                                                                                                                                                                                                                                                                           
model.summary()                                                                                                                                                                                                                                                                                                        
model.save('float.h5')
`

zhenzhen-AMD, 2023-09-20:
Closing this for now. Please re-open if there are any other concerns. Thanks.

closed 2023-09-20 by zhenzhen-AMD

zhenzhen-AMD, 2023-09-20:
You're welcome.

jeinstei, 2023-11-03:
Even with the new technique, I am still getting the "name" key error as above

jeinstei, 2023-11-03:
Correction, a new error:

```
  "`tf.keras.backend.set_learning_phase` is deprecated and "
Traceback (most recent call last):
  File "runme_tf2.py", line 31, in <module>
    mm = tf.keras.models.load_model("model_tuned/", compile=False)
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/keras/utils/traceback_utils.py", line 70, in error_handler
    raise e.with_traceback(filtered_tb) from None
  File "/opt/vitis_ai/conda/envs/vitis-ai-tensorflow2/lib/python3.7/site-packages/tensorflow/python/framework/type_spec.py", line 469, in _deserialize
    return cls(*serialization)  # pytype: disable=not-instantiable  # trace-all-classes
TypeError: __init__() takes from 2 to 4 positional arguments but 5 were given
```

jeinstei, 2023-11-03:
Updating to 3.5 fixed this -- apologies for commenting on a dead issue.

zhenzhen-AMD, 2023-11-04:
Could you please provide the float model files? These would be very helpful in resolving the issue.

zhenzhen-AMD, 2023-11-04:
Could you please specify the Docker version and Conda environment details needed to run this project?