Xilinx/Vitis-AI #1305: cannot execute yolov6 example on the board

Mushegh097, 2023-07-29:
So I am trying to test the yolov6 on ZCU104 board. I have run the ./build.sh (examples/vai_library/samples/yolov6) and the executable files were generated. I copied the directory to the board from the host and when tried to run one of the executable files it didn't run.
The host has x86_64 architecture but the board is aarch64. I think that's the issue. How can I solve this??

qianglin-xlnx, 2023-07-29:
@Mushegh097 
You need to install the cross-compiler on the host and use cross-compiler to compile the sample, then copy the executable file to the board and run it.
For vai3.0, you can refer to https://xilinx.github.io/Vitis-AI/3.0/html/docs/quickstart/mpsoc.html#setup-the-host to setup the host.

Mushegh097, 2023-07-30:
@qianglin-xlnx 

I have tried this tutorial but the issue still remains. The "source ~/petalinux_sdk_2022.2/environment-setup-cortexa72-cortexa53-xilinx-linux" works fine and after that the executable file for resnet50 example is generated without any errors. But when I copy the executable to the board and try to run, it gives this error:
[image]
the reason is the same. The executable which was generated on the host machine is based on x86-64 arch.

quentonh, 2023-08-10:
@Mushegh097 

quentonh@XSJ-QUENTONH-L1:/mnt/c/Users/quentonh$ echo $CXX

aarch64-xilinx-linux-g++ -mcpu=cortex-a72.cortex-a53 -march=armv8-a+crc -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/home/quentonh/petalinux_sdk_2022.2/sysroots/cortexa72-cortexa53-xilinx-linux