Xilinx/Vitis-AI #1321: unsuccessfully trying to significantly reduce boot time in ZYNQ MPSOC environment 2021.2

hudini87, 2023-08-23:
unsuccessfully trying to significantly reduce boot time in ZYNQ MPSOC
environment 2021.2 .

it takes more than a minute to get the login as a root.

can you give some tips?

i am using petalinux 2021.2 to create linux boot image on Mercury XU8 SOM , zynq mpsoc xzcu7ev SOC assembled on it.

 

it takes more than a minute(!!!!) to complete boot and get into system as a root user, it doesn't matter if i do the boot from SD card (mmc1) or emmc (mmc0), it took the same time and more than a minute.

 

**2 main blocks i have to fix:**

 

it can be seen live during boot that i am waiting (emmc or sd card boot) 20 seconds when Trying to load boot images from mmc1 or mmc0:

 

sswitch to partitions #0, OK
mmc0(part 0) is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
2710 bytes read in 15 ms (175.8 KiB/s)
## Executing script at 20000000
Trying to load boot images from mmc0
**305488600 bytes read in 20165 ms (14.4 MiB/s)**

 

it can be seen live during boot that i am waiting (emmc or sd card) 14 seconds when Trying to unpack rootfs:

 

 

[    2.597160] RPC: Registered tcp transport module.
[    2.601832] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    2.608826] PCI: CLS 0 bytes, default 64
**[    2.612230] Trying to unpack rootfs image as initramfs...
[   16.407064] Freeing initrd memory: 289108K**
[   16.468780] Initialise system trusted keyrings
[   16.468925] workingset: timestamp_bits=46 max_order=20 bucket_order=0
 
 

please advice.

 

thanks and best regards.

quentonh, 2023-08-23:
@hudini87 There are potentially many paths that you can explore to reduce the size of initramfs, but this is very much outside the scope of Vitis AI.

You should first seek to understand precisely what is included in your initramfs and also, determine precisely what is needed for the initial boot.  There is nothing saying that everything that is included in the initramfs in a default PetaLinux build is required for this initial boot, and as a user you can choose to slim this down to the point where the system will still boot and mount an additional filesystem that contains other required packages.  You can also explore the possibility of different compression schemes which may have an impact on unpacking performance, but here again, this is outside the scope of Vitis AI.  These are really Linux development related issues, not specific to Vitis AI.

https://wiki.debian.org/initramfs
https://wiki.ubuntu.com/Initramfs
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842473/Build+and+Modify+a+Rootfs

The best path for you to explore this is not via filing Vitis AI Github issues (there is no bug here), but rather through the use of the Vitis AI forums, Xilinx technical support or your local FAE.

quentonh, 2023-08-23:
Also, perhaps see https://xilinx.github.io/Vitis-Tutorials/2021-1/build/html/docs/Vitis_Platform_Creation/Introduction/02-Edge-AI-ZCU104/step2.html#add-ext4-rootfs-support-optional-but-recommended

quentonh, 2023-08-30:
@hudini87 My recommendation is that you post Linux optimization related questions [here](https://support.xilinx.com/s/topic/0TO2E000000YKXXWA4/embedded-linux) and potentially also that you consult with a local Field Applications Engineer.