Xilinx/Vitis-AI #1323: Issues while compiling model with Conv2d Transpose and Leaky Relu

wilfredkisku, 2023-08-26:
While the Documentation for Vitis-AI 2.5, 3.0 and 3.5 all show that Conv2dTransposed has been accommodated in DPU unit, I am getting a warning that they will be assigned to the CPU. I am not sure why this issue is coming up.

And help will be appreciated.

```
**************************************************
* VITIS_AI Compilation - Xilinx Inc.
**************************************************
[INFO] Namespace(batchsize=1, inputs_shape=None, layout='NHWC', model_files=['./generator_quantized.h5'], model_type='tensorflow2', named_inputs_shape=None, out_filename='/tmp/generator_model_DPUCZDX8G_ISA1_B4096_org.xmodel', proto=None)
[INFO] tensorflow2 model: /workspace/DPU-DCGAN/generator_quantized.h5
[INFO] keras version: 2.12.0
[INFO] Tensorflow Keras model type: functional
[INFO] parse raw model     :100%|█| 12/12 [00:00<00:00, 4745.58it/s]            
[INFO] infer shape (NHWC)  :100%|█| 20/20 [00:00<00:00, 70.97it/s]              
[INFO] perform level-0 opt :100%|█| 1/1 [00:00<00:00, 649.17it/s]               
[INFO] perform level-1 opt :100%|█| 2/2 [00:00<00:00, 2160.90it/s]              
[INFO] infer shape (NHWC)  :100%|█| 22/22 [00:00<00:00, 666.40it/s]             
[INFO] generate xmodel     :100%|█| 22/22 [00:00<00:00, 75.28it/s]              
[INFO] dump xmodel: /tmp/generator_model_DPUCZDX8G_ISA1_B4096_org.xmodel
[UNILOG][INFO] Compile mode: dpu
[UNILOG][INFO] Debug mode: null
[UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
[UNILOG][INFO] Graph name: generator_model, with op num: 40
[UNILOG][INFO] Begin to compile...
[UNILOG][WARNING] xir::Op{name = quant_conv2d_transpose, type = transposed-conv2d-fix} has been assigned to CPU: [DPU does not support activation type: LEAKYRELU. Its alpha is 0.200000, but DPU only support 0.1015625.].
[UNILOG][WARNING] xir::Op{name = quant_conv2d_transpose_1, type = transposed-conv2d-fix} has been assigned to CPU: [DPU does not support activation type: LEAKYRELU. Its alpha is 0.200000, but DPU only support 0.1015625.].
[UNILOG][WARNING] xir::Op{name = quant_conv2d_transpose_2, type = transposed-conv2d-fix} has been assigned to CPU: [DPU does not support activation type: LEAKYRELU. Its alpha is 0.200000, but DPU only support 0.1015625.].
[UNILOG][WARNING] xir::Op{name = quant_conv2d_transpose_weights, type = const-fix} has been assigned to CPU: [Has no fanout or at least one fanout is out of DPU subgraph.].
[UNILOG][WARNING] xir::Op{name = quant_conv2d_transpose_bias, type = const-fix} has been assigned to CPU: [Has no fanout or at least one fanout is out of DPU subgraph.].
[UNILOG][WARNING] xir::Op{name = quant_conv2d_transpose_1_weights, type = const-fix} has been assigned to CPU: [Has no fanout or at least one fanout is out of DPU subgraph.].
[UNILOG][WARNING] xir::Op{name = quant_conv2d_transpose_1_bias, type = const-fix} has been assigned to CPU: [Has no fanout or at least one fanout is out of DPU subgraph.].
[UNILOG][WARNING] xir::Op{name = quant_conv2d_transpose_2_weights, type = const-fix} has been assigned to CPU: [Has no fanout or at least one fanout is out of DPU subgraph.].
[UNILOG][WARNING] xir::Op{name = quant_conv2d_transpose_2_bias, type = const-fix} has been assigned to CPU: [Has no fanout or at least one fanout is out of DPU subgraph.].
[UNILOG][INFO] Total device subgraph number 5, DPU subgraph number 2
[UNILOG][INFO] Compile done.
[UNILOG][INFO] The meta json is saved to "/workspace/DPU-DCGAN/./meta.json"
[UNILOG][INFO] The compiled xmodel is saved to "/workspace/DPU-DCGAN/./generator_model.xmodel"
[UNILOG][INFO] The compiled xmodel's md5sum is eaffda3ecbaa7e2b6df4a557e16c4ec2, and has been saved to "/workspace/DPU-DCGAN/./md5sum.txt"
```

zhenzhen-AMD, 2023-08-28:
Hi @wilfredkisku ，

Could you please provide the float.h5 and quantized.h5 files? These would be very helpful in resolving the issue.

wilfredkisku, 2023-08-28:
Thank you for the reply. I have attached the [files](https://drive.google.com/file/d/1yegarRlNzxdcFuWskTlVMRXvgQ34fPgp/view?usp=drive_link) for your reference. Also is there any workaround for leaky ReLU with alpha as 0.2?

I don't want the model to use any other value of alpha as it does not perform well with any other value of alpha other than 2.0.

Thank you once again for your help.

zhenzhen-AMD, 2023-09-06:
Hi @wilfredkisku ，
Leakyrelu is allocated to the DPU only when its alpha is set to 0.1; if alpha is 0.2, it defaults to the CPU, with no available workaround.

zhenzhen-AMD, 2023-09-20:
Closing this for now. Please re-open if there are any other concerns. Thanks.

closed 2023-09-20 by zhenzhen-AMD