Xilinx/Vitis-AI #1325: Docker image that contains ONNX quantizer Release Date

farah-rostom98, 2023-08-27:
The latest docker image :docker pull xilinx/vitis-ai-cpu:latest  
contained vitis ai v2.5 which doesn't contain ONNX quantizer so when will the docker image that contains ONNX quantizer be released?

quentonh, 2023-08-28:
@farah-rostom98 vitis-ai-cpu:latest refers to a deprecated container name that was associated with previous releases which incorporated more than one framework into a single container.  Those images are still on Docker Hub but are not updated.  

Vitis AI 3.5 uses a different container naming scheme that is per framework.  See:  https://xilinx.github.io/Vitis-AI/3.5/html/docs/install/install.html#option-1-leverage-the-pre-built-docker.

So for instance:

`./docker_run.sh xilinx/vitis-ai-pytorch-cpu:latest`

Did you find the command "docker pull xilinx/vitis-ai-cpu:latest" in association with documentation for our current release?  If so, please let us know so that we can ensure that this is updated.

farah-rostom98, 2023-08-28:
> You don't often get email from ***@***.*** Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
>
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

My problem is I'd like to use onnx and I don't know which framework supports it?

CONFIDENTIALITY NOTICE: This e-mail transmission (and/or the attachments accompanying it) may contain confidential information belonging to the sender. The information is intended only for the use of the intended recipient. If you are not the intended recipient, you are hereby notified that any disclosure, copying, distribution or the taking of any action in reliance on the contents of this information is strictly prohibited. Any unauthorized interception of this transmission is illegal under the law. If you have received this transmission in error, please promptly notify the sender by reply e-mail, and then destroy all copies of the transmission.

quentonh, 2023-08-28:
@farah-rostom98 The ONNX Quantizer was introduced in Vitis AI 3.5, so the link that I provided is the correct starting point.

quentonh, 2023-08-28:
`docker pull vitis-ai-pytorch-cpu:latest`

farah-rostom98, 2023-08-28:
Thank you so much I’ll hive it a try and get back to you

quentonh, 2023-08-28:
Sorry, I was distracted by the use of the old container nomenclature.  What you really want, if you want native ONNX Quantization is to follow the separate and distinct steps for that tool.  I would recommend that you start here:  https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Installing-vai_q_onnx and also here:  https://github.com/Xilinx/Vitis-AI/tree/v3.5/src/vai_quantizer/vai_q_onnx

farah-rostom98, 2023-08-28:
Thank you so much I’m downloading the PyTorch-cpu latest and I’ll go on from there

quentonh, 2023-08-28:
FWIW, I just pulled vitis-ai-pytorch-cpu:latest, [built ](https://github.com/Xilinx/Vitis-AI/tree/v3.5/src/vai_quantizer/vai_q_onnx#install-from-source-code-with-wheel-package) the ONNX Quantizer from source, installed the .whl package and was able to run quantization of this [example ](https://github.com/Xilinx/Vitis-AI/tree/v3.5/src/vai_quantizer/vai_q_onnx/example/resnet_ptq_cpu) inside the Pytorch container without any issue.

farah-rostom98, 2023-08-29:
Hi Quenton,
The quantizer did build and I tried it on the model example but now I have another question.
Is there a onnx compiler ?? Or how am I going to deploy the onnx if there is no compiler ?

farah-rostom98, 2023-08-30:
Thank you so much for your assistance

@farah-rostom98<https://github.com/farah-rostom98> In the Vitis AI flow for ONNX models, the model compiler is directly integrated into the Vitis AI ONNX Runtime Engine (aka "VOE). Please see the diagram here<https://xilinx.github.io/Vitis-AI/3.5/html/docs/workflow-third-party.html#onnx-runtime>. Please review the VOE section in UG1414 here<https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Programming-with-VOE>.

closed 2023-08-30 by farah-rostom98