Xilinx/Vitis-AI #1326: Vitis AI Execution Provider- ONNX runtime inference error

farah-rostom98, 2023-08-30:
`
import onnnxruntime

session = onnxruntime.InferenceSession(
'[model_file].onnx',
providers=["VitisAIExecutionProvider"],
provider_options=[{"config_file":"/etc/vaip_config.json"}])
`
os images : ubuntu 22.04
vitis version: v2.5

error 

`
UserWarning: Specified provider 'VitisAIExecutionProvider' is not in available provider names.Available providers: 'CPUExecutionProvider'
`

quentonh, 2023-08-30:
@farah-rostom98 Are you trying to run the model on your PC or on a target board?  It almost appears to me that you are attempting to run it on your PC, which to me implies your target may be Alveo.  Vitis AI ONNX Runtime support was first released with Vitis AI 3.0 and supports Edge targets and not Alveo.  The first release of the Vitis AI ONNX model Quantizer was v3.5 of Vitis AI.  There is some historic Alveo ONNX Runtime support which may still be functional with Alveo, but which hasn't been tested with more recent versions of Vitis AI.

Perhaps tell us more about what you are trying to do.  You quantized an ONNX model.  To what target do you want to deploy that model, and with what release of Vitis AI?

farah-rostom98, 2023-08-30:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

No I’m running it on kv260 board using the provided os image for Ubuntu 22.04 and it has vitis AI version 2.5

CONFIDENTIALITY NOTICE: This e-mail transmission (and/or the attachments accompanying it) may contain confidential information belonging to the sender. The information is intended only for the use of the intended recipient. If you are not the intended recipient, you are hereby notified that any disclosure, copying, distribution or the taking of any action in reliance on the contents of this information is strictly prohibited. Any unauthorized interception of this transmission is illegal under the law. If you have received this transmission in error, please promptly notify the sender by reply e-mail, and then destroy all copies of the transmission.

quentonh, 2023-08-30:
@farah-rostom98 Right, that makes sense....Ubuntu + VAI2.5.  

Officially the Vitis AI 3.0 VOE (ORT) package has been tested on the Vitis AI 3.0 pre-built board images.  There does exist a board image for the KV260 in 3.0 (different than the Ubuntu board image that you are working with).  I cannot say offhand what will happen if you attempt to deploy our VOE package on a Vitis AI 2.5 image.  The KV260 image that I refer to is linked in [Step 2 of this section](https://xilinx.github.io/Vitis-AI/3.0/html/docs/quickstart/mpsoc.html#setup-the-target) of the docs.