Xilinx/Vitis-AI #1328: Yolov8 Vai Quantizer

farah-rostom98, 2023-09-04:
Used the static quantizer on yolov8 architecture and the resulting model have changes in the shape of the weights which is not logical for example a layer original shape is (96,96,1,1) and the resulting shape is (3,)

Alexiazzf, 2023-09-06:
Hi @farah-rostom98 ,

What is the kernal shape? and can you upload your code and log?

farah-rostom98, 2023-09-06:
I tried the Vai Onnx quantizer on this model [Yolov8n-seg](https://docs.ultralytics.com/tasks/segment/)

1- used yolo export to export the model in onnx form with fp32 weights.
2- Prepared the caliration data reader
3- called vai_q_onnx. 

the code 
[vai_quantizer.zip](https://github.com/Xilinx/Vitis-AI/files/12534805/vai_quantizer.zip)

Log (docker image: xilinx/vitis-ai-pytorch-cpu:latest)
`RuntimeError: module compiled against API version 0x10 but this version of numpy is 0xf . Check the section C-API incompatibility at the Troubleshooting ImportError section at https://numpy.org/devdocs/user/troubleshooting-importerror.html#c-api-incompatibility for indications on how to solve this problem .
images
Finding optimal threshold for each tensor using PowerOfTwoMethod.MinMSE algorithm ...
Calibrated and quantized model saved.
`

SKKUJoseph, 2024-10-16:
Hi.
Did u solve issue?