Xilinx/Vitis-AI #1334: There is a huge difference between my quantitative output and deployment output

njustlxk, 2023-09-08:
I am trying to deploy an unet generative neural network on the ZCU102 development board and the quantization result error is very small.But during deployment, almost a completely incorrect output was generated.My network is based on the Pytorch framework, and my Vitis AI Docker is version 2.5.Here is my deployment code.Both network input and output are 256 × 256 single channel graph as data for a certain scientific research purpose.Both network input and output are 256 × The 256 single channel graph serves as data for a certain scientific research purpose rather than a civilian use, so the network output is not intuitive.Can you tell me how to solve this problem or help me check my code? Thank you very much!

# deploy code
```
from ctypes import *
from typing import List
import os
import cv2
import time
import argparse
import numpy as np
from PIL import Image, ImageDraw
import xir
import vart
import sys
from unet_prepost import UnetPrePost

def make_parser():
    parser = argparse.ArgumentParser("Unet Demo!")
    parser.add_argument("--model_path", default="model_data/unet.xmodel", help="path to model")

    parser.add_argument("--data_path", default="data", help="path to images")

    parser.add_argument("--save_path", default="out", help="path to save quant results")

    return parser

class UnetVart():
    def __init__(self,
                 model_path='model_data/Unet_int.xmodel'):
        super(UnetVart, self).__init__()

        self.pre_post = UnetPrePost()

        self.ModelInit(model_path)

    """
    obtain dpu subgrah
    """

    def get_child_subgraph_dpu(self, graph: "Graph") -> List["Subgraph"]:
        assert graph is not None, "'graph' should not be None."
        root_subgraph = graph.get_root_subgraph()
        print("root_subgraph:", root_subgraph)
        assert (root_subgraph
                is not None), "Failed to get root subgraph of input Graph object."
        if root_subgraph.is_leaf:
            return []
        child_subgraphs = root_subgraph.toposort_child_subgraph()
        assert child_subgraphs is not None and len(child_subgraphs) > 0
        return [
            cs for cs in child_subgraphs
            if cs.has_attr("device") and cs.get_attr("device").upper() == "DPU"
        ]

    def InputOutInit(self, runner: "Runner"):
        """get tensor"""
        inputTensors = runner.get_input_tensors()
        input_ndim = tuple(inputTensors[0].dims)
        """prepare batch input/output """
        inputData = np.empty(input_ndim, dtype=np.int8, order="C")
        input_fixpos = inputTensors[0].get_attr("fix_point")
        input_scale = 2 ** input_fixpos

        outputDatas = []
        output_scales = []
        outputTensors = runner.get_output_tensors()
        for outputTensor in outputTensors:
            output_ndim = tuple(outputTensor.dims)
            output_fixpos = outputTensor.get_attr("fix_point")
[... 165 lines cut ...]

```
# deploy preprocess code
```
import cv2
import random
import colorsys
import numpy as np

class UnetPrePost():
    def __init__(self):
        super(UnetPrePost, self).__init__()

    def postprocess(self, model_outputs):

        array = ((model_outputs - model_outputs.min()) / (model_outputs.max() - model_outputs.min())) * 255
        array = array.astype(np.uint8)
        
        return array
    
    def get_metrics(self):
        return self.psnr, self.ssim
```
[image]
[xmodel and model.zip](https://github.com/Xilinx/Vitis-AI/files/12555952/xmodel.and.model.zip)

jiangaojie, 2023-09-09:
We meet the same problem when deploy yolov5s model on zcu102 with 2.0 version vitis ai. So far, we don't know what the reason is

njustlxk, 2023-09-11:
> We meet the same problem when deploy yolov5s model on zcu102 with 2.0 version vitis ai. There is a significant difference between quantitative results and deployment results . So far, we don't know what the reason is.

I want to know how to look at the middle layer of the network, but I can't find any way, so we can only see the results of DPU input and output when debugging now. The debugging method provided in the ug1414 document seems to only be for TensorFlow

njustlxk, 2023-09-13:
Sorry, I think I know where the error occurred. The quantified 8-bit data is a signed number, and I mistakenly read them in Python and directly output their size to display the image, resulting in a huge error

jiangaojie, 2023-09-13:
We've been stuck here for a long time now. Can you share your quantization code and deploy code ?
We will express our great gratitude to you!
my email is aojie@smail.nju.edu.cn

njustlxk, 2023-09-13:
> We've been stuck here for a long time now. Can you share your quantization code and deploy code ? We will express our great gratitude to you! my email is [aojie@smail.nju.edu.cn](mailto:aojie@smail.nju.edu.cn)

My deployment and quantification code are attached in the question, and you can freely read it if you need it. It is worth noting that as long as the input is correct, the output of the DPU should be correct, and fixed point arithmetic should be used with caution

jiangaojie, 2023-09-13:
> My deployment and quantification code are attached in the question, and you can freely read it if you need it. It is worth noting that as long as the input is correct, the output of the DPU should [...]

Thank you very much !

Fomalhautwx, 2023-09-19:
> Sorry, I think I know where the error occurred. The quantified 8-bit data is a signed number, and I mistakenly read them in Python and directly output their size to display the image, resulting in a huge error

Sorry but may you elucidate the solution in detail? eg. which code is incorrect and how to rectify the input of 8-bit data. I am still quite illusioned 'bout the very bug. Thank u very much for that favor. 🥺🥺🥺

njustlxk, 2023-09-25:
> Sorry but may you elucidate the solution in detail? eg. which code is incorrect and how to rectify the input of 8-bit data. I am still quite illusioned 'bout the very bug. Thank u very much for that [...]

Sorry, I just saw your request.I found out during the debugging process that my DPU was actually completely fine, but I later realized the importance of formatting. Data converted to int8 cannot be easily manipulated because there are many differences between integer data and floating-point data. For example, in some versions of the Python compiler, it reads 11111111 (int8) as -1, while the other part reads 255 for actual floating-point operations. On the other hand, the operational process of DPU and the quantization output network are not completely equivalent, and it may introduce some errors to some extent, which may lead to