Xilinx/Vitis-AI #1335: VEK280 DPU-TRD Build Issue

AlbertaBeef, 2023-09-11:
Tool Version : 2023.1
Vitis-AI Version : 3.5
Targetted Platform : VEK280
Related to : https://xilinx.github.io/Vitis-AI/3.5/html/docs/workflow-system-integration.html#vek280-dpu-reference-design
Design Files Used : https://www.xilinx.com/bin/public/openDownload?filename=DPUCV2DX8G_VAI_v3.5.tar.gz

I was able to successfully build the vek280_platform.

For the vitis project, however, I have encountered the following build issue:

...
DEBUG:AIECDOGenerator:### Entering AIE CDO Generator pass
INFO: [aiecompiler 77-43964] Opened file : ./Work/ps/cdo/generated-source/cdo_main.cpp
INFO: [aiecompiler 77-43964] Opened file : ./Work/ps/cdo/generated-source/gen_cdo.cpp
INFO: [aiecompiler 77-43964] Opened file : ./Work/ps/cdo/generated-source/gen_cdo.h
INFO: [aiecompiler 77-43964] Opened file : ./Work/ps/cdo/generateAIEConfig
INFO: [aiecompiler 77-43964] Opened file : ./Work/ps/cdo/Makefile
INFO: [aiecompiler 77-404] Executing Cmd: make -C ./Work/ps/cdo -f Makefile all
(rm -rf generated-objects/*  *.bin)
(/tools/Xilinx/Vitis/2023.1/aietools/tps/lnx64/gcc/bin/g++ -fPIC -c -std=c++17 -I /tools/Xilinx/Vivado/2023.1/include  -D__AIEARCH__=20 -D__LOCK_FENCE_MODE__=0 -DAIE_OPTION_SCALAR_FLOAT_ON_VECTOR -DAIE2_FP32_EMULATION_ACCURACY_FAST -Wno-deprecated-declarations -D__AIESIM__ -D__CDO__ -D__PS_INIT_AIE__ -Og -flto -I /tools/Xilinx/Vitis/2023.1/aietools/include -I /tools/Xilinx/Vitis/2023.1/aietools/include/adf -I /tools/Xilinx/Vitis/2023.1/aietools/tps/lnx64/gcc/include/c++/6.2.0 -I /tools/Xilinx/Vitis/2023.1/aietools/tps/lnx64/gcc/include/c++/6.2.0/backward -I /tools/Xilinx/Vitis/2023.1/aietools/tps/lnx64/gcc/include/c++/6.2.0/x86_64-pc-linux-gnu -I /tools/Xilinx/Vitis/2023.1/aietools/include/common_cpp/common_cpp_v1_0/include -I /tools/Xilinx/Vitis/2023.1/aietools/include/drivers/aiengine -I . -I . -I ../../../  -I ../../.././direct_conv_int8x8_C16 -o "generated-objects/gen_cdo.o" "generated-source/gen_cdo.cpp")
(/tools/Xilinx/Vitis/2023.1/aietools/tps/lnx64/gcc/bin/g++ -fPIC -c -std=c++17 -I . -I /tools/Xilinx/Vitis/2023.1/aietools/include -o "generated-objects/cdo_main.o" "generated-source/cdo_main.cpp")
(/tools/Xilinx/Vitis/2023.1/aietools/tps/lnx64/gcc/bin/g++ -o "generated-objects/cdo_main.out" generated-objects/gen_cdo.o generated-objects/cdo_main.o -Wl,--allow-shlib-undefined -L /tools/Xilinx/Vitis/2023.1/aietools/lib/lnx64.o -lcdo_driver -lxioutils -lxaiengine -ladf_api -Wl,-R/tools/Xilinx/Vitis/2023.1/aietools/tps/lnx64/gcc/lib64 )
(./generateAIEConfig)
Initializing AIE driver...
Initializing ADF API...
XAIEFAL: INFO: Resource group Avail is created.
XAIEFAL: INFO: Resource group Static is created.
XAIEFAL: INFO: Resource group Generic is created.
**[AIE ERROR] XAieSim_GetStackRange():522: Invalid Map file, 2: No such file or directory
[AIE ERROR] XAie_LoadElf():707: Stack range definition failed
ERROR: Failed to load elf for core(4,4)**
Generating: /media/albertabeef/TheExpanse/Xilinx_2023_1/tmp/DPUCV2DX8G_VAI_v3.5-work01/vitis_prj/aie/ip/Work/ps/cdo/aie_cdo_reset.bin
Generating: /media/albertabeef/TheExpanse/Xilinx_2023_1/tmp/DPUCV2DX8G_VAI_v3.5-work01/vitis_prj/aie/ip/Work/ps/cdo/aie_cdo_clock_gating.bin
Generating: /media/albertabeef/TheExpanse/Xilinx_2023_1/tmp/DPUCV2DX8G_VAI_v3.5-work01/vitis_prj/aie/ip/Work/ps/cdo/aie_cdo_mem_clear.bin
Generating: /media/albertabeef/TheExpanse/Xilinx_2023_1/tmp/DPUCV2DX8G_VAI_v3.5-work01/vitis_prj/aie/ip/Work/ps/cdo/aie_cdo_error_handling.bin
Generating: /media/albertabeef/TheExpanse/Xilinx_2023_1/tmp/DPUCV2DX8G_VAI_v3.5-work01/vitis_prj/aie/ip/Work/ps/cdo/aie_cdo_elfs.bin
Loading elfs of graph g...
make[2]: *** [Makefile:21: run] Error 1
**ERROR: [aiecompiler 77-4505] Failed in generating CDO.**
INFO: [aiecompiler 77-44877] Run completed. Find additional information in:
     Guidance: ./Work/reports/guidance.html

INFO: [aiecompiler 77-44878] Use the vitis_analyzer tool to visualize and navigate the relevant reports. Run: 
     vitis_analyzer ./Work/graph_py.aiecompile_summary

Compilation Failed

(WARNING:42, CRITICAL-WARNING:0, ERROR:1)
/tools/Xilinx/Vitis/2023.1/aietools/bin/rdiArgs.sh: line 352: 154381 Segmentation fault      (core dumped) "$RDI_PROG" "$@"
segfault in /tools/Xilinx/Vitis/2023.1/aietools/bin/unwrapped/lnx64.o/aieir_be -exec aieir_be --time-passes=0 --disable-om-fifo-insertion=false --logcfg-file= --trace-plio-width=64 --aie2ipu-base-addr=1073741824 --pl-freq=100 --enable-light-cdo=false --show-loggers=false --fast-nonlinearfloats=false --broadcast-enable-core=true --high-performance=false --multi-layer-opt=0 --kernel-address-location=false --target=hw --swfifo-threshold=40 --single-mm2s-channel=false --workdir=./Work --multi-layer-enable-shared-channel-opt=true --exit-after=complete --event-trace-config= --test-iterations=-1 --stacksize=1280 --known-tripcount=false --enable-core-processor-bus=false --platform=/media/albertabeef/TheExpanse/Xilinx_2023_1/tmp/DPUCV2DX8G_VAI_v3.5-work01/vek280_platform/platform_repo/xilinx_vek280_es1_base_202310_1/export/xilinx_vek280_es1_base_202310_1/xilinx_vek280_es1_base_202310_1.xpfm --event-trace-custom-config= --disable-dma-cmd-alignment=false --enable-ecc-scrubbing=true --float-accuracy=fast --output-archive=libadf.a --write-partitioned-file=true --print-timer-multi-layer-ctrl=false --schemafile=AIEGraphSchema.json --disable-multirate=false --multi-layer-controller-parallel=false --include=./direct_conv_int8x8_C16 --verify-switchconfig=true --device= --enable-peano-flow=false --Xelfgen=-j 8 --disable-multirate-analysis=false --fastmath=false --cpp-std=-std=c++17 --max-layer-ctrl-param-size=256 --enable-mapper-hints-constraints=false --multi-layer-auto-phase=false --event-trace-advanced-mapping=0 --multi-layer-plc-constraints= --lock-fence-mode=0 --enable-reconfig=false --compute-heapsize=true --log-level=5 --aiesim-xrt-api=false --Xmapper=outgoingpliolimitpertile:6 --Xmapper=incomingpliolimitpertile:8 --evaluate-fifo-depth=false --gen-graph-cleanup=false --use-canonical-net-names=false --event-trace-port=gmio --use-phy-shim=true --print-timer-multi-layer-core=false --xlopt=1 --graph-iterator-event=false --pre-compile-kernels=false --trace-aiesim-option=0 --aiearch=aie-ml --mapped-soln-udm= --optimize-pktids=false --no-init=false --adf-api-log-level=2 --num-trace-streams=16 --aie-heat-map=false --enable-partition=false --fast-floats=true --quiet=false --enable-multi-layer=false --multi-layer-enable-tct=false --multi-layer-pipelining=false --exec-timed=0 --pl-auto-restart=false --routed-soln-udm= --large-kernel-program=false --read-mem-multi-layer=false --enable-profiling=false --runtime-opt=true --part= --disable-transform-merge-broadcast=false --verbose=true --use-async-rtp-locks=true --repo-path= --genArchive=false --kernel-compile-replacement= --pl-axi-lite=false --event-trace-bounding-box= --multi-layer-shim-dma-bandwidth=0 --heapsize=1024 --logical-arch= --enable-reconfig-dma-autostart=false --nodot-graph=false --use-core-control-api=false --disable-dma-autostart=false --disable-transform-broadcast-split=true --json=./Work/temp/graph_py.json --frontend-warning=0 --frontend-critical-warning=0 --frontend-summary-name=./Work/graph_py.aiecompile_summary --frontend-files-to-report=Guidance:./Work/reports/guidance.html --sdf-graph=/media/albertabeef/TheExpanse/Xilinx_2023_1/tmp/DPUCV2DX8G_VAI_v3.5-work01/vitis_prj/aie/graph/graph_py.cpp, exiting...
/tools/Xilinx/Vitis/2023.1/aietools/bin/aieir_be: line 98: kill: (-154351) - No such process
/tools/Xilinx/Vitis/2023.1/aietools/bin/aiecompiler: line 83: kill: (-154244) - No such process
make[1]: *** [Makefile:83: ip] Error 255
make: *** [Makefile:148: /media/albertabeef/TheExpanse/Xilinx_2023_1/tmp/DPUCV2DX8G_VAI_v3.5-work01/vitis_prj/aie/ip/libadf.a] Error 2

I am looking for ideas as to what could be causing this error.
Any help, feedback, or suggestions are welcome.

Macro-A, 2023-09-14:
please try with command:   apt-get install util-linux

quentonh, 2023-09-22:
Solution in this case was to force the reinstallation of rename, which may conflict with the Perl version by default.  

$ which rename
/usr/bin/rename
 
$ ls -l /usr/bin/rename
lrwxrwxrwx 1 root root 24 May 19 09:06 /usr/bin/rename -> /etc/alternatives/rename
 
$ ls -l /etc/alternatives/rename
lrwxrwxrwx 1 root root 20 May 19 09:06 /etc/alternatives/rename -> /usr/bin/file-rename
 
$ file /usr/bin/file-rename 
 /usr/bin/file-rename: Perl script text executable
 
$ rename –version
/usr/bin/rename using File::Rename version 1.30, File::Rename::Options version 1.10

Simply re-installing rename is insufficient to solve this (ie, sudo apt-get reinstall util-linux will not do the job).  

Users need to execute the following commands in order to resolve this:

wget https://mirrors.edge.kernel.org/pub/linux/utils/util-linux/v2.35/util-linux-2.35-rc1.tar.gz
cd util-linux-2.35-rc1
./configure
make rename
cp rename ~/bin
export PATH="$HOME/bin:$PATH"

AlbertaBeef, 2023-09-23:
Yes, thank you for the solution :)
Very much appreciated.

closed 2023-09-23 by AlbertaBeef