Xilinx/Vitis-AI #1338: Can not execute the model from model zoo on FPGA board

Mushegh097, 2023-09-13:
Hi,

I have downloaded the HRNET model for zcu104/zcu102 boards from the Vitis-AI v3.0 model zoo. I want to execute the model on the FPGA board but the directory doesn't contain the executable files for the target board. Can you please explain how should I run inference on the board. My main goal for this is to compare the results from the GPU and the FPGA

Thanks,
Mushegh

lishixlnx, 2023-09-22:
1. can you please check if the hrnet model is in the directory /usr/share/vitis_ai_library/models/;
2. can you please check if this directory exist?
  /home/root/Vitis-AI/examples/vai_library/samples/segmentation
  if not, please search similar directory in /home/root/Vitis-AI/

Mushegh097, 2023-09-25:
Hi @lishixlnx ,

Thanks! I was able to find the model.

But I am getting an error when trying the inference.
I have run the "./test_jpeg_segmentation HRNet_pt" and as an input gave an image from the COCO val2017 dataset but got this error:
"F0925 19:48:41.571030  1010 buffer_object_xrt_imp.cpp:78] Check failed: bo_ != XRT_NULL_BO  allocation failure:  xrt_.handle 0xaaaae25928d0 xrt_.device_id 0 size  536850432 xrt_.flags 0x1000000"

I think it is a memory issue but when I check the memory on my ZCU104 board it shows that 1.8gb memory is free:

MemTotal:        2033108 kB
MemFree:         1808088 kB
MemAvailable:    1838296 kB

Can you please help to understand the issue? I am using the latest DPU trd image (Vitis-Ai v3.0).

Thanks,
Mushegh

lishixlnx, 2023-09-26:
please run below commands and paste result here:
# cat /proc/meminfo
#  xdputil xmodel -l /usr/share/vitis_ai_library/models/HRNet_pt/HRNet_pt.xmodel
# xdputil query

Mushegh097, 2023-09-26:
Meminfo: [image]
Query:[image]
xmodel -l : [image]
[image]

lishixlnx, 2023-09-27:
from pic1, you can see your CMA total is 1572864KB  ( about 1.5G)
from pic3, please add all the REG size from the "reg info" section (in this version, some seg info is missing), it may exceed the size of your CMA.
That causes the memory problem.  You can try with vck190 or vek280 board to test this model.

Mushegh097, 2023-09-27:
Hi @lishixlnx,

I have downloaded the model from the model zoo of Vitis-AI v3.0 and it says that it supports zcu102 and zcu104.
info from model.yml:
********************************************************************************************************************************************
  board: zcu102 & zcu104 & kv260
  download link: https://www.xilinx.com/bin/public/openDownload?filename=HRNet_pt-zcu102_zcu104_kv260-r3.0.0.tar.gz
  checksum: bea55b0b98e7460895d68b3d30d460f4
********************************************************************************************************************************************

The sum of the regs exceeds the CmaFree that's the reason of the error I think. Is there a way to optimize the model?

Thanks,
Mushegh

quentonh, 2023-09-27:
This may help.....

DPU required Memory in Linux
CMA
This driver depends on CMA and DMA_CMA by default, which enables the Contiguous Memory Allocator which allows the driver to allocate big physically-contiguous blocks of memory for use with DPU. If you don't want to use the default CMA pool in Linux, you can specify a reserved_memory in the DTS and manage it by yourself. For more details please refer to [Reserved_Memory](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18841683/Linux+Reserved+Memory), the following demonstrates how to configure the default CMA pool size.

Using Petalinux kernel menuconfig option
Run kernel menuconfig using the below cmds
petalinux : "petalinux-config -c kernel"
yocto: " bitbake -f virtual/kernel -c menuconfig"
Go to Device Drivers→Generic Driver options→ DMA Contiguous Memory Allocator
Set default contiguous memory size as per your preferred value depending upon your use case
Save the configuration, exit the menuconfig and build images

Using bootargs at uboot prompt:
During boot-up stop at uboot prompt by continuously pressing any key during bootup and set the custom CMA size using the below cmd

setenv bootargs '$bootargs cma=512m'
boot

Using custom uEnv.txt

There is a custom uEnv.txt file in sdcard "boot" partition being generated by yocto/petalinux build which can be edited to set custom cma size.

Modify uEnv.txt bootargs option to set the requried cma size as below
"bootargs=earlycon clk_ignore_unused root=/dev/mmcblk0p2 rw rootwait cma=1500m"
Copy uEnv.txt having CMA boot args setting in boot partition (i.e the first partition) of sdcard and boot the board.