Xilinx/Vitis-AI #1343: Wrong onnxruntime import - vai_q_onnx

ZEB1CLJ, 2023-09-27:
**Summary**
Installing vai_q_onnx from the `Vitis-AI` repository (according to: https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Installing-vai_q_onnx) and trying to import the `vai_q_onnx` package causes an *ImportError*

**Environment**
```
onnx                     1.14.1
onnxruntime              1.16.0
onnxruntime-extensions   0.9.0
vai-q-onnx               1.14.0
```

**How to reproduce**
```
# Clone git repo
git clone https://github.com/Xilinx/Vitis-AI
# Get to vai_q_onnx library folder
cd Vitis-AI/src/vai_quantizer/vai_q_onnx
# Install library
chmod +x build.sh
sh build.sh
pip install pkgs/*.whl
# Start python interpreter
python3
# Inside the live interpreter, run:
import vai_q_onnx
```
As you run the last step, an error should be raised, saying:
```
ImportError: cannot import name 'clone_model_with_shape_infer' from 'onnxruntime.quantization.quant_utils' (<home_folder>/.local/lib/python3.8/site-packages/onnxruntime/quantization/quant_utils.py)
```

**Line generating the error**
https://github.com/Xilinx/Vitis-AI/blob/bbccde60d062403fc8d764b9adb2055527097e75/src/vai_quantizer/vai_q_onnx/vai_q_onnx/calibrate.py#L26

Reference to correct function name (from onnxruntime repo) [here](https://github.com/microsoft/onnxruntime/blob/276e8733bda76e6e42845af1b795fcb8189aba96/onnxruntime/python/tools/quantization/quant_utils.py#L662)

ZEB1CLJ, 2023-09-27:
It seems multiple function imports are incompatible with `onnxruntime==1.16.0`.
Temporary fix: install `onnxruntime==1.14.0`

Zeosleus, 2024-04-23:
> It seems multiple function imports are incompatible with `onnxruntime==1.16.0`. Temporary fix: install `onnxruntime==1.14.0`

I can confirm this workaround.
Thanks @ZEB1CLJ  !

atoposyz, 2025-02-21:
Thanks! Your solution fixed my issue perfectly!