Xilinx/Vitis-AI #1344: DPU TRD build fails due to missing 'aie/ip/libadf.a'

danielstumpp, 2023-09-27:
I am attempting to build the DPU TRD for the vck190 using the provided Vitis flow. I have followed the README documentation in the TRD downloaded from [here](https://www.xilinx.com/bin/public/openDownload?filename=DPUCVDX8G_VAI_v3.0.tar.gz) and have also attempted to use the build flow provided in the WAA ResNet50 example [here](https://github.com/Xilinx/Vitis-AI/tree/master/examples/waa/apps/resnet50/build_flow/DPUCVDX8G_vck190).

I am using Vitis 2022.2 and Petalinux 2022.2 on Ubuntu 22.04. 

Both of the above methods result in an error along the lines of what is attached below. The apparent error is on the second to last line: `cp: cannot stat 'aie/ip/libadf.a': No such file or directory`

I have confirmed that the `libadf.a` file does not exist. Any ideas on the cause of this? 

```
+ export CUR_DIR=/home/dstumpp/Documents/Vitis-AI/examples/waa/apps/resnet50/build_flow/DPUCVDX8G_vck190
+ CUR_DIR=/home/dstumpp/Documents/Vitis-AI/examples/waa/apps/resnet50/build_flow/DPUCVDX8G_vck190
+ cd /home/dstumpp/Documents/Vitis-AI/examples/waa/apps/resnet50/build_flow/DPUCVDX8G_vck190/../../../../plugins/blobfromimage/pl
+ make cleanall
rm -rf build_dir.sw_emu.xilinx_vck190_base_202220_1/blobfromimage vitis_* TempConfig system_estimate.xtxt *.rpt .run/
rm -rf src/*.ll _xocc_* .Xil dltmp* xmltmp* *.log *.jou *.wcfg *.wdb sample_link.ini sample_compile.ini obj* bin* *.csv *.jpg *.jpeg
rm -rf build_dir.sw_emu.xilinx_vck190_base_202220_1/*.xclbin _vimage *xclbin.run_summary qemu-memory-_* emulation/ _vimage/ start_simulation.sh *.xclbin
rm -rf _x_temp.*/_x.* _x_temp.*/.Xil _x_temp.*/profile_summary.* xo_* _x*
rm -rf _x_temp.*/dltmp* _x_temp.*/kernel_info.dat _x_temp.*/*.log 
rm -rf _x_temp.* 
rm -rf build_dir.sw_emu.xilinx_vck190_base_202220_1 sd_card* build_dir.* emconfig.json *.html _x_temp.sw_emu.xilinx_vck190_base_202220_1 /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/reports *.csv *.run_summary /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/*.raw package_* run_script.sh .ipcache *.str /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/*.xo
rm -rf /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/Makefile/common/data/*.xe2xd* /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/Makefile/common/data/*.orig*
rm -rf  /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/Work /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/*.xpe /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/hw.o /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/*.xsa /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/xnwOut aiesimulator_output .AIE_SIM_CMD_LINE_OPTIONS
+ make xo TARGET=hw BLOB_CHANNEL_SWAP_EN=0 BLOB_CROP_EN=0 BLOB_LETTERBOX_EN=0 BLOB_JPEG_EN=0 BLOB_NPC=1
Compiling Kernel: blobfromimage_accel
mkdir -p _x_temp.hw.xilinx_vck190_base_202220_1
v++ -c  -t hw --platform /tools/Xilinx/Vitis/2022.2/base_platforms/xilinx_vck190_base_202220_1/xilinx_vck190_base_202220_1.xpfm --save-temps --optimize 2 --hls.jobs 8 --hls.clock 300000000:blobfromimage_accel -I/home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/../../include/pl/ -D_BLOB_CHANNEL_SWAP_EN_=0 -D_BLOB_JPEG_EN_=0  -D_BLOB_CROP_EN_=0 -D_BLOB_LETTERBOX_EN_=0 -D_BLOB_NPC_=1 -k blobfromimage_accel -I'/home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl' --temp_dir _x_temp.hw.xilinx_vck190_base_202220_1 --report_dir /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/reports/_x.hw.xilinx_vck190_base_202220_1 -o'_x_temp.hw.xilinx_vck190_base_202220_1/blobfromimage_accel.xo' /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/xf_blobfromimage_accel.cpp
Option Map File Used: '/tools/Xilinx/Vitis/2022.2/data/vitis/vpp/optMap.xml'

****** v++ v2022.2 (64-bit)
  **** SW Build 3671529 on 2022-10-13-17:52:11
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

INFO: [v++ 60-1306] Additional information associated with this v++ compile can be found at:
	Reports: /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/reports/_x.hw.xilinx_vck190_base_202220_1/blobfromimage_accel
	Log files: /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/_x_temp.hw.xilinx_vck190_base_202220_1/logs/blobfromimage_accel
Running Dispatch Server on port: 41693
INFO: [v++ 60-1548] Creating build summary session with primary output /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/_x_temp.hw.xilinx_vck190_base_202220_1/blobfromimage_accel.xo.compile_summary, at Wed Sep 27 13:10:34 2023
INFO: [v++ 60-1315] Creating rulecheck session with output '/home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/reports/_x.hw.xilinx_vck190_base_202220_1/blobfromimage_accel/v++_compile_blobfromimage_accel_guidance.html', at Wed Sep 27 13:10:34 2023
INFO: [v++ 60-895]   Target platform: /tools/Xilinx/Vitis/2022.2/base_platforms/xilinx_vck190_base_202220_1/xilinx_vck190_base_202220_1.xpfm
INFO: [v++ 60-1578]   This platform contains Xilinx Shell Archive '/tools/Xilinx/Vitis/2022.2/base_platforms/xilinx_vck190_base_202220_1/hw/hw.xsa'
INFO: [v++ 60-585] Compiling for hardware target
INFO: [v++ 60-423]   Target device: xilinx_vck190_base_202220_1
INFO: [v++ 60-242] Creating kernel: 'blobfromimage_accel'
INFO: [v++ 60-1616] Creating a HLS clock using hls.clock option: 300 MHz

===>The following messages were generated while  performing high-level synthesis for kernel: blobfromimage_accel Log file: /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/_x_temp.hw.xilinx_vck190_base_202220_1/blobfromimage_accel/blobfromimage_accel/vitis_hls.log :
INFO: [v++ 204-61] Pipelining loop 'MMIterInLoop1'.
INFO: [v++ 200-1470] Pipelining result : Target II = NA, Final II = 1, Depth = 3, loop 'MMIterInLoop1'
INFO: [v++ 204-61] Pipelining loop 'MMIterInLoopRow'.
INFO: [v++ 200-1470] Pipelining result : Target II = NA, Final II = 1, Depth = 6, loop 'MMIterInLoopRow'
INFO: [v++ 204-61] Pipelining loop 'VITIS_LOOP_339_1_VITIS_LOOP_344_2'.
INFO: [v++ 200-1470] Pipelining result : Target II = NA, Final II = 1, Depth = 3, loop 'VITIS_LOOP_339_1_VITIS_LOOP_344_2'
INFO: [v++ 204-61] Pipelining function 'scaleCompute<17, 42, 20, 48, 16, 1>'.
INFO: [v++ 200-1470] Pipelining result : Target II = NA, Final II = 1, Depth = 3, function 'scaleCompute<17, 42, 20, 48, 16, 1>'
INFO: [v++ 204-61] Pipelining loop 'VITIS_LOOP_396_5'.
INFO: [v++ 200-1470] Pipelining result : Target II = NA, Final II = 1, Depth = 15, loop 'VITIS_LOOP_396_5'
INFO: [v++ 204-61] Pipelining loop 'VITIS_LOOP_115_1'.
INFO: [v++ 200-1470] Pipelining result : Target II = 1, Final II = 1, Depth = 6, loop 'VITIS_LOOP_115_1'
INFO: [v++ 204-61] Pipelining loop 'VITIS_LOOP_49_1'.
INFO: [v++ 200-1470] Pipelining result : Target II = 1, Final II = 1, Depth = 4, loop 'VITIS_LOOP_49_1'
INFO: [v++ 204-61] Pipelining loop 'MMIterOutRow_MMIterOutCol'.
INFO: [v++ 200-1470] Pipelining result : Target II = NA, Final II = 1, Depth = 4, loop 'MMIterOutRow_MMIterOutCol'
INFO: [v++ 204-61] Pipelining loop 'MMIterOutLoop2'.
INFO: [v++ 200-1470] Pipelining result : Target II = NA, Final II = 1, Depth = 3, loop 'MMIterOutLoop2'
INFO: [v++ 200-789] **** Estimated Fmax: 406.01 MHz
INFO: [v++ 60-594] Finished kernel compilation
INFO: [v++ 60-244] Generating system estimate report...
INFO: [v++ 60-1092] Generated system estimate report: /home/dstumpp/Documents/Vitis-AI/examples/waa/plugins/blobfromimage/pl/reports/_x.hw.xilinx_vck190_base_202220_1/blobfromimage_accel/system_estimate_blobfromimage_accel.xtxt
INFO: [v++ 60-586] Created _x_temp.hw.xilinx_vck190_base_202220_1/blobfromimage_accel.xo
INFO: [v++ 60-2343] Use the vitis_analyzer tool to visualize and navigate the relevant reports. Run the following command. 
[... 81 lines cut ...]
INFO: [IP_Flow 19-4728] Bus Interface 'm_axi_aresetn': Added interface parameter 'POLARITY' with value 'ACTIVE_LOW'.
INFO: [IP_Flow 19-4728] Bus Interface 'interrupt': Added interface parameter 'SENSITIVITY' with value 'LEVEL_HIGH'.
INFO: [IP_Flow 19-818] Not transferring value dependency attribute "((spirit:decode(id('MODELPARAM_VALUE.ARCH_OCP')) = 16) ? 64 : 128)" into user parameter "SHIM_IFM_DATA_BW".
WARNING: [IP_Flow 19-991] Unrecognized or unsupported file 'src/fingerprint_json.ttcl' found in file group 'Synthesis'.
[... cut ...]
WARNING: [Vivado 12-12407] VLNV in kernel.xml does not match VLNV in any of the IPs specified with the ip_directory option: xilinx.com:ip:DPUCVDX8G:0.0
INFO: [Common 17-206] Exiting Vivado at Wed Sep 27 13:11:45 2023...
INFO:- - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - - -
INFO:Copying libadf.a...
cp: cannot stat 'aie/ip/libadf.a': No such file or directory
make: *** [Makefile:134: hw/binary_container_1/libadf.a] Error 1
```

quentonh, 2023-09-27:
I have seen this before and was trying to help the user resolve but ultimately did not get their feedback.  

[image]

ABS_PATH is defined by the current working directory in which the \vitis_prj\makefile is launched.

I would be inclined to go to DPUCVDX8G_VAI_v3.0\vitis_prj and then execute "make aie" to see what happens.

danielstumpp, 2023-09-27:
Hi @quentonh, thanks for that suggestion. Running `make aie` revealed that I did not have the appropriate license file. I had to generate and install the license for AIE compilation (as directed in https://support.xilinx.com/s/article/76792?language=en_US).

After that, the build resulted in this error: `nvalid device name xcvc1902-vsva2197-1LP-e-S-es1 for NOC/AIE model creation. Please swithc to using HDARPart to create the HIPDevice.` I resolved that by updating the aiecompiler platform as noted in this post: https://support.xilinx.com/s/question/0D54U00006qVqWkSAK/cannot-integrate-dpu-into-vck190-critical-warning-aiecompiler-75674-the-use-of-device-name-xcvc-to-create-hipdevice-is-going-to-be-deprecated?language=en_US

Building the rest of the design now. Will close this when I confirm a successful build. Thanks for the help!

quentonh, 2023-09-27:
@danielstumpp Oof!  OK, it seems like we have to update the tutorial archive to fix that problem.  Thanks for highlighting it.  I will look at modifying the script and get a new version uploaded.

danielstumpp, 2023-09-29:
Ok, I reference design successfully built. One other note is that the PLATFORM definition in the makefile was incorrect in the downloaded TRD. When running the platform build the actual output location is `$TRD_HOME/vck190_platform/platforms/xilinx_vck190_mipiRxSingle_hdmiTx_202220_1/vck190_mipiRxSingle_hdmiTx.xpfm`

Thanks for the assistance @quentonh

closed 2023-09-29 by danielstumpp