Xilinx/Vitis-AI #1350: ./docker_build.sh -t gpu -f pytorch failed.

MiaoHe0306, 2023-10-15:
I try to install vitis-ai in my Ubuntu 20.04, and I use kv260. But every time I do "./docker_build.sh -t gpu -f pytorch", I got this error "ERROR: failed to solve: process "/bin/bash -c if [[ -n \"${TARGET_FRAMEWORK}\" ]]; then  bash ./install_${TARGET_FRAMEWORK}.sh; fi" did not complete successfully: exit code: 1"

Did anyone meet this problem and know how to solve it?

quentonh, 2023-11-01:
@MiaoHe0306 I don't see anything wrong with the formatting of your command.  Have you confirmed that bash is your default shell?

jonGuti13, 2024-02-06:
@ElonGUI @MiaoHe0306 Did you manage to solve it? I have the same problem for TF2 now even though in another computer I managed to build the image 2 months ago, so something must have changed.