Xilinx/Vitis-AI #1353: zcu102 burning image startup failed

zhangapeng, 2023-10-16:
The SD card cannot start after burning the zcu102 onboard image given in vitis-ai. I confirmed that I have set the SD card startup mode, and I tried the 2023.1 image burning given by the wiki official website and it was successful. This shows that There is no problem with my board. I also tried different versions of the images provided by the vitis-ai library, but they couldn’t be started after burning them.

lishixlnx, 2023-10-26:
If all images from vitis-ai-library fails, then maybe something is wrong with the board. 
Otherwise, all customer will complain.

quentonh, 2023-11-01:
@zhangapeng what is the failure mode when you attempt to boot your board from the pre-built Vitis AI 3.0 image for the ZCU102? 

https://www.xilinx.com/member/forms/download/design-license-xef.html?filename=xilinx-zcu102-dpu-v2022.2-v3.0.0.img.gz

Please share any terminal output that can be captured during the boot process.  Also, please confirm the specific board revision that you are using:

https://support.xilinx.com/s/article/71961?language=en_US

zhangapeng, 2023-11-03:
@quentonh When I use balenaEtcher to burn the image img provided by vitis-ai to the SD card, after powering on, the serial port just outputs "xilinx zynq mp first stage boot loader, Release 2022.0
Dec 15 2022-22:48:43". I also tried to flash different versions of vitis-ai images, but the output was the same. I suspected that there was a problem with my board, but I found a flash on the official wiki website After writing, the serial port generates an image of the petalinux terminal, https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/2654765057/2023.1+Release. This shows that there is no problem with my board. My board model is HW- Z1-ZCU102 REVISION1.1

zhangapeng, 2023-11-03:
@lishixlnx I also suspected that there was a problem with my board, but I found an image on the official wiki website that generated a petalinux terminal through the serial port after programming, https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/2654765057/ 2023.1+Release. This means there is nothing wrong with my board.

zhangapeng, 2023-11-03:
I just wanted to run quick start based on the official image and tutorial, but I failed in the first step. I was very frustrated because I didn't know what went wrong. Is there something wrong with my board or do I need to set something up. I followed the tutorial exactly.