Xilinx/Vitis-AI #1358: Vitis-AI Assertion Error (pytorch)

wilfredkisku, 2023-10-23:
I am training and compiling the Resnet50 model using ```pytorch 1.13.1```. I am able to obtain the ```xmodel``` after quantization and compilation but while running the model on the hardware I get the resultant error of AssertionError.

```
I20231023 11:48:41.588702   116 compile_pass_manager.cpp:352] [UNILOG][INFO] Compile mode: dpu
I20231023 11:48:41.588901   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:41.589030   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:41.589396   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_nndct_relu_VLxS1y2CcsMpr5dJ, with op num: 9
I20231023 11:48:41.589457   116 compile_pass_manager.cpp:478] [UNILOG][INFO] Begin to compile...
I20231023 11:48:41.786698   116 compile_pass_manager.cpp:489] [UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
I20231023 11:48:41.786796   116 compile_pass_manager.cpp:504] [UNILOG][INFO] Compile done.
I20231023 11:48:41.805989   116 compile_pass_manager.cpp:352] [UNILOG][INFO] Compile mode: dpu
I20231023 11:48:41.806100   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:41.806129   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:41.806367   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_nndct_relu_43I6OX57yrbj9Ynl, with op num: 9
I20231023 11:48:41.806412   116 compile_pass_manager.cpp:478] [UNILOG][INFO] Begin to compile...
I20231023 11:48:41.881876   116 compile_pass_manager.cpp:489] [UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
I20231023 11:48:41.881914   116 compile_pass_manager.cpp:504] [UNILOG][INFO] Compile done.
I20231023 11:48:41.886736   116 compile_pass_manager.cpp:352] [UNILOG][INFO] Compile mode: dpu
I20231023 11:48:41.886767   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:41.886780   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:41.886844   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_nndct_relu_cKxi3t8kWa74dnms, with op num: 9
I20231023 11:48:41.886852   116 compile_pass_manager.cpp:478] [UNILOG][INFO] Begin to compile...
I20231023 11:48:41.895570   116 compile_pass_manager.cpp:489] [UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
I20231023 11:48:41.895675   116 compile_pass_manager.cpp:504] [UNILOG][INFO] Compile done.
I20231023 11:48:41.902637   116 compile_pass_manager.cpp:352] [UNILOG][INFO] Compile mode: dpu
I20231023 11:48:41.902671   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:41.902679   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:41.902751   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_nndct_relu_IHPg2Qk561yv4OKf, with op num: 9
I20231023 11:48:41.902760   116 compile_pass_manager.cpp:478] [UNILOG][INFO] Begin to compile...
I20231023 11:48:41.927364   116 compile_pass_manager.cpp:489] [UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
I20231023 11:48:41.927435   116 compile_pass_manager.cpp:504] [UNILOG][INFO] Compile done.
I20231023 11:48:41.932706   116 compile_pass_manager.cpp:352] [UNILOG][INFO] Compile mode: dpu
I20231023 11:48:41.932730   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:41.932736   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:41.932793   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_nndct_relu_BpSwtqRoWxXz1vdL, with op num: 9
I20231023 11:48:41.932801   116 compile_pass_manager.cpp:478] [UNILOG][INFO] Begin to compile...
I20231023 11:48:41.937515   116 compile_pass_manager.cpp:489] [UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
I20231023 11:48:41.937546   116 compile_pass_manager.cpp:504] [UNILOG][INFO] Compile done.
I20231023 11:48:41.942870   116 compile_pass_manager.cpp:352] [UNILOG][INFO] Compile mode: dpu
I20231023 11:48:41.942906   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:41.942921   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:41.943043   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_nndct_relu_Upt5HjRGdCFkXiLx, with op num: 9
I20231023 11:48:41.943059   116 compile_pass_manager.cpp:478] [UNILOG][INFO] Begin to compile...
I20231023 11:48:41.952984   116 compile_pass_manager.cpp:489] [UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
I20231023 11:48:41.953132   116 compile_pass_manager.cpp:504] [UNILOG][INFO] Compile done.
I20231023 11:48:41.972256   116 compile_pass_manager.cpp:352] [UNILOG][INFO] Compile mode: dpu
I20231023 11:48:41.972330   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:41.972337   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:41.972397   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_nndct_relu_iVtHb582EQwFRmDW, with op num: 9
I20231023 11:48:41.972404   116 compile_pass_manager.cpp:478] [UNILOG][INFO] Begin to compile...
I20231023 11:48:42.104565   116 compile_pass_manager.cpp:489] [UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
I20231023 11:48:42.104604   116 compile_pass_manager.cpp:504] [UNILOG][INFO] Compile done.
I20231023 11:48:42.116638   116 compile_pass_manager.cpp:352] [UNILOG][INFO] Compile mode: dpu
I20231023 11:48:42.116672   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:42.116684   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:42.116768   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_nndct_relu_bwSCk8tHGsZMqcuP, with op num: 9
I20231023 11:48:42.116779   116 compile_pass_manager.cpp:478] [UNILOG][INFO] Begin to compile...
I20231023 11:48:42.135566   116 compile_pass_manager.cpp:489] [UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
I20231023 11:48:42.135604   116 compile_pass_manager.cpp:504] [UNILOG][INFO] Compile done.
I20231023 11:48:42.147694   116 compile_pass_manager.cpp:352] [UNILOG][INFO] Compile mode: dpu
I20231023 11:48:42.147728   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:42.147739   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:42.147819   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_nndct_relu_0TcliqtnuGQ1fWF3, with op num: 9
I20231023 11:48:42.147833   116 compile_pass_manager.cpp:478] [UNILOG][INFO] Begin to compile...
I20231023 11:48:42.164856   116 compile_pass_manager.cpp:489] [UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
I20231023 11:48:42.164909   116 compile_pass_manager.cpp:504] [UNILOG][INFO] Compile done.
I20231023 11:48:42.170341   116 compile_pass_manager.cpp:352] [UNILOG][INFO] Compile mode: dpu
I20231023 11:48:42.170374   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:42.170388   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:42.170464   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_nndct_relu_3VQPTvf1BrAKZi7Y, with op num: 9
[... 439 lines cut ...]
I20231023 11:48:44.203816   116 compile_pass_manager.cpp:353] [UNILOG][INFO] Debug mode: null
I20231023 11:48:44.203822   116 compile_pass_manager.cpp:357] [UNILOG][INFO] Target architecture: DPUCZDX8G_ISA1_B4096
I20231023 11:48:44.203879   116 compile_pass_manager.cpp:465] [UNILOG][INFO] Graph name: nndct_conv2d_iTGMctlVNvja2Qh3, with op num: 8
I20231023 11:48:44.203886   116 compile_pass_manager.cpp:478] [UNILOG][INFO] Begin to compile...
I20231023 11:48:44.220065   116 compile_pass_manager.cpp:489] [UNILOG][INFO] Total device subgraph number 3, DPU subgraph number 1
I20231023 11:48:44.220105   116 compile_pass_manager.cpp:504] [UNILOG][INFO] Compile done.
```
The error on while loading it onto the hardware (using python APIs).

```
---------------------------------------------------------------------------
[... cut ...]
/usr/local/share/pynq-venv/lib/python3.8/site-packages/pynq_dpu/dpu.py in load_model(self, model)
    169             self.graph = xir.Graph.deserialize(abs_model)
    170             subgraphs = get_child_subgraph_dpu(self.graph)
--> 171             assert len(subgraphs) == 1
    172             self.runner = vart.Runner.create_runner(subgraphs[0], "run")

AssertionError: 
```

Is there anything wrong that I am doing according to the logs, I was unable to figure out any. Any help would be great. Thanks!

lishixlnx, 2023-10-25:
from the above log info, you can see you get a lot of dpu subgraphs. 
But this sample code need meet the condition that dpu subgraph number is 1.  
So error occurs.  Please open the compiled xmodel with netron to check the cpu subgraph and find what op is not supported by dpu.  then try to replace these ops by the supported ops.

wilfredkisku, 2023-10-25:
This is the snippet of the netron generated graph. You are right (about the subgraphs generated). But I tried to create a single subgraph for a ResNet Model. I think I will have to simplify the network code for the model.

I am not sure how multiple subgraphs are getting generated.

[image]

Thank you for the help.