Xilinx/Vitis-AI #1362: open sourceing xcompiler package?

t-kuha, 2023-10-29:
Is is possible to make xcompiler package (now distributed in binary only) open source?

quentonh, 2023-11-01:
@t-kuha We have considered this.  Please tell us more about your use case and share additional motivations as to why we should evaluate this further.

t-kuha, 2023-11-01:
@quentonh 
Thanks for your reply.
I want to run the Vitis AI Quantizer on a ARM64 Linux PC.
Moreover, current Python version requirement (3.6 & 3.8 I think) is a bit prohibitive for developers who are using newer version of Python.

quentonh, 2023-11-01:
@t-kuha Thanks for your inputs.  Just to confirm, you mean that you want to run the compiler on an ARM64 Linux PC (the Quantizer is already open-source)?

t-kuha, 2023-11-01:
@quentonh 
Yes, I want use ARM 64 Linux machine (, which tends to be more power efficient than x84/64 one).
And, yes, I know the Quantizer is open source, but some parts of it (like [here](https://github.com/Xilinx/Vitis-AI/blob/bbccde60d062403fc8d764b9adb2055527097e75/src/vai_optimizer/nndct_shared/inspector/dpu_pattern_transform.py#L262) and [here](https://github.com/Xilinx/Vitis-AI/blob/bbccde60d062403fc8d764b9adb2055527097e75/src/vai_optimizer/nndct_shared/compile/compiler_factory.py#L27)) seems to depend on _xcompiler_.

quentonh, 2023-11-01:
@t-kuha I see.....yes, there are indeed hooks into xcompiler.

We will consider this further internally.  The major concern is the potential support burden, but there are good reasons for us to expose the source for this tool as well.

t-kuha, 2023-11-01:
@quentonh 
Thank you for the prompt responses.
I now understand your situation and am looking forward to the next release!