Xilinx/Vitis-AI #1363: irps5410 error

FBlueN, 2023-10-31:
I have built and ran the tutorial with no modification successfully but when using Petalinux to configure the Common Software the dpu_sw_optimize.sh fails with:

Auto resize ext4 partition ...[✔]
Start QoS config ...[✔]
Config PMIC irps5401 .../dev/i2c-4: Device or resource busy
Couldn't attach to device 0x43

I included the i2ctool in my Petalinux build but this still errors out. I am using the same base platform as the tutorial.
I have see the problem, but I don't see the solution.

Vivado verision : 2022.2
Broad : zcu104
vitis-ai : 3.0

quentonh, 2023-11-01:
@FBlueN I am of the understanding that dpu_sw_optimize.sh was developed prior to the inclusion of an IRPS5401 driver in the Linux upstream.  The dpu_sw_optimize implementation leverages IOCTL access to the I2C device which does not function if the IRPS5401 driver has been probed which binds the upstream driver to the hardware.

A potential work-around to this is to disable CONFIG_SENSORS_IRPS5401.  If CONFIG_SENSORS_IRPS5401=y the hwmon/pmbus/ir5401.c is probed and the user space dpu_sw_optimize application will fail.

BTW, the sole purpose of this application is to modify the maximum output power available from the regulator subsystem on the ZCU104.  Without these modifications, the maximum power available to the ZU+ device is insufficient to run some (many) models with multiple threads.

FBlueN, 2023-11-02:
> @FBlueN I am of the understanding that dpu_sw_optimize.sh was developed prior to the inclusion of an IRPS5401 driver in the Linux upstream. The dpu_sw_optimize implementation leverages IOCTL access to the I2C device. IOCTL cannot read and write to this hardware device if the IRPS5401 driver has been probed and bound to the hardware instance. The upstream 5401 driver currently does not have the hooks to read/write the necessary registers in the device, so it is not straightforward to modify dpu_sw_optimize to simply leverage the IRPS5401 driver.
> 
> A potential work-around to this is to disable CONFIG_SENSORS_IRPS5401. If CONFIG_SENSORS_IRPS5401=y the hwmon/pmbus/ir5401.c is probed and the user space dpu_sw_optimize application will fail.
> 
> BTW, the sole purpose of this application is to modify the maximum output power available from the regulator subsystem on the ZCU104. Without these modifications, the maximum power available to the ZU+ device is insufficient to run some (many) models with multiple threads. Given that our use of dpu_sw_optimize is purely for ZCU104 max power adjustment, we have not ourselves invested in modifying the upstream driver to support the required register accesses.

I understand what do you mean.However when I try to disable CONFIG_SENSORS_IRPS5401, I didn't find how to do.So can you teach me how to disable CONFIG_SENSORS_IRPS5401.Thank you.

FBlueN, 2023-11-02:
> @FBlueN I am of the understanding that dpu_sw_optimize.sh was developed prior to the inclusion of an IRPS5401 driver in the Linux upstream. The dpu_sw_optimize implementation leverages IOCTL access [...]

Hey, I have solved this problem. Thank you.

quentonh, 2023-11-03:
Thanks @FBlueN we are pleased to hear this.  We will close out this issue but please reopen if needed.