Xilinx/Vitis-AI #1368: [UNILOG][FATAL][XCOM_ACGEN_UNSUPPORT_QUANTIZATION][unsupport quantization bit shift while assembly code generation.] Model_Model_Detect_model_Detect_77_ret_335_fix_FixShifterprecheck fail the delta of fix_point between input and output is over bit width fix_point_min of input is7, output fix point is -2

alvarffff, 2023-11-07:
Hi, the main problem is that when Im trying to compile the quantized xmodel with vai_c_xir for kv260 im getting this error:

[image]

Context:

I have trained a model using tiny yolo from ( https://github.com/WongKinYiu/yolov7.git ) with a custom dataset with only 1 class. The training command looks like this:

`python train.py --batch-size 32 --epochs 100 --img 320 320 --data license_data/license_plates.yaml --hyp data/hyp.scratch.tiny.yaml --cfg cfg/deploy/yolov7-tiny-custom.yaml --weights yolov7-tiny.pt --name yolov7-tiny-custom` 

When I try to quantize it, I got the following warnings 
[image]

The problem of this is that I dont know where I am using the atten::pow, the nndct_strided_slice_inplace_copy and the nndct_select. If I inspect the quantized model using netron app looks like some nodes are fixed (maybe because the quantizer nor the xir recognises some layers). 

My quantization script looks like this 

`import torch
from models.experimental import attempt_load
from models.yolo import Model
import yaml
from torch.utils.data import Dataset
from torchvision import transforms
from PIL import Image
from pytorch_nndct.apis import torch_quantizer, dump_xmodel
import argparse
import os
import sys
from tqdm import tqdm
import copy

class LicensePlatesDataset(Dataset):
    def __init__(self, root_dir, transform=None):
        self.root_dir = root_dir
        self.transform = transform
        self.image_paths = sorted(os.listdir(os.path.join(root_dir, 'images', 'val')))
        self.label_paths = sorted(os.listdir(os.path.join(root_dir, 'labels', 'val')))

    def __len__(self):
        return len(self.image_paths)

    def __getitem__(self, idx):
        img_name = os.path.join(self.root_dir, 'images', 'val', self.image_paths[idx])
        label_name = os.path.join(self.root_dir, 'labels', 'val', self.label_paths[idx])

        image = Image.open(img_name).convert("RGB")
        with open(label_name, "r") as file:
            # Assuming label file contains bounding box coordinates and class information
            # Adjust this part based on your label format
            label_info = file.readline().strip().split()
            boxes = torch.tensor([float(coord) for coord in label_info[1:4]], dtype=torch.float32)
            class_label = int(label_info[0])  # Assuming class label is an integer

        if self.transform:
            image = self.transform(image)

        return image, class_label

def main():
    ap = argparse.ArgumentParser()
    ap.add_argument('-q',  '--quant_mode',  type=str, default='calib',    help='Path to build folder. Default is build')
    args = ap.parse_args()
    quant_mode = args.quant_mode
    device = torch.device('cpu')
    batchsize = 8

    # Hyperparameters
    with open('./data/hyp.scratch.tiny.yaml') as f:
        hyp = yaml.load(f, Loader=yaml.SafeLoader)  # load hyps

    model = Model('./cfg/deploy/yolov7-tiny-custom.yaml', ch=3, nc=1, anchors=hyp.get('anchors')).to(device)
    
    checkpoint = torch.load('best.pt', map_location=device)

    for key in checkpoint.keys():
        print(key)
    # Eliminar claves no deseadas
    # unexpected_keys = ['epoch', 'best_fitness', 'training_results', 'ema', 'updates', 'optimizer', 'wandb_id']
    # for key in unexpected_keys:
    #     del checkpoint[key]

    # Cargar el modelo
    model.load_state_dict(checkpoint, strict=False)   

    if(quant_mode == 'test'):
        batchsize = 1

    rand_in = torch.randn([batchsize, 3, 320, 320])
    print("Torch_quantizer")
    quantizer = torch_quantizer(quant_mode, model, rand_in, output_dir="./prueba") 
    quantized_model = quantizer.quant_model

    test_transform = transforms.Compose([
        transforms.Resize((320, 320)),
        transforms.ToTensor(),
    ])

    dataset_root = 'license_data'
    # data loader
    test_dataset = LicensePlatesDataset(root_dir=dataset_root, transform=test_transform)
    
    test_loader = torch.utils.data.DataLoader(test_dataset,
                                           batch_size=batchsize, 
                                           shuffle=False)

    quantized_model.eval()
    #model = model.to(device)
   
    # Realizar inferencia en datos de prueba
    for iteraction, (inputs, labels) in tqdm(
      enumerate(test_loader), total=len(test_loader)):
        inputs = inputs.to(device)
        labels = labels.to(device)
        outputs = quantized_model(inputs)
       

    print("Exporting...")
    # export config
    if quant_mode == 'calib':
        quantizer.export_quant_config()
    if quant_mode == 'test':
        quantizer.export_xmodel(deploy_check=False, output_dir="./prueba")

if __name__ == '__main__':
    main()`

alvarffff, 2023-11-07:
I have run the Inspector on the float model for the DPU model. Attached 
[inspect_DPUCZDX8G_ISA1_B4096.txt](https://github.com/Xilinx/Vitis-AI/files/13290354/inspect_DPUCZDX8G_ISA1_B4096.txt)

ThomasMeng0515, 2023-11-13:
Try to remove the postprocessing part out of quantizer. You can try to use the following example to avoid postprocess part input to the quantizer.

alvarffff, 2023-11-13:
> Try to remove the postprocessing part out of quantizer. You can try to use the following example to avoid postprocess part input to the quantizer.

Hi Thomas, thank you for your response but I think you forgot to add the example.

ThomasMeng0515, 2023-11-20:
@alvarffff Sorry for the mistake. Here is the official user guide link for quant and dequant operation:
https://docs.xilinx.com/r/en-US/ug1414-vitis-ai/Module-Partial-Quantization