Xilinx/Vitis-AI #1378: nvcc: command not found (Vitis-AI-3.0)

A7eNg, 2023-12-06:
After building and entering docker, met this error 
`nvcc: command not found`

[image]
 And there is no bin file folder in path: /usr/local/cuda-11.3/
[image]

How to fix it in Vitis-AI-3.0? I have not met this problem in version 3.5, but i am using MPSoC, so whether 3.5 can be used for MPSoC devices？Or I must change to 3.0?

Thanks for your kind reply!

lishixlnx, 2023-12-11:
Yes, you must change to v3.0.  
MPSoC is only supported to v3.0.
If you want to use cuda, can you try to install the missing part manually?

A7eNg, 2023-12-11:
Thanks for your comment, should I use "opt_pytorch" for the target framework?

cchalou98, 2023-12-19:
Hello, please did you solve this issue? cuz am having the same issue? I am using Vitis 3.0 to quantize yolov5, Ihave GPU NVIDIA GeForce RTX 4060, and am using CUDA 11.3.1 in vitis?
[image]

A7eNg, 2023-12-20:
Yeah, I have solved this issue by installing CUDA 11.3 and cudnn manually

cchalou98, 2023-12-20:
> Yeah, I have solved this issue by installing CUDA 11.3 and cudnn manually

Could you please share with me the command lines how you installed it?? much appreciated, thank you.