Xilinx/Vitis-AI #1385: Generatin Xmodel from onnx model directly

shaibal13, 2023-12-24:
Hello,
I have a quantized model using torch quantization library and exported it in onnx format.
Later on I run `vai_c_xir -x model.onnx -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/ZCU104/arch.json` to generate the xmodel. I can't view that xmodel with netron but I can view onnx model .it gives me below error. Can anyone give me how to verify that xmodel generation is right or not?
[image]

ahnhy1324, 2024-01-17:
try this on your board
xdputil benchmark <xmodel> <-i subgraph_index> <num_of_threads>
if it works, your xmodel is excutable for your board
and check this link
https://docs.xilinx.com/r/en-US/ug1354-xilinx-ai-sdk/Using-the-xdputil-Tool

sandyagur8, 2024-11-05:
@shaibal13 @ahnhy1324  I am having the same issue . Have you found the solution?