Xilinx/Vitis-AI #1394: vaitrace not generating the xrt.run_summary and summary.csv files

Shreyas-NR, 2024-01-18:
Hi,
I'm trying to run Vitis AI trace on the resnet50 application, but for some reason, there is no **summary.csv** and **xrt.run_summary** files generated.
I'm using Vivado flow, DPU IPv 4.1, Petalinux-2022.1, and updated Vitis AI library to 3.5 from the latest recipes.

```
root@xilinx-kv260-starterkit-20222:~# xdputil query
{
    "DPU IP Spec":{
        "DPU Core Count":2,
        "IP version":"v4.1.0",
        "enable softmax":"False"
    },
    "VAI Version":{
        "libvart-runner.so":"Xilinx vart-runner Version: 3.5.0-b7953a2a9f60e23efdfced5c186328dd1449665c  2024-01-10-05:17:16 ",
        "libvitis_ai_library-dpu_task.so":"Advanced Micro Devices vitis_ai_library dpu_task Version: 3.5.0-b7953a2a9f60e23efdfced5c186328dd1449665c  2023-06-29 03:20:28 [UTC] ",
        "libxir.so":"Xilinx xir Version: xir-b7953a2a9f60e23efdfced5c186328dd1449665c 2024-01-10-05:16:38",
        "target_factory":"target-factory.3.5.0 b7953a2a9f60e23efdfced5c186328dd1449665c"
    },
    "kernels":[
        {
            "DPU Arch":"DPUCZDX8G_ISA1_B1600",
            "DPU Frequency (MHz)":325,
            "XRT Frequency (MHz)":100,
            "cu_idx":0,
            "fingerprint":"0x101000056010404",
            "is_vivado_flow":true,
            "name":"DPU Core 0"
        },
        {
            "DPU Arch":"DPUCZDX8G_ISA1_B1600",
            "DPU Frequency (MHz)":325,
            "XRT Frequency (MHz)":100,
            "cu_idx":1,
            "fingerprint":"0x101000056010404",
            "is_vivado_flow":true,
            "name":"DPU Core 1"
        }
    ]
}
```
```
root@xilinx-kv260-starterkit-20222:~# vart_version
libxir.so.3: Xilinx xir Version: xir-b7953a2a9f60e23efdfced5c186328dd1449665c 2024-01-10-05:16:38
libvart-buffer-object.so.3: Xilinx vart-buffer-object Version: 3.5.0-b7953a2a9f60e23efdfced5c186328dd1449665c  2024-01-10-05:17:16
libvart-dpu-controller.so.3: Xilinx vart-dpu-controller Version: 3.5.0-b7953a2a9f60e23efdfced5c186328dd1449665c  2024-01-10-05:17:16
libvart-dpu-runner.so.3: Xilinx vart-dpu-runner Version: 3.5.0-b7953a2a9f60e23efdfced5c186328dd1449665c  2024-01-10-05:17:16
libvart-mem-manager.so.3: Xilinx vart-mem-manager Version: 3.5.0-b7953a2a9f60e23efdfced5c186328dd1449665c  2024-01-10-05:17:16
libvart-runner.so.3: Xilinx vart-runner Version: 3.5.0-b7953a2a9f60e23efdfced5c186328dd1449665c  2024-01-10-05:17:16
libvart-util.so.3: Xilinx vart-util Version: 3.5.0-b7953a2a9f60e23efdfced5c186328dd1449665c  2024-01-10-05:17:16
libvart-xrt-device-handle.so.3: Xilinx vart-xrt-device-handle Version: 3.5.0-b7953a2a9f60e23efdfced5c186328dd1449665c  2024-01-10-05:17:16
```
```
root@xilinx-kv260-starterkit-20222:~#vaitrace -v
vaitrace ver:
3.5.0_b7953a2a9_2024-01-10-05:17:16
Project:
vart
Internal Git Ver:
[671d17e, -]

```

I tried to debug using **strace** on the vaitrace process ( log file attached below), from the log I don't see a file operation on summary.csv begin done.
 
A snapshot of the log is here,

[image]

Further debug shows that only profile_summary.csv, vart_trace.csv, vitis_ai_profile.csv are written from the [source code from the vitis AI repo](https://github.com/Xilinx/Vitis-AI/blob/master/src/vai_runtime/vart/trace/vaitrace/writer/vtf/convert.py#L376) 

```
def output(OUTPUT_PATH, options):
    output_vart_trace(OUTPUT_PATH)
    output_nmu_trace(OUTPUT_PATH, options)
    output_profile_summary(OUTPUT_PATH)
    output_vitis_ai_profile(OUTPUT_PATH)
```

**Which part of the code writes summary.csv and xrt.run_summary?**

Can anyone please help me to get the summary.csv and xrt.run_summary?

The console log of vaitrace with debug, B1600_2 core dtbo and bin files, python application, and compiled models are attached below.
[B1600_2.zip](https://github.com/Xilinx/Vitis-AI/files/13982664/B1600_2.zip)
[python.zip](https://github.com/Xilinx/Vitis-AI/files/13982665/python.zip)
[vaitrace_consoleLOG.txt](https://github.com/Xilinx/Vitis-AI/files/13982666/vaitrace_consoleLOG.txt)
[strace_vaitraceLOG.txt](https://github.com/Xilinx/Vitis-AI/files/13982669/strace_vaitraceLOG.txt)

Shreyas-NR, 2024-01-22:
Hi,
Update on my debugging progress,

1. I tried to use the pre-built image for KV260 that comes with a single core B4096 arch DPU.
2. vaitrace work on this and all the csv and xrt.run_summary is generated.
3. I did stack trace of the vaitrace process to get more information and to know which file is creating the summary.csv and xrt.run_summary.
4. Observing, I see just before writing the summary.csv file libxdp_core.so.2 is opened. See below image
[image]

Using the strings command on the library to check it it is built to create summary file turned out to be true
```
root@xilinx-kv260-starterkit-20222:/usr/bin#  strings /usr/lib/libxdp_core.so.2|grep summary
summary.csv
xrt.run_summary
Hardware counters added to summary file
Collection of data from PL monitors and added to summary and trace
Generation of Vitis AI summary and trace (Vitis AI designs only)
Cannot create profile summary checkpoint file
root@xilinx-kv260-starterkit-20222:/usr/bin#
```

This libxdp_core is from XRT, 
[https://github.com/Xilinx/XRT xrt-git](https://github.com/Xilinx/XRT xrt-git)

After doing dnf install xrt,

Still the summary and xrt_run_summary files are not generated.

Can anyone help me?

Best regards,
Shreyas NR

MRFiruzabadi, 2024-09-02:
hi @Shreyas-NR,
I'm also having the same issue, Have you found a solution yet?

Shreyas-NR, 2024-09-03:
Hey,
I couldn't get this working. I had to do some manual patching to get the
trace waveform.
But I remember reading somewhere that vaitrace is not supported for vivado
flow.