Xilinx/Vitis-AI #1399: vitis-ai-library_3.0.bb:do_compile) failed with exit code '1'

heroamd, 2024-01-25:
I am tring to integrate DPU4.1 to ZYNQ-Z2 board. So I choose Vivado 2022.2 and PetaLinux 2022.2 to generate linux image for SD card.
but when I following[ THIS TUTORIAL](https://github.com/Xilinx/Vitis-AI/tree/3.0/src/vai_petalinux_recipes), trying to add VISTIS-AI driver into the rootfs. I am sure each step is same with the README.MD. But I still meet trouble:

> ERROR: Task (/home/xilinx/dpu_pynq_z2/project-spec/meta-user/recipes-vitis-ai/vitis-ai-library/vitis-ai-library_3.0.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 6198 tasks of which 6189 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/xilinx/dpu_pynq_z2/project-spec/meta-user/recipes-vitis-ai/vitis-ai-library/vitis-ai-library_3.0.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
ERROR: Failed to build project. Check the /home/xilinx/dpu_pynq_z2/build/build.log file for more details...

and the content in build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/vitis-ai-library/3.0-r0/temp/log.do_compile is: numorous such this error. The whole file has been uploaded.

> /home/xilinx/dpu_pynq_z2/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/vitis-ai-library/3.0-r0/recipe-sysroot-native/usr/lib/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/11.2.0/include/arm_neon.h:4403:50: error: argument 2 must be a constant immediate
 4403 |   return (uint8x16_t)__builtin_neon_vshru_nv16qi ((int8x16_t) __a, __b);
      |                      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In function 'uint8x16_t vshrq_n_u8(uint8x16_t, int)',
    inlined from 'vitis::ai::transform_mean_scale(int, int, const uint8_t*, int8_t*, const std::vector<float>&, const std::vector<int>&, bool)::<lambda(const uint8x16_t&, int, const uint8x16_t&, const uint8x16_t&)>' at /home/xilinx/dpu_pynq_z2/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/vitis-ai-library/3.0-r0/git/src/vai_library/math/test/../src/image_util.cpp:554:33,
    inlined from 'vitis::ai::transform_mean_scale(int, int, const uint8_t*, int8_t*, const std::vector<float>&, const std::vector<int>&, bool)::<lambda(const uint8x16_t&, int, const uint8x16_t&, const uint8x16_t&)>' at /home/xilinx/dpu_pynq_z2/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/vitis-ai-library/3.0-r0/git/src/vai_library/math/test/../src/image_util.cpp:542:26,
    inlined from 'void vitis::ai::transform_mean_scale(int, int, const uint8_t*, int8_t*, const std::vector<float>&, const std::vector<int>&, bool)' at /home/xilinx/dpu_pynq_z2/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/vitis-ai-library/3.0-r0/git/src/vai_library/math/test/../src/image_util.cpp:608:32:
/home/xilinx/dpu_pynq_z2/build/tmp/work/cortexa9t2hf-neon-xilinx-linux-gnueabi/vitis-ai-library/3.0-r0/recipe-sysroot-native/usr/lib/arm-xilinx-linux-gnueabi/gcc/arm-xilinx-linux-gnueabi/11.2.0/include/arm_neon.h:4403:50: error: argument 2 must be a constant immediate

Then How can i solve this problem?
[log.do_compile.txt](https://github.com/Xilinx/Vitis-AI/files/14048772/log.do_compile.txt)

quentonh, 2024-01-25:
@heroamd We have internally built Vitis AI 3.0 for Zynq-7000, and yes, changes to the NEON code are required.  There are differences in the supported NEON intrinsics for the A9 versus the A53.  If you want to reach out to me directly (quentonh@amd.com) I believe that we can provide you a workaround that you can test.  We have done some degree of validation on these code changes, but the fix has not been fully validated.  quentonh@ A M D.com

heroamd, 2024-01-25:
> @heroamd We have internally built Vitis AI 3.0 for Zynq-7000, and yes, changes to the NEON code are required. There are differences in the supported NEON intrinsics for the A9 versus the A53. If you want to reach out to me directly ([quentonh@amd.com](mailto:quentonh@amd.com)) I believe that we can provide you a workaround that you can test. We have done some degree of validation on these code changes, but the fix has not been fully validated. quentonh@ A M D.com

Thank you!!!!

duh1duh, 2024-06-13:
Hello 
I run into the same problem. Can you share with me please your work around the issue ? 
Thank you.

heroamd, 2024-06-16:
Hello,
I had try to use the new IP or other files. But it still didn't work. I had made real so much more modification for the files step-by-step followed by their support.

---- Replied Message ----
| From | ***@***.***> |
| Date | 6/13/2024 22:33 |
| To | ***@***.***> |
| Cc | ***@***.***> ,
***@***.***> |
| Subject | Re: [Xilinx/Vitis-AI] vitis-ai-library_3.0.bb:do_compile) failed with exit code '1' (Issue #1399) |

Hello
I run into the same problem. Can you share with me please your work around the issue ?
Thank you.