Xilinx/Vitis-AI #1400: xdputil Fails Checking DPU Status

danielstumpp, 2024-01-25:
Hello, I'm expanding the WAA example with pre- and post- processing shown [here](https://github.com/Xilinx/Vitis-AI-Tutorials/tree/3.0/Tutorials/18-mpsocdpu-pre-post-pl-acc/) for my own application. So far these are the relevant changes I have made:
- Switched target device from ZCU102 to ZCU104
- Updated DPU configuration to enable URAM for ZCU104
- Changed the number of DPU CUs to 1 for resource utilization
- Changed the pre- and post- processing IP to run on 200MHz clock to meet timing

When I load the generated image and run the tests, both the pre- and post-processing accelerators pass the test and operate as expected. The end-to-end accelerator however fails with the following error:

```
F0125 17:51:24.579839  1436 xrt_device_handle_imp.cpp:327] Check failed: r == 0 cannot set read range! cu_index 2 cu_base_addr 2147680256 fingerprint 0x0 : Invalid argument [22]
*** Check failure stack trace: ***
```

I also get the same error when running `xdputil status`:

```
F0125 17:51:24.579839  1436 xrt_device_handle_imp.cpp:327] Check failed: r == 0 cannot set read range! cu_index 2 cu_base_addr 2147680256 fingerprint 0x0 : Invalid argument [22]
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:  1436 Aborted                 /usr/bin/python3 -m xdputil $*
```

The fact that it is seemingly checking CU index 2 is concerning, but I'm not sure why it's doing that or what the actual root cause of this error may be. Does anyone have any ideas? Any help would be much appreciated.

AlbertaBeef, 2024-01-25:
My guess would be that the run-time is referring to a stale (previous) version of the dpu.xclbin file.
Did you change the .xclbin file on your embedded platform ?
I think the default search path for this file is defined in :
   /etc/vart.conf
Make sure the .xclbin file defined in this file is pointing to your .xclbin

danielstumpp, 2024-01-25:
Hello @AlbertaBeef, thanks for the response. I have confirmed that `etc/vart.conf` is pointing to the correct `dpu.xclbin` file. In this case located at `/run/media/mmcblk0p1/dpu.xclbin`.

I also confirmed that this dpu binary has each of the three kernels I expect (see output below). Any other idea what may be causing this issue?

partial output of `xclbinutil --info --input dpu.xclbin`:

```
root@zynqmp-common-20222:/run/media/mmcblk0p1# xclbinutil --info --input dpu.xclbin 
XRT Build Version: 2.14.0 (2022.2)
       Build Date: 2022-10-07 05:12:02
          Hash ID: 43926231f7183688add2dccfd391b36a1f000bea
------------------------------------------------------------------------------
Warning: The option '--output' has not been specified. All operations will    
         be done in memory with the exception of the '--dump-section' command.
------------------------------------------------------------------------------
Reading xclbin file into memory.  File: dpu.xclbin

==============================================================================
XRT Build Version: 2.14.0 (2022.2)
       Build Date: 2022-10-07 05:12:02
          Hash ID: 43926231f7183688add2dccfd391b36a1f000bea
==============================================================================
xclbin Information
------------------
   Generated by:           v++ (2022.2) on 2022-10-13-17:52:11
   Version:                2.14.0
   Kernels:                hls_dpupostproc_m_axi, hls_dpupreproc_m_axi, sfm_xrt_top, DPUCZDX8G
   Signature:              
   Content:                Bitstream
   UUID (xclbin):          a654ee8b-cbef-0ad8-8da0-641b68c74203
   Sections:               BITSTREAM, MEM_TOPOLOGY, IP_LAYOUT, CONNECTIVITY, 
                           BUILD_METADATA, EMBEDDED_METADATA, SYSTEM_METADATA, 
                           GROUP_CONNECTIVITY, GROUP_TOPOLOGY
==============================================================================
Hardware Platform (Shell) Information
-------------------------------------
   Vendor:                 xilinx.com
   Board:                  xd
   Name:                   xilinx_zcu104_base_202220_1
   Version:                202220.1
   Generated Version:      Vivado 2022.2 (SW Build: 3668458)
   Created:
               Tue Oct 11 16:36:36 2022   FPGA Device:            xczu7ev
   Board Vendor:           xilinx.com
   Board Name:             xilinx.com:zcu104:1.1
   Board Part:             xilinx.com:zcu104:part0:1.1
   Platform VBNV:          xilinx.com_xd_xilinx_zcu104_base_202220_1_202220_1
   Static UUID:            00000000-0000-0000-0000-000000000000
   Feature ROM TimeStamp:  0

Scalable Clocks
---------------
   No scalable clock data available.

System Clocks
------
   Name:           clk_wiz_0_clk_out1 
   Type:           FIXED 
   Default Freq:   150 MHz

   Name:           clk_wiz_0_clk_out2 
   Type:           FIXED 
   Default Freq:   300 MHz

   Name:           clk_wiz_0_clk_out3 
   Type:           FIXED 
   Default Freq:   75 MHz

   Name:           clk_wiz_0_clk_out4 
   Type:           FIXED 
[... 350 lines cut ...]
   Port:              M_AXI_HP0
   Memory:            HP0 (MEM_DRAM)

   Argument:          dpu_base4_addr
   Register Offset:   0x80
   Port:              M_AXI_HP2
   Memory:            HP1 (MEM_DRAM)

   Argument:          dpu_base5_addr
   Register Offset:   0x88
   Port:              M_AXI_HP2
   Memory:            HP1 (MEM_DRAM)

   Argument:          dpu_base6_addr
   Register Offset:   0x90
   Port:              M_AXI_HP2
   Memory:            HP1 (MEM_DRAM)

   Argument:          dpu_base7_addr
   Register Offset:   0x98
   Port:              M_AXI_HP2
   Memory:            HP1 (MEM_DRAM)
==============================================================================

```

danielstumpp, 2024-01-25:
@AlbertaBeef Another update here... I updated some of the connectivity to reflect the 1 CU default and now the error message does not refer to a cu_index of 2; it is zero now. It seems from reading elsewhere that the following two lines are separate and are the actual cause:
```
*** Check failure stack trace: ***
/usr/bin/xdputil: line 20:   923 Aborted                 /usr/bin/python3 -m xdputil $*
```

Any insight into the cause of this? The `xdputil` source code doesn't indicate anything obvious to me.

quentonh, 2024-01-25:
@danielstumpp Quick question....what does show_dpu report?  Also, is (and was) Softmax enabled in the DPU IP?

danielstumpp, 2024-01-25:
@quentonh Here is the output of `show_dpu`, same type of behavior:

```
root@zynqmp-common-20222:~# show_dpu
WARNING: Logging before InitGoogleLogging() is written to STDERR
F0125 23:44:38.437368  1160 xrt_device_handle_imp.cpp:327] Check failed: r == 0 cannot set read range! cu_index 0 cu_base_addr 2147680256 fingerprint 0x0 : Invalid argument [22]
*** Check failure stack trace: ***
Aborted
```

I didn't actively enable Softmax, and I don't see an option to do that in `dpu_conf.vh`. Where can I verify that?

danielstumpp, 2024-02-10:
Hi @quentonh @AlbertaBeef,

I ported my design to the VCK190 and am encountering the same issue when I have multiple kernels in my `dpu.xclbin` file. The error message states `cu_base_addr 2147549184` which corresponds the hls preprocessing kernel I have implemented.

It looks to me like vart/xdputil is looking at the first kernel in the xclbin and assuming it is the DPU, which fails in the case where the DPU is not the first kernel. How can I reorder the kernels in `dpu.xclbin`? That seems to be the quickest fix

danielstumpp, 2024-03-15:
For those who may be wondering, this was fixed by reverting to VART 2.5. I have not explored further yet, but it seems that VART 3.0 is mistaking any additional kernels as DPUs and tries to configure the memory space as such, leading to the failure.