Xilinx/Vitis-AI #1403: Can't build docker image for TF2 and GPU

jonGuti13, 2024-02-06:
I am trying to build the Vitis AI 3.5 image for GPU and tf2 by running the following command
```
./docker_build.sh -t cpu -f tf2
```
and it fails, as it can be seen in

[image]

This is a screenshot of my CI pipeline, updated with upstream as of 06/02/2024, which I have included in .

By the way, #1350 has the same problem except for the framework, which is pytorch for them.

vaibuild, 2024-03-15:
Hi @jonGuti13,
Could you please try with the latest commit from master branch, this issue has been solved.

jonGuti13, 2024-03-15:
Hi @vaibuild,

Thank you for your response, but I have tried it and it still fails for me. This a screenshot telling that [my branch](https://github.com/jonGuti13/Vitis-AI/tree/ci) is ahead of Xilinx/Vitis-AI-master.

[image]

The [first commit](776e5399efa6a6d991652f9871e2444da8b522e1) has to do with removing a confirm option for CI, the [second one](da247c10a3952ce2b4449bf8a6a44e2fa3880efa) is the creation of the `.github/worflows/ci.yml` and the [third one](7e6bbea6bded947da23a997cb4ec5799ca568c98) is changing the `.github/worflows/ci.yml` to build image for GPU.

Both actions have failed, although the reason of the one related to the GPU is not being enough space left on device, so I will just attach the screenshot of the [failure of the CPU image](https://github.com/jonGuti13/Vitis-AI/actions/runs/8292936607/job/22695096311) (the one I was originally reporting in this issue).

[image]

vaibuild, 2024-03-15:
Hi @jonGuti13,
 can you try with another server and directly from master? I cannot reproduce your problem, as for the coredump, maybe you can see if this helps?  https://github.com/conda/conda/issues/7815
conda clean -a add it in install_tf2.sh #107 befgore mamba install command excute?

jonGuti13, 2024-03-15:
Hi @vaibuild,

Thank you for your rapid response. I have been able to build both GPU and CPU images locally so the problem has been solved with the latest commit. However, I will try to make it work in CI also, as it is a valuable tool for checking compatibility issues. I will follow your piece of advice and try to fix it. 

I close the issue as the problem has been solved, but I will still add information regarding the CI issue.

closed 2024-03-15 by jonGuti13

jonGuti13, 2024-03-15:
After adding `conda clean -a` in [install_tf2.sh](https://github.com/Xilinx/Vitis-AI/blob/master/docker/common/install_tf2.sh) it fails as a consequence of not being enough space left on device. However, I have removed some of the unnecesary packages of `ubuntu-latest` thanks to a Github Action from @jlumbroso https://github.com/marketplace/actions/free-disk-space-ubuntu and now the CPU image is succesfully built.