Xilinx/Vitis-AI #1411: Can not run the compiled xmodel (pytorch) in inference phase on KV260 (B4096 & B3136)

rezajahadi, 2024-03-08:
Hello,

I have quantized a resnet18 model using the resnet18_quant.py script provided by Vitis AI in the "example/vai_quantizer/pytorch" folder.
The xmodel was successfully created for the quantized model. I used this command to run the script:

python resnet18_quant.py --quant_mode test --model_dir pre_trained_models/ --data_dir tiny-imagenet-200/ --target DPUCZDX8G_ISA1_B3136 --batch_size=1 --deploy --subset_len 1

Then I compile the xmodel by this command:

vai_c_xir -x ResNet_int.xmodel -a /opt/vitis_ai/compiler/arch/DPUCZDX8G/KV260/arch.json -o customized_xmodel_pytorch/ -n resnet18

This command gives me the final xmodel file for DPU implementation. However, when I run DPU using this xmodel on my target device (KV260), I face this error:

""
Kernel Restarting
The kernel appears to have died. It will restart automatically.
"""

KrishnaGaihre, 2024-03-15:
Which DPU image are you running on KV260? Have you checked "[Vitis AI runtime example](https://github.com/Xilinx/Vitis-AI/tree/3.0/examples/vai_runtime/resnet50) on that board" for sanity check if only the boot image is created by you?

rezajahadi, 2024-03-15:
At the time of writing the issue, I couldn't have the inference on KV260.
Right after that, I could run the xmodel on KV260. It was because the Vitis
ai libraries were not installed on Ubuntu OS, that was running on Kria
KV260. I got the last update of Vitis AI image from Xilinx website and
flashed the SD card. The problem was solved.

But I face another problem. Once I want to use the models from pre
trained weights, using the command
"model.load_state_dict(torch.load(pretrained_weights_path))", the model
cant be quantized. The error is that the model is not traceable. Could you
please help me on this matter?